scikit-rf/scikit-rf #402: Allow file-like input to create a network object creation

vik-s, 2020-10-19:
I was trying to create a Network object from a file-like input but the constructor does not seem to allow this. For example:

```
import skrf as rf
f = open('file.s2p')
t = rf.Network(f)
f.close()
```
The above fails to create object. However, the Touchstone class seems to allow object creation with file-like objects. Foe example:

`
t = rf.Touchstone(f)
`

It would be nice if the Network class can support the same style of object creation. The reason I cannot do a conventional file load as in `t = rf.Network('file.s2p')` is because I am writing an app and the file uploader in the app returns a file-like object which I would like to handle using scikit-rf.

Meanwhile, is there a way I can use the Touchstone class to create a network object?

When I do `t.load_data(fid)`, I get a bunch of empty numpy arrays and I'm not sure what is going on. But just want to make sure that I am not missing something obvious.

thanks for your help in advance!

jhillairet, 2020-10-19:
You can use the `read_touchstone()` method :

```python
import skrf as rf
f = open('test.s2p')
t = rf.Network()
t.read_touchstone(f)
f.close()
```

vik-s, 2020-10-19:
That works! Thanks a bunch...

arsenovic, 2020-10-20:
so we say Network can be init'd from a file object, so we should maybe support this. ... ill re-open for now. 

https://github.com/scikit-rf/scikit-rf/blob/master/skrf/network.py#L344-L347

vik-s, 2020-10-22:
If I understand this right, when a file object is passed, the constructor assumes it is a pickle file and then fails. To load the file object, it should be calling the read_touchstone method instead. Is that right? 

I can make a PR for this if I can get it working. Any suggestions?

arsenovic, 2020-10-22:
that sounds right. 

here is the  test which should have failed https://github.com/scikit-rf/scikit-rf/blob/master/skrf/tests/test_network.py#L111-L114.

arsenovic, 2021-02-19:
Reading a network from a stringio interface would be useful, and the way things work now is confusing.

vik-s, 2021-03-19:
I've created [PR#437](https://github.com/scikit-rf/scikit-rf/pull/437) for this issue. By catching a TypeError as an exception, file-like objects can be used for network object construction. I'd love to hear any thoughts on this.

jhillairet, 2021-03-19:
@vik-s , the PR #433  has been merged, which should fix this issue. Could you check?