scikit-rf/scikit-rf #404: .SNP files are not read in

horror-vacui, 2020-11-16:
There are softwares and instruments which saves the touchstone datafile with upper case extention in the filename. For example: my_great_amplifier.S2P. The load_file function in the touchstone.py file raises an exception.

It would be great of skrf would allow SNP extension format as well.

JAnderson419, 2020-11-16:
Hi Horror-vacui,

Would you be able to provide a minimal working example and a copy of the exception raised? I am guessing you are running on Linux? I have no problems trying to load a file with an SNP extension, but I am on Windows and these case-sensitive file issues are usually Linux-only. 

Jackson

jhillairet, 2020-11-17:
Dear @horror-vacui ,

I can't reproduce the problem, even under lInux, for example:

```
In [6]: rf.Network('gamma_f.S1P')                                               
Out[6]: 1-Port Network: 'gamma_f',  1.0-100.0 GHz, 401 pts, z0=[50.+0.j]

In [7]: rf.Network('line2p3mm.S2P')                                             
Out[7]: 2-Port Network: 'line2p3mm',  1.0-100.0 GHz, 401 pts, z0=[50.+0.j 50.+0.j]
```

can you give more details?

horror-vacui, 2020-11-17:
I am terribly sorry guys, I can not reproduce it...
Yesterday the traceback show me an error which led me to the source code, where it seemed to me that the filename extension was only written for lowercase extensions. Renaming all my .S1P files to .s1p fixed my issue without any change in my code, but as it has now turned out, it was not a skrf bug, but some strange interaction with my code.

closed 2020-11-17 by horror-vacui

jhillairet, 2020-11-17:
ok, thank you.