scikit-rf/scikit-rf #405: Data missing for One-Port Two-Tier calibration

scimax, 2020-11-16:
I wanted to go through the example case of the One-port two-tier calibration as explained by @arsenovic in the paper from 2018. There exists an example also in the docs, under [Metrology > One Port Tiered Calibration](https://scikit-rf.readthedocs.io/en/latest/examples/metrology/One%20Port%20Tiered%20Calibration.html). But the repo lacks the data, isn't it? The link provided in the docs is wrong, but nevertheless there exist no directory in the example folder.
Or am I wrong and it is supposed to be somewhere else?

JAnderson419, 2020-11-16:
Hey Scimax,

Is [this](https://github.com/scikit-rf/scikit-rf/tree/master/doc/source/examples/metrology/oneport_tiered_calibration) the data you are looking for?

scimax, 2020-11-16:
Thanks! I'm really sorry! This issue was definitely not necessary. I somehow missed it...

closed 2020-11-16 by scimax

JAnderson419, 2020-11-17:
No problem! It's good you brought this up as it allows us to fix the link for the next person.