scikit-rf/scikit-rf #407: Cannot connect to any PNAs

cafeclimber, 2020-11-21:
Trying to connect to both a FieldFox (N9923A) and an Agilent PNA (not in the lab atm for the model number). Using this code from the docs:

**For the PNA**
```python
from skrf.vi import vna

p =vna.PNA("TCPIP0::10.0.0.3::INSTR")
```

** For the FieldFox**
```python
from skrf.vi import vna

p =vna.FieldFox("TCPIP0::10.0.0.3::INSTR")
```

Both fail at the class instantiation with the following:

```bash
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "$USER/.virtualenvs/scratch/lib/python3.8/site-packages/skrf/vi/vna/keysight_fieldfox.py", line 36, in __init__
    super(FieldFox, self).__init__(address, **kwargs)
  File "$USER/.virtualenvs/scratch/lib/python3.8/site-packages/skrf/vi/vna/abcvna.py", line 106, in __init__
    self.resource.control_ren(2)
  File "$USER/.virtualenvs/scratch/lib/python3.8/site-packages/pyvisa/resources/messagebased.py", line 54, in control_ren
    return self.visalib.gpib_control_ren(self.session, mode)
  File "$USER/.virtualenvs/scratch/lib/python3.8/site-packages/pyvisa_py/highlevel.py", line 325, in gpib_control_ren
    return self.handle_return_value(
  File "$USER/.virtualenvs/scratch/lib/python3.8/site-packages/pyvisa/highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_NSUP_OPER (-1073807257): The given session or object reference does not support this operation.
```

I've installed pyvisa-py as the visa backend, but I'm not sure which is correct. If I can be pointed somewhere, I'm happy to make a PR.

JAnderson419, 2020-11-21:
I have not personally used the skrf bindings as I typically using pyvisa directly, but this looks like it is using pyvisa as the backend correctly as you are getting a VisaIOError. It is likely something to do with how your instruments are set up/connected to the control PC. Looking at [NI VISA documentation](https://zone.ni.com/reference/en-XX/help/370131S-01/ni-visa/vigpibcontrolren/), I found the following:

`This operation is valid only if the GPIB interface associated with the session specified by vi is currently the system controller.`

Is there any other hardware or software connected to the PNA that might be acting as a system controller and locking python out? You could also try instantiation the instrument in pyvisa directly for troubleshooting:

```
import pyvisa

rm = pyvisa.ResourceManager()
pna = rm.open_resource("TCPIP0::10.0.0.3::INSTR")
print(pna.query("*IDN?"))
pna.control_ren(2)
```

I have not used a fieldfox, but on the larger PNAs I believe the status is listed in the bottom status bar (below windows, where it shows calibration status, etc). You may be able to see if it is in LCL or REM there, and whether that changes with your code running.

Hope that helps,
Jackson

cafeclimber, 2020-11-23:
So I went through pyvisa directly and was able to query the FieldFox with `pna.query("*IDN?")` however trying to run `pna.control_ren(2)` raises the same error I was experiencing in my above post:

```bash
>>> pna.control_ren(2)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "$USER/.virtualenvs/scratch/lib/python3.8/site-packages/pyvisa/resources/messagebased.py", line 54, in control_ren
    return self.visalib.gpib_control_ren(self.session, mode)
  File "$USER/.virtualenvs/scratch/lib/python3.8/site-packages/pyvisa_py/highlevel.py", line 325, in gpib_control_ren
    return self.handle_return_value(
  File "$USER/.virtualenvs/scratch/lib/python3.8/site-packages/pyvisa/highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_NSUP_OPER (-1073807257): The given session or object reference does not support this operation.
```

JAnderson419, 2020-11-27:
Hello Bailey, 

My apologies for the delay around the holiday.

Looking online, it seems like a number of different things may cause the viGpibControlREN command in NI VISA. Some people had luck with it using Agilent VISA and some said that certain instruments don't support it. Looking into the skrf codebase, it seems like it was originally tested on a Fieldfox using A.06.23 firmware. As of A.07.75, the instrument supports `INSTrument:GTL` and `INSTrument:GTR` SCPI commands that switch remote mode on and off. This may be an alternative that Keysight wishes users to use instead. The PNA documentation indicates that you can send the GPIB GTL and LLO (local lockout) commands to a PNA and that it should switch to remote automatically if using SCPI commands. This has been my experience using it as well - I typically can connect and send commands for it to execute without first defining remote operation. 

I am not sure what the "correct" way to handle these things are as far as the virtual instrument standards go, but a fix could be to put the control_ren command in abcvna.py in a try statement and throw a warning on exception that skrf could not explicitly exert remote control. As I said, in my experience with a VNA I have never needed to do this before executing code. You could try this on a local fork of skrf and let me know if it works with your instruments. @dvincentwest may have some ideas as well - I believe he wrote the original code for skrf instrument control.

Jackson

cafeclimber, 2020-11-27:
No worries about the delay.  I noticed that function call in `abcvna.py` and tried to execute the commands around it without issue. I think your solution seems appropriate and I'll submit a pull request with this in it after experimenting with the equipment I have. I also considered the firmware issue but haven't had a chance to update the firmware (which I think might be beneficial. It would be nice to not require a firmware update to use the `vi` module.) Thanks!

cafeclimber, 2020-11-27:
So I've found a deeper issue with the `vi` module. It appears to be using an outdated version of `pyvisa` that utilizes some deprecated commands like `query_values()`. I can try to fix this, but it demands rewriting the parser script which generates commands from each instrument class's `.yaml` file. I can try to figure this out, but if you can provide any insight that would be great!

jhillairet, 2020-11-27:
Indeed, it's probably related to #393 and maybe also to the [other issues labelled "instruments"](https://github.com/scikit-rf/scikit-rf/issues?q=is%3Aissue+is%3Aopen+label%3AInstruments)

cafeclimber, 2020-11-27:
Mmk. I'm writing up a PR at the moment. Will post shortly.

closed 2022-08-21 by cafeclimber