scikit-rf/scikit-rf #411: Something wrong in the simulations

Antozamm, 2020-12-03:
Hello

I'm simulating a simple network with two different frequency settings, the first time from 10MHz to 2.5GHz, have a look below
[image]
the second time from 2GHz to 2.5Ghz
[image]
As you can see the second smith chart plot is not a subset of the first plot, as it should be (because is a reduced frequency range).

Is my code wrong or there is something broken in scikit-rf?

thanks

jhillairet, 2020-12-03:
Dear @Antozamm ,

Thank you for reporting.

The behaviour you observe is "normal" (but I admit it can be surprising). The reason is that `DistributedCircuit` is a dispersive media. So when one creates a line defined on an electrical length, the question that arises is: at which frequency does this hold? 

The present choice in scikit-rf is that the translation between time-based units (such as angle in degree) to a distance is defined on the _mean group velocity_ (this is done the the `media.to_meters()` method). In practice, it means that the line delay defined in terms of electric length (in degree) is the one you will have at the mid-band of the `Frequency`. 

Examples: in the figure below you can see that the 100-degree delay is obtained in the middle of the frequency bands
```

import skrf as rf
from skrf.media import DistributedCircuit, RectangularWaveguide

f = rf.Frequency(1, 2, npoints=101, unit='GHz')
tem = DistributedCircuit(f, z0=50)
tem.line(100, unit="deg").plot_s_deg(m=1, n=0)

f2 = rf.Frequency(2, 3, npoints=101, unit='GHz')
tem2 = DistributedCircuit(f2, z0=50)
tem2.line(100, unit="deg").plot_s_deg(m=1, n=0)
```
[image]

So, in your example, the lines does depend of the frequency band they have defined for and thus you get different results. If you want to avoid this, the lines should be defined in terms of unit of length: 

```
def create_circuit(start, stop, npoints):
    f = rf.Frequency(start, stop, npoints, unit='GHz')
    tem = DistributedCircuit(f, z0=50)
    
    gamma1 = rf.zl_2_Gamma0(50,40)
    z1 = tem.load(gamma1)
    l1 = tem.line(107, unit='mm', z0=64)
    c1 = tem.capacitor(100e-9)
    l2 = tem.line(4.3, unit='mm', z0=57)
    c2 = tem.shunt_capacitor(0.9e-12)
    l3 = tem.line(11.1, unit='mm', z0=57)
    c3 = tem.shunt_capacitor(1.8e-12)
    
    match = c3 ** l3 ** c2 ** l2 ** c1 ** l1 ** z1
    return match

create_circuit(1, 3, npoints=1001).plot_s_db()
create_circuit(1, 2, npoints=1001).plot_s_db(ls='--')
create_circuit(2, 3, npoints=1001).plot_s_db(ls=':')
```

[image]

Antozamm, 2020-12-04:
Dear @jhillairet 

thank you for taking the time to llok into it, and for the good explanation.

"at which frequency does this hold?", of course! I should have relized by myself!
Indeed, in commercial tools one define the electrical length and the frequency where it apply.

Btw, once you know it is just a matter of getting used to it.

Thanks again!

Have a nice weekend

closed 2020-12-04 by Antozamm