scikit-rf/scikit-rf #413: Media class cannot create lengths of mismatched transmission line

DavidPowell, 2020-12-10:
I have created a transmission line using the `skrf.media.DefinedGammaZ0`, where I defined both `Z0` (the characteristic impedance of the transmission line), and `z0` (the reference impedance for measurement) to be different from each other.

However, when I use the `line()` method on my media, it ignores the characteristic impedance `Z0`, and assumes that the transmission line is matched.

Looking at the [source code](https://github.com/scikit-rf/scikit-rf/blob/0ee9e387dd537b46f9f9d9fac61209cf8289e4f6/skrf/media/media.py#L745-L746), I see that the `line()` method always assumes that the reference impedance is the same as the characteristic impedance, thus it always gives |S_{11}|=0 and |S_{21}|=1.

A more general approach would be to use the known ABCD matrix solution for a transmission line, which can then be converted to S parameters with arbitrary reference impedance

[image]

I need mismatched lengths of transmission line for modelling a plane wave impinging on a dielectric layer. I can imagine many other situations where they would also be needed, such as stepped impedance filters.

arsenovic, 2020-12-10:
you can use line() to create mismatched lines.  you need to use the embed option. 
https://scikit-rf.readthedocs.io/en/latest/api/media/generated/skrf.media.media.Media.line.html#skrf.media.media.Media.line

post your code.  if you still have problems.

DavidPowell, 2020-12-22:
@arsenovic thanks for the info. Using the `embed` option doesn't quite do what I want, because the resulting network has a reference impedance equal to the characteristic impedance of the line.

However, if I cascade this section of line with S parameters calculated with the required reference impedance, I see that the the network is automaticlaly transformed to the correct S parameters. So I think this suffices for my needs.

closed 2020-12-22 by DavidPowell

arsenovic, 2020-12-22:
ok good deal. i use media.thru() a lot for this type of thing.