scikit-rf/scikit-rf #419: No display found. Using non-interactive Agg backend.

Vinc0110, 2021-01-21:
There is an issue related to the matplotlib output in the Jupyter notebooks. In basically *every* notebook for the tutorials and the examples for skrf on [readthedocs](https://scikit-rf.readthedocs.io), the first `import skrf` results in an ugly warning message and most (?) subsequent plot outputs are suppressed.

See for instance the [Networks tutorial](https://scikit-rf.readthedocs.io/en/latest/tutorials/Networks.html):

1. Warning after the import in code block [1]: `No display found. Using non-interactive Agg backend.`
2. Call of `ntwk.plot_s_db()` in code block [5] does nothing (not plot).
3. The plots from code blocks [16], [17] and everything below work for some reason.
4. Maybe the execution of code block [15] did the trick? Unfortunately, it throws another warning: 
`/home/docs/checkouts/readthedocs.org/user_builds/scikit-rf/envs/latest/lib/python3.7/site-packages/skrf/plotting.py:1446: UserWarning: Style includes a parameter, 'interactive', that is not related to style.  Ignoring
  mpl.style.use(os.path.join(pwd, style_file))`

Ironically, all of the plotting examples in the [Plotting Tutorial](https://scikit-rf.readthedocs.io/en/latest/tutorials/Plotting.html) fail. They work fine if I run the notebook on my local machine.

It all seems to come from this setting:
https://github.com/scikit-rf/scikit-rf/blob/49d80d3adc4905b8a281162d471763acdddbe59e/skrf/plotting.py#L60-L64

Is there a good reason for this setting or can we maybe just delete that? I don't see the point. To be clear: this is not about the warning, but about the missing plot outputs.

I'd be happy to implement the changes (delete the backend reconfiguration) and create a PR.

jhillairet, 2021-01-21:
Indeed. This change was intended to solve the failing tests with Python 2 on Github Actions, but I didn't pay attention to the effect on the documentation rendering (but some plots remain, don't know exactly why). 

[As you noticed, ](https://github.com/scikit-rf/scikit-rf/pull/416#issuecomment-757189103), the cause of the Python2 bug was related to something else. So please do.

But finally, the bug with Python2 came from Please do.

Vinc0110, 2021-01-21:
I hope my PR actually works. It's hard to verify, because I cannot reproduce the problem locally with my Python3.8. As I wrote above, the notebooks are fine in any case when I create the docs on my local machine.

closed 2021-01-21 by jhillairet