scikit-rf/scikit-rf #431: Problem creating network from s4p file

amenychaabanii, 2021-02-25:
Good morning , 
I jsut started using scikit-rf  and I have problem creating a network from a s4p file .I get the following error s :
from skrf.data import meas
Traceback (most recent call last):

  File "<ipython-input-19-02f15f787888>", line 1, in <module>
    from skrf.data import meas

ImportError: cannot import name 'meas' from 'skrf.data' (C:\Users\Achraf\anaconda3\lib\site-packages\skrf\data\__init__.py)

example1=rf.Network('data/meas.s4p')
Traceback (most recent call last):

  File "<ipython-input-20-7d6659bbb526>", line 1, in <module>
    example1=rf.Network('data/meas.s4p')

  File "C:\Users\Achraf\anaconda3\lib\site-packages\skrf\network.py", line 420, in __init__
    fid = get_fid(file, 'rb')

  File "C:\Users\Achraf\anaconda3\lib\site-packages\skrf\util.py", line 168, in get_fid
    return open(file, *args, **kwargs)

FileNotFoundError: [Errno 2] No such file or directory: 'data/meas.s4p'

arsenovic, 2021-02-25:
> File "", line 1, in

you probably have to go find where the file is ?
which example are you working with?

you could give it the full path like `rf.Network('/full/path/to/meas.s4p')`

also checkout the mailing list, perfect place for use-questions like this.

amenychaabanii, 2021-02-25:
Thank you for your answer. I am just following the tutorials on scikit-RF webpage https://scikit-rf.readthedocs.io/en/latest/tutorials/index.html . I wanted to try this with an s4p file that I have  but i still have errors , even though  i wrote the full path to the s4p file : 
example1=rf.network('C:/Users/Achraf/meas.s4p')
Traceback (most recent call last):

  File "<ipython-input-14-73cef53ba99f>", line 1, in <module>
    example1=rf.network('C:/Users/Achraf/meas.s4p')

TypeError: 'module' object is not callable

arsenovic, 2021-02-25:
use capital N for network. the network is a  module. Network is a class.
a little confusing as first, but this is common in python.

amenychaabanii, 2021-02-26:
Solved Problem.Thank youu

closed 2021-02-26 by jhillairet