scikit-rf/scikit-rf #434: Non Harmonic sweep

FranzForstmayr, 2021-03-11:
I'm using a non-harmonic sweep and wanted to calculate the TDR. I recognized, that the step_response function does not check if the sweep is harmonic and assumes it is. 
Wouldn't it be better, if the frequency class could check if the sweep is harmonic, so that the step_response could use a short time fourier transformation or interpolate the support points?
Can you confirm this behavior or did I just something wrong?

jhillairet, 2021-03-13:
What do you mean by non-harmonics: frequency points are not evenly spaced (changing frequency step size)? 

Indeed, the code does use standard fft for time gating, so it will not work properly on unevenly spaced data. 

You are right that a test should be made somewhere to indicate users that something is wrong. It has to be discussed if the data should be interpolated under the hood silently or not... could you provide a data example so that one could add this feature and forge a dedicated test?

FranzForstmayr, 2021-03-15:
Yes, I changed the step size using a segmented sweep in a R&S VNA. I think there would be an similar issue using an logarithmic sweep.

I would say either throw a Warning and interpolate or throw an Error that the feature is not implemented (yet).
Probably the mentioned non-uniform fourier transformation could help without interpolating. 

I can deliver test data for a segmented use case. Should I integrate it in skrf/tests and create a testcase?

jhillairet, 2021-03-15:
I would personally be in favour of throwing a not implement error, as interpolating may result in specific case-to-case problems... If you want to make a pull request, with a test that check an error is raiser in such case, you are very welcome!

FranzForstmayr, 2021-03-16:
I'll give it a try :)

jhillairet, 2021-03-29:
fixes with #435. thank you!

closed 2021-03-29 by jhillairet