scikit-rf/scikit-rf #440: Plots in readthedocs don't work (Agg backend, again)

Vinc0110, 2021-04-01:
The old problem with the `Agg` backend for matplotlib seems to be back. 
The issue is the same as already reported in issue https://github.com/scikit-rf/scikit-rf/issues/419, so most plot outputs of the Jupyter notebooks on readthedocs don't work anymore. A recent change must have caused it, maybe related to PR https://github.com/scikit-rf/scikit-rf/pull/438?

jhillairet, 2021-04-01:
where did you spotted the problem? On your own RTD rendering? (the official is still OK because it is run manually on demand currently)

Vinc0110, 2021-04-01:
This is very strange. It's about the official RTD. When I check now, everything is fine again, but I am 100% sure the plot outputs were missing 6 hours ago when I created this issue.

jhillairet, 2021-04-01:
well, the last time they have been rendered was a month ago.

Vinc0110, 2021-04-02:
I guess we can close this issue for now. Maybe the browser can sometimes get stuck while loading the plots?

closed 2021-04-02 by Vinc0110