scikit-rf/scikit-rf #443: Applicable to discrete component assemblies for HF?

jaredd, 2021-04-09:
I ran into this package as I was researching potential enablers for a frequency response analysis tool I am conceptualizing to support the design of a high frequency (HF) (i.e. 3 MHz - 30 MHz) prototype radio. 

My idea was to have a composable set of objects that represented discrete components who's dependent characteristics (either conventional datasheet specs like insertion loss, gain, etc.) or s-parameters could be used to describe its frequency response. I was also going to include tolerances (e.g. max input power) and constraint blocks (e.g. a spectrum mask) to ensure part / regulatory compliance. By linking component objects' ports I could design and analyze the overall response of an RF chain like you'd have in an RF electronics lab, including some of the non-linear affects.

I see there's a Device class but I'm still unsure whether this package would be a natural starting point for creating say, a heterodyne front-end consisting of filters, amplifiers, and mixers?

I would certainly be willing to contribute if I can make use of this framework.

Any advice would be appreciated.

jhillairet, 2021-04-10:
In order to build a circuit composed of lumped elements and N-ports devices (described by their S/Z/Y/ABCD-params), you can use the `Circuit` class. You will find [a tutorial here](https://scikit-rf.readthedocs.io/en/latest/tutorials/Circuit.html) and there are in addition also[ few examples](https://scikit-rf.readthedocs.io/en/latest/examples/index.html#circuit) available in the documentation. 

The `Device` class probably does not fit your need and is still under development. It could be the opportunity to develop more this aspect eventually.

Have also a look at [vector fitting tools](https://scikit-rf.readthedocs.io/en/latest/tutorials/VectorFitting.html), they can be of use for active circuit modelling.

Vinc0110, 2021-04-10:
That sounds like a fun project, but probably not a simple one.

It seems like you don't just want to connect a bunch of linear networks, but also include nonlinear active devices. For that, I don't think scikit-rf is sufficient, but it might still be helpful:

Why not start with a proper circuit simulator such as SPICE? I can recommend QUCS and specifically [QUCS-S](https://ra3xdh.github.io/) as a GUI if you want to use SPICE. It seems like the natural approach, as you application is a (partly nonlinear) circuit. There are many device macromodels available for SPICE, for example from [Analog Devices](https://www.analog.com/en/design-center/simulation-models/spice-models.html), and many more.

For the (linear) frequency responses, such as filter characteristics, you could either directly implement the respective circuit with its components (R, C, L) in SPICE, or use the datasheet information as you already mentioned. For that, scikit-rf can be useful, as @jhillairet already pointed out. In addition to the scikit-rf `Circuits` feature, the `VectorFitting` class can export equivalent circuits of those `Networks` for SPICE. Have a look at [this example](https://scikit-rf.readthedocs.io/en/latest/examples/vectorfitting/vectorfitting_ringslot.html).

jaredd, 2021-04-12:
Thanks to everyone for these suggestions. The VectorFitting feature seems particularly relevant. I had not thought of using SPICE as it's been 20 years since I've last used that in undergrad. We use a lot of mini-circuits components, so it seems they only have models for Keysight SystemVue (closed source, commercial).

While doing some further research based on this feedback I ran into this tool by Analog Devices called [ADISimRF](https://www.analog.com/en/design-center/adisimrf.html) which is about as close to what I'm looking for as I think exists right now. I imagine it would be a heavy lift to replicate in a reasonable amount of time. Again, I appreciate all the comments.

closed 2021-04-14 by jhillairet