scikit-rf/scikit-rf #449: Typehints in scikit-rf

FranzForstmayr, 2021-04-19:
I would like to add python typehints in skrf, to improve the quality of the autosuggestions in IDEs.
I use vscode, which takes the information out of typehints, I think most IDEs do it like that.

I would start a pull request, if this is ok?

arsenovic, 2021-04-19:
it is welcomed.

FranzForstmayr, 2021-05-05:
Need your opinion here...

Type annotations, especially with mypy, work very well with pure python types, but get tricky with numpy types.
For example this function:
```python
def complex_2_magnitude(input):
    '''
    returns the magnitude of a complex number.
    '''
    return abs(input)
```
returns a `float` when called with any Python `number` and returns a `npy.ndarray`, if called with an `npy.ndarray`. So the return type would basically be a `Union`, however `Union`s as return type is not recommended.
https://mypy.readthedocs.io/en/stable/kinds_of_types.html#union-types

The big players (Numpy/Scipy/Pandas) haven't adopted annotations completely, so there's less documentation about this right now.

**Inline Type Hints**
This annotations are interpreted correct when doint the following (in vscode at least)

```python
@overload
def complex_2_magnitude(input: complex) -> float: ...
@overload
def complex_2_magnitude(input: npt.ArrayLike) -> npy.ndarray: ...

## simple conversions
def complex_2_magnitude(input):
    '''
    returns the magnitude of a complex number.
    '''
    return abs(input)
```
This method works for most of the function in math_functions.py, however we would add lot of boilerplate code.

**Type Hints in a stub file**
To minimize boilerplate code within the functions, it's also possible to generate stub files, like in this repository:
https://github.com/python/typeshed/tree/master/stubs
Typeshed is a collection of stubfiles, especially to support static typing in IDEs and for mypy.
One yould also add and package the stubs directly in the skrf repository.
However another file containing stubs seem to me prone to get out of sync.

The first method seems reliable, with the drawback to add lot's of code.
Maybe anyone has some thoughts about this issue?

PS:
Most of the other files could work very well, but I wanted to start with the base functions, as a bottom up approach looks best to me. That's why the math_functions made me some headache.

jhillairet, 2021-05-06:
> **Inline Type Hints**
> This annotations are interpreted correct when doint the following (in vscode at least)
>
> ```python
> @overload
> def complex_2_magnitude(input: complex) -> float: ...
> @overload
> def [...]

I personally don't like much of this overloading things, at the end it looks like Java ;-)... I though that `ArrayLike ` was the recommand way, but I've not looked into to be honest.

FranzForstmayr, 2021-05-06:
> I personally don't like much of this overloading things, at the end it looks like Java ;-)... I though that `ArrayLike ` was the recommanded way, but I've not looked into to be honest.

I thought so too, however ArrayLike can be neary anything from integers to nested sequences of strings, as `ArrayLike` means you can cast the value into a `npy.ndarray`.
This makes sense for input types, but don't work very well with return types.

I'm not very satisfied of this approach too :D

Vinc0110, 2021-05-06:
Isn't this kind of flexibility due to undefined/dynamic data types one of the big advantages of Python? To me, it seems counterproductive to try and force (define) some sort of data type in places where they are simply not well defined or ambivalent.

Of course, type hints can be useful, but maybe it is better to leave them out where it's not that clear?

jhillairet, 2021-05-06:
> Of course, type hints can be useful, but maybe it is better to leave them out where it's not that clear?

I agree with that.

adamtec, 2021-05-06:
Just chiming in. I use scikit-rf a lot and would love type hints to be added. While the dynamic, duck typing *feature* of Python can be useful it also leads to subtle bugs and becomes problematic on larger code bases. For example if you take the function `complex_2_magnitude` above it uses built-in `abs` whereas `complex_2_db` uses numpy `abs`. If you pass a numpy array to `complex_2_magnitude` you'll get back a numpy array but if you pass a python array it’ll raise an exception. However, for  `complex_2_db` if you pass either you always get back a numpy array. It’s a rather contrived example but type hints could certainly help.

FranzForstmayr, 2021-05-06:
This was also my first thought, and it works well for python objects. Numpy functions behave differently, and there's not even guide how to handle typing for this case. Hopefully numpy will add a sophisticated method in the near future, as there are much more libraries which depend on this. I saw similar discussions in scipy, scikit-learn and other libraries.

Some functions are straight forward to migrate, for these I'll add types as far as possible. It makes sense most of the time, as long we don't return `Unions` or `ArrayLike`, as they are not handlet very well by the IDEs. (It's perfectly okay to accept Unions for Inputs though)

Adding two overloads for a oneliner function seems a overkill to me, but I'm open for future improvements in this area.

FranzForstmayr, 2021-05-08:
From an users perspective, type annotations in network.py and frequency.py are very clear, as the return type is always constant. The math_functions are not called from users anyway. At least not from me :)

jhillairet, 2021-05-26:
A lot has been done in #451. I keep the issue open for the moment, for future discussions

jhillairet, 2021-08-29:
A more lot  has been made recently. I think we can close it for now.

closed 2021-08-29 by jhillairet