scikit-rf/scikit-rf #462: Example notebooks are broken: ModuleNotFoundError

Vinc0110, 2021-05-07:
The example notebooks on RTD are broken: `ModuleNotFoundError: No module named 'skrf'`.
I'm not yet sure what's going on.

jhillairet, 2021-05-07:
seems linked to the suppression of the line : 
`git+https://github.com/scikit-rf/scikit-rf.git@master`
in `requirements.txt`. 

But there is the following line in `doc/source/conf.py`:
`sys.path.insert(0, os.path.abspath('../../'))`
which should allow importing the package correctly... I'm puzzled.

FranzForstmayr, 2021-05-07:
> But there is the following line in `doc/source/conf.py`:
> `sys.path.insert(0, os.path.abspath('../../'))`
> which should allow importing the package correctly... I'm puzzled.

This is only valid for the conf.py file I assume, the runipy will run in a seperate session.

jhillairet, 2021-05-07:
OK, fixed with #464

closed 2021-05-07 by jhillairet