scikit-rf/scikit-rf #466: Problem with the circuit builder - “LinAlgError: Singular matrix” error

andresmmera, 2021-05-13:
Hello all,

As reported in the scikit-rf google group, the bandpass filter example [1] is not working. I copied and pasted it in my jupyter notebook just changing the freq definition line, but it breaks where the connections matrix is passed and it gives the following error: “LinAlgError: Singular matrix”.

Please find the source and the complete error log in the links [2] and [3].

[1] https://scikit-rf.readthedocs.io/en/latest/examples/circuit/Lumped%20Element%20Circuits.html#Pass-band-filter

[2] Complete error log: https://gist.github.com/andresmmera/c459801297078a5bc7120a5f55b65131

[3] Source: https://gist.github.com/andresmmera/42e6c29625f55251fa90689c22d8ccc5

jhillairet, 2021-05-13:
Thank you for reporting. 

This bug is interesting:
- the code passes under my Windows 10 Python 3.7.8 and 3.8.10
- it passes under my Linux Python 3.7.7 but fails with Python 3.8.8
- The example is tested in the Continuous Integration which is supposed to be Linux environment : it passes Python 3.6, 3.7 and 3.8, 

@andresmmera what is your configuration?

andresmmera, 2021-05-13:
Thank @jhillairet for your support.

I'm using Python 3.8.0 (through a jupyter notebook) with Linux. My system settings are the following:

  Operating System: Ubuntu 18.04.5 LTS
  Kernel: Linux 4.15.0-20-generic
  Architecture: x86-64

[image]

[image]

jhillairet, 2021-05-13:
Thank you. The error comes from the frequency point in your code which is exactly 0, which creates NaN in the scattering parameters (for the inductance) and Numpy can honestly complain about that.  

Could you retry changing:

`freq = rf.Frequency(start=0, stop=1, npoints=501, unit='GHz')`

into, for example:

`freq = rf.Frequency(start=0.1, stop=1, npoints=501, unit='GHz')`

andresmmera, 2021-05-13:
Indeed, having f_start > 0 solved the problem.

Thank you very much for your help!

closed 2021-05-13 by andresmmera