scikit-rf/scikit-rf #467: Problem with Network.is_symmetric()

Vinc0110, 2021-05-14:
There seems to be an issue with the `is_symmetric()` function in the `Network` class. Instead of a return value, I'm getting an error: `ValueError: shape too large to be a matrix.`

Apart from this error, the use of NumPy's `Matrix` class is [no longer recommended](https://numpy.org/doc/stable/reference/generated/numpy.matrix.html). Maybe it's time to upgrade `is_symmetric()`?

Minimum example:
```
import skrf
nw1 = skrf.data.ring_slot
nw1.is_symmetric()
```
Error output:
```
  File "/home/vinc/PycharmProjects/scikit-rf/skrf/network.py", line 1600, in is_symmetric
    mat = npy.matrix(test_network.s)
  File "/home/vinc/miniconda3/envs/skrf-dev/lib/python3.8/site-packages/numpy/matrixlib/defmatrix.py", line 135, in __new__
    new = data.view(subtype)
  File "/home/vinc/miniconda3/envs/skrf-dev/lib/python3.8/site-packages/numpy/matrixlib/defmatrix.py", line 180, in __array_finalize__
    raise ValueError("shape too large to be a matrix.")
ValueError: shape too large to be a matrix.
```

jhillairet, 2021-05-18:
Yes indeed. It also means this function is not tested.

Vinc0110, 2021-05-18:
There is a testcase, but apparently only with single-frequency parameters. That might be why this issue was not caught by the tests.

For reference, here is one of the tests:
https://github.com/scikit-rf/scikit-rf/blob/aaf6dee8eaa70038fbf388120dc0bf4ab33b13ff/skrf/tests/test_network.py#L594-L602

Vinc0110, 2021-07-05:
Before I continue and write a fix for this bug, I would like to discuss a general issue related to the comment in `Network.is_symmetric()` and the entire functionality of this function:
https://github.com/scikit-rf/scikit-rf/blob/f43c748fc767b2633f52778a24a0382027ab5c26/skrf/network.py#L1557-L1563

I don't think this is correct.

To my knowledge and from what I've found online (links below), only a reciprocal network has a symmetric S matrix. Strictly speaking, the network itself is not symmetric, but the S matrix is. In that case, `S = transpose(S)` is true. That means `S_ij = S_ji`, but it does **not** mean `S_ii = S_jj`, even though the Wikipedia article says so. The way I see it, network reciprocity and S matrix symmetry are two terms for the same thing. Moreover, these terms do not imply network passivity (even though the report below says so).

Unfortunately, I currently don't have access to a decent microwave textbook to properly look this up. I hope someone of you can help out here.

On [microwaves101.com](https://www.microwaves101.com/encyclopedias/basic-network-theory#reciprocal):
`A reciprocal network always has a symmetric S-parameter matrix. That means that S21=S12, S13=S31, etc. All values along the lower-left to upper right diagonal must be equal.`

In [an old report](https://apps.dtic.mil/dtic/tr/fulltext/u2/a213437.pdf), page 8 in the PDF:
`An important result from N-port network theory is that when an N-port network is passive and reciprocal, its scattering matrix S is symmetric, i.e., Sj = Sji.`

Vinc0110, 2021-07-05:
I just found another definition of n-port symmetry in a german rf textbook referring to the (spatial) layout symmetry of the network, which confirms the current implementation of `Network.is_symmetrc()` based on the reflection coefficients. 
Still, I find the hole topic is rather confusion, because I'm thinking of mathematical (matrix) symmetry when I hear that expression.

Vinc0110, 2021-07-26:
There does not seem to be much interest in this discussion, so I'm leaving the functionality as is and only fix the implementation.

For future reference, see these lecture notes on circuit symmetry for a better explanation: http://www.ittc.ku.edu/~jstiles/723/handouts/Circuit%20Symmetry.pdf

jhillairet, 2021-07-27:
> To my knowledge and from what I've found online (links below), only a reciprocal network has a symmetric S matrix. Strictly speaking, the network itself is not symmetric, but the S matrix is. In [...]

I agree with your analyze. Maybe @arsenovic has the history about the name and purpose of this function?

PR #502 (2021-07-27): Bugfix for issue #467 and extended testcase.
Fixes #467
closed 2021-07-27 by jhillairet