scikit-rf/scikit-rf #468: How can I propagate a differential signal over Channel with given S-parameters

caffery-chen, 2021-05-18:
Is there a simple API function for this purpose?

denzchoe, 2021-05-18:
See here:
https://scikit-rf.readthedocs.io/en/latest/examples/mixedmodeanalysis/Mixed%20Mode%20Basics.html

here:
https://scikit-rf.readthedocs.io/en/latest/api/generated/skrf.network.Network.html#skrf.network.Network

and here:
https://groups.google.com/g/scikit-rf/
<use search term, mixed-mode>

caffery-chen, 2021-05-18:
The pages help getting the mixed mode S parameter and it works fine. My question is if I have a differential signal say X= [x1,x2] with x1 represents time domain positive signal and x2 the negative signal. 

Can I do something like Y = S_{mixed mode} * X and the Y will be the signal that passes this channel? Is there an API for this?

JAnderson419, 2021-05-20:
Hello,

There may be a better way to do this (if there isn't, we should probably work on an API for it) but here is an example that I believe multiplies an input signal through an s-parameter network and gives you an output signal. If you are using the skrf default mixed mode matrix, your signal ports should be [diff+, diff-, com+, com-], so your input signal would just have zero for the last two input ports. Your output signal would be in the same format. Just be aware of port impedances if you are trying to do other calculations or transformations, as your resulting signal array will have z0=[100, 100, 25, 25]. 

```
import skrf as rf
import numpy as np
#  Backend TkAgg is interactive backend. Turning interactive mode on.

n = rf.Network(r'BFP540_w_noise_VCE_2.0V_IC_12mA.S2P')
#  2-Port Network: 'BFP540_w_noise_VCE_2.0V_IC_12mA',  0.01-6.0 GHz, 210 pts, z0=[50.+0.j 50.+0.j]

X = np.array((np.ones(210),np.ones(210)*.5)).T #  generate dummy input power signal for two ports, with 210 points to match len( n.f )
Y = np.zeros(np.shape(X),dtype = 'complex_')
for f,i in enumerate(X[:,0]):
    Y[f] = np.matmul(n.s[f,:,:], X[f,:])

#  Double check the matmul at one freq

n.s[0,:,:]
#  array([[ 6.30154217e-01-1.76018011e-02j, -1.47161080e-04+7.86348279e-04j],
#         [-2.92136056e+01+8.16009892e-01j,  9.66868565e-01-2.53183444e-02j]])
X[0,:]
#  array([1. , 0.5])
Y[0,:]
#  array([  0.63008064-0.01720863j, -28.73017133+0.80335072j])
```

jhillairet, 2021-06-02:
@caffery-chen, is the solution proposed fits your needs?

caffery-chen, 2021-06-02:
Dear @JAnderson419 @jhillairet , the solution is about right. I will have a try and come back. Thanks.

closed 2021-06-02 by caffery-chen