scikit-rf/scikit-rf #470: HFSS Touchstone File - Incorrect Port Impedance Array

lstuemke, 2021-05-18:
I'm having a problem with sNp files exported from HFSS having the wrong port impedance. Specifically, every port after port1 is labeled as 50ohm.

In skrf.io.touchstone, there is a function called get_gamma_z0() that expects the port impedance matrix to be a single line:

[image]

However, the port impedance matrix is not always listed as a single line. For example:

[image]

Clearly, each of the 4-ports is 50ohm. But since get_gamma_z0() only looks at the line starts with '! port_impedance', it misses the other 3 matrix rows, labeling them as 0ohm.

It looks like there's code that might address this in the 'else' clause, but this code never executes because nb_elem is in fact equal to 2*self.rank:

[image]

jhillairet, 2021-05-18:
Thanks for reporting. Is it a Touchstone file from an HFSS Driven terminal solution or from HFSS 2020 R1?

lstuemke, 2021-05-18:
It's Driven Modal, HFSS 2020R1.

jhillairet, 2021-05-18:
Yes. That very particular version of HFSS exports gamma and Z0 in matrix form (why?), then went back to the previous convention in the following version. (Note matrix form is also the case when generated from a Driven terminal solution.) Can you upgrade your version? kind of duplicate of #354

jhillairet, 2021-05-18:
I mean, it's not a bug from scikit-rf, it's a change of the way HFSS formats its Touchstone file (gamma and z0 at not part of the Touchstone format, so it's a feature from HFSS they implement into the comments). Since they have changed back this formatting in 2020R2 and later, the question for us is should we support this particular version of HFSS?

lstuemke, 2021-05-18:
Ahh okay, yeah if they changed back in the next release then this should go away when I update my Ansys. No need to support 2020R1 on my account.

closed 2021-05-18 by lstuemke

jhillairet, 2021-05-18:
a workaround for HFSS 2020R1 is to manually set the characteristic impedance :
`ntw.z0 = 50`
and eventually the same for gamma if you need it...