scikit-rf/scikit-rf #475: ModuleNotFoundError: No module named 'pandas'

sgherbst, 2021-05-30:
Thanks for creating this awesome library!  Just wanted to report that, starting a few days ago, the latest version of ``scikit-rf`` won't load without manually installing ``pandas``:

```shell
$ pip install scikit-rf
$ python -c "import skrf"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/path/to/skrf/__init__.py", line 15, in <module>
    from . import network
  File "/path/to/skrf/network.py", line 170, in <module>
    import pandas as pd
ModuleNotFoundError: No module named 'pandas'
$ pip install pandas
$ python -c "import skrf"
(works)
```

Looks like the related change was made in a10aaf9afe6e0cae3dc4b24f9aae6f5418957f03 four days ago.  For completeness, I am running this on Python 3.7.10 on macOS 10.15.7, although I originally saw this issue because a nightly regression test for one of my projects started failing.

FranzForstmayr, 2021-05-30:
There's already an PR regarding this issue: #474

sgherbst, 2021-05-30:
Thanks, sorry I missed that.  Will close the issue.

jhillairet, 2021-05-30:
> Looks like the related change was made in [a10aaf9](https://github.com/scikit-rf/scikit-rf/commit/a10aaf9afe6e0cae3dc4b24f9aae6f5418957f03) four days ago. For completeness, I am running this on Python 3.7.10 on macOS 10.15.7, although I originally saw this issue because a nightly regression test for one of my projects started failing.

Thanks for reporting. Seems like we should release a minor version asap.

jhillairet, 2021-06-02:
should be close with #474 and integrated into the next version. Please let us know if it resolves your regression tests.

By the way, could give us some info on your project? We are always interested in the various use cases.

sgherbst, 2021-06-04:
Thanks!  The regression test now passes with the latest version of ``scikit-rf``.

In terms of my use case, I am using ``scikit-rf`` to read Touchstone S4P files representing channels in high-speed wireline links.  The S-parameters are used to generate synthesizable analog models for use in FPGA emulation.  The main application for this is the DragonPHY chip design (https://git.io/dragonphy), although the specific regression test that failed was in ``msdsl`` (https://git.io/msdsl), which is my emulation model generation tool.

jhillairet, 2021-06-25:
I'm closing this issue which should have been solved with #474

closed 2021-06-25 by jhillairet