scikit-rf/scikit-rf #486: Academic paper for scikit-rf

jhillairet, 2021-06-29:
An important number of the scikit-rf users present their works during conferences or in papers published in journals. It has already been asked "how to cite scikit-rf?" (#73) and [this question is also answered in the wiki](https://github.com/scikit-rf/scikit-rf/wiki/Citations-Acknowledgments). 

A previous attempt has been made a few years ago to publish a paper presenting scikit-rf, but was finally not done. It is maybe time to retry now, to: i) make a reference paper about scikit-rf (*), ii) ease future citations,  and iii) inform and spread the toolkit to more users and community.

Few options were considered considering the target journal. The [Journal of Open Source (JOSS)](https://joss.theoj.org/) was a logical choice, however, the RF community will not be directly reached. Instead, we proposed to target a IEEE journal, for example [IEEE Journal of Microwaves](https://mtt.org/publications/journal-of-microwaves/information-for-authors/) which is:
> a fully open access publication covering the complete scope of the Microwave Theory and Techniques Society which includes articles on the theory, techniques and applications of guided wave and wireless technologies and spanning the electromagnetic spectrum from RF/microwave through millimeter-waves and terahertz, covering the aspects of materials, components, devices, circuits, modules, and systems which involve the generation, modulation, demodulation, control, transmission, sensing and effects of electromagnetic signals.

Of course, this choice can be discussed...

A draft paper is currently under preparation (using Overleaf). If you have contributed to scikit-rf and/or if you want to help us to read/review/improve the paper and having your name and affiliation mentioned in the list of the authors, please let us know here or in [Slack](https://scikit-rf.slack.com/). 

*: like [Numpy ](https://www.nature.com/articles/s41586-020-2649-2)papers, [scikit-learn](http://jmlr.org/papers/v12/pedregosa11a.html), [scipy](https://www.nature.com/articles/s41592-019-0686-2), etc

vik-s, 2021-06-29:
This is a great idea! Please count me in. I'd like to help in any way I can, including help review/proof the paper.

dvincentwest, 2021-06-29:
I think its a great idea.  I remember the proposal flaming out a few years ago, which is unfortunate, but I think that usage has really ticked up in the past few years and is likely to continue to grow.  As for as authorship, I'd be happy to look over the draft and contribute where I can, and I think that if the paper highlights instrument control or the graphical components at all, then it would be appropriate to be an author on the paper, with affiliation being listed as 3M Company.  If those components are not to be addressed in the paper, then maybe its a bit more ambiguous.

FranzForstmayr, 2021-06-29:
Great idea, I'm happy to help where needed :)

Vinc0110, 2021-06-29:
Cool! I'm offering to write about the vector fitting feature and do some general review regarding network theory.

JAnderson419, 2021-06-29:
I am glad to see so much interest in the paper! One comment is that there is a link to the Github contributors page in the acknowledgements, so even if you are not directly involved in the writing/review of the paper your contributions will be noted. 

That being said, I started updating the summary of features with bullet points on newer stuff like vector fitting and Dash-based GUIs, but I think having more info on alternative GUIs by Vince W. and the vector fitting/SPICE extraction would be excellent. It would not be a bad idea to mention the type-hinting work as well, and perhaps Vikram could check over and update the previously written sections on deembedding/calibration if needed, as I know you have been doing a good bit of that?

The journal recommends staying within 10 pages for everything including references. It is currently at 6, so there is plenty of room for authors to contribute a few paragraphs on the theory behind their work and/or small use-case examples.

FormRF, 2021-06-29:
I am happy to help review the paper as well and provide comments.

Vinc0110, 2021-06-29:
> The journal recommends staying within 10 pages for everything including references. It is currently at 6, so there is plenty of room for authors to contribute a few paragraphs on the theory behind their work and/or small use-case examples.

In that case - if desired - I could write roughly 1 page about vector fitting. I'm thinking about a summary/compilation of the tutorial and the examples I wrote for the documentation on readthedocs.

vik-s, 2021-06-29:
> 
>  perhaps Vikram could check over and update the previously written sections on deembedding/calibration if needed, as I know you have been doing a good bit of that?
> 
Sounds good, I can definitely do that.

jhillairet, 2021-06-29:
I've updated the message above with a read-only link to the overleaf project to give you an idea of the present draft. If you want to modify directly the project, please ask @JAnderson419 for an editing link (but not here, preferably on Slack as there is no more private messages on GitHub)

ivonnics, 2021-06-29:
Happy to help reviewing it as well as with potential translations into Spanish and Portuguese...

jhillairet, 2021-06-30:
> 
> Happy to help reviewing it as well as with potential translations into Spanish and Portuguese...

That's a good idea, thanks!

arsenovic, 2021-06-30:
great to hear there is so much interest!  i look forward to getting it
published.
alex

jjacampion, 2021-07-07:
I think this is a great idea. I have cited the project in multiple papers (many of which were with @beuerle) as suggested, but it would be nice to be able to link to a permanent reference that provides casual readers with an overview of scikit-rf and its capabilities without scaring them off with GitHub/software development stuff.

The choice of journal seems like a good one, as the main user group is the RF/microwave community. Hopefully publishing in an established journal in that field can improve the reputation of the project. From my experience scikit-rf is still rather unknown in the RF community at large and there is still a high level of scepticism to open-source tools (most would rather stick with established commercial packages, even if it's a closed black-box...). Alternative journals might be [IEEE Trans. on Inst. and Meas.](https://ieeexplore.ieee.org/xpl/RecentIssue.jsp?punumber=19) or perhaps [Metrologia](https://iopscience.iop.org/journal/0026-1394).

However, I do wonder whether the paper would be accepted in such a journal in its current format, which is somewhat tutorial-esque (no disrespect intended!). The Journal of Microwaves (JOM) lists its screening criteria as:

> New and original articles describing significant work and/or ideas not found elsewhere in the literature;
Theoretical work should be backed up by experiments whenever possible, and if not possible, it should be made very clear in the paper how the theory can be applied, why it is important for the general community, and how it can be verified in an application;

Given that much of the current material in the paper is available via the docs, one could ask how much original material there is and in general how much of the content is scientific. I could see the paper being bounced back due to a lack of scientific content. Is there any way in which the style and content of the paper could be changed to beef it up and make it more academically rigorous? Perhaps by including more complex examples, with measurement data, processing workflows, setup images, etc.? Benchmarking/comparison with existing packages? 
One area where this is crucial is in statistical and uncertainty analysis - comparison with international guidelines such as [BIPM's GUM](https://www.bipm.org/en/committees/jc/jcgm/publications) would be good to prove that the built in methods are up to scratch. Understanding that currently requires that readers first look up skrf, then go to the scipy docs, find the correct method, make a testbench to compare to GUM, etc.. I know from personal experience that this is of concern to those working with metrology.

One very boring, but important question: how will the publishing fees be paid? Publishing a >6 page non-open paper in any other MTT-S journal will cost several hundred $. IEEE open access fees for their hybrid journals are [$2045](https://journals.ieeeauthorcenter.ieee.org/choose-a-publishing-agreement/open-access-rights-management/), while IEEE Access costs $1750. I guess JOM will cost the same. Perhaps this can be covered through one of the author's affiliations under an agreement with IEEE?

I haven't been an active developer of the project (I'm no dev,  @beuerle took care of anything we ever needed) but use it heavily. I'd be happy to help out in any way possible, potentially by providing measurement data, examples or simply reviewing a draft. 

/James

rakeshsinha30, 2021-07-07:
I think IEEE Microwave Magazine is the best option. Development of tools
and software module can be presented in IEEE Microwave Magazine.

jhillairet, 2021-07-07:
Thanks @jjacampion and @rakeshsinha30 for your comments. You are right, the current form is maybe (certainly?) not compatible with usual scientific paper. Maybe IEEE Microwave Magazine would be better suited for this (with the advantage to not require a lot of changes)... 

Anyway, if you want to have a look on the paper and eventually to contribute on it, you are welcome. Please come to scikit-rf Slack channel.

jjacampion, 2021-07-07:
Good idea @rakeshsinha30 - I forgot about Microwave Magazine. They should be open to a more tutorial style paper, which would avoid a major re-draft etc. If the goal of the paper is to provide a permanent reference and to help raise the profile of the project I guess an article there is totally sufficient.

Vinc0110, 2021-07-07:
@jjacampion, you have a good point there regarding the scientific level and the style of the current draft. I was actually thinking the same when I read it and I have already changed many paragraphs that contained very colloquial language.

The Microwave Magazine sounds about perfect for what we currently have. Check out [Guidelines for Authors](https://mtt.org/publications/ieee-microwave-magazine/guidelines-for-magazine-authors/):
- Articles should be of interest to a broad audience and understandable by non-experts in the field. Equations should be used sparingly. Qualitative descriptions or graphs are preferred tools for explaining relationships.
- The language of the article can be less formal than that desired for our journals
- Feature articles are typically about 8-10 pages in length, which is around 6,000 words.
- Microwave Magazine does not use a submission template. Authors are encouraged to prepare their manuscripts using a single column format with double spacing.

However, I did not find information about publication fees...

wbarnha, 2021-07-07:
I agree- the paper in its current state is more fit for IEEE Microwave Magazine. Also worth noting:
>The technical content of the magazine is about 55 pages per issue, consisting of 4-5 feature articles and 1-2 application notes
>Feature articles are typically about 8-10 pages in length
>Each paper in the focus section should be no more than 10 magazine pages in length
>Application notes tend to run two or three pages, but can run as high as five: this is about 3,000 words.

The current word count is 2820, so this could be interesting to see where things go as an application note. 

>The primary goal of an application note could be to describe how to accomplish a specific task, or to describe a particular piece of equipment or software. This should include the illustration of a technique, measurement, or product application that will be newsworthy and of benefit to our readers as microwave engineers. It should also try to show the problem being solved is general, that several alternative techniques (i.e., products) exist; the application note may be using one product as an illustration. Application notes tend to run two or three pages, but can run as high as five: this is about 3,000 words.

It does satisfy the criteria above. However, we have many figures and additional content that need to be removed. I'm not eager to remove some of these figures.

In its current state, it's not very academically rigorous because there's not a clear answer to "what's new" about this library. The library itself is new however its techniques aren't necessarily novel. If we want to stretch this to being academically rigorous, then we need to ask ourselves what methods in this library are new. We may also want to consider: IEEE Journal on Multiscale and Multiphysics Computational Techniques, IEEE Transactions on Microwave Theory and Techniques

jjacampion, 2021-07-07:
> In its current state, it's not very academically rigorous because there's not a clear answer to "what's new" about this library. The library itself is new however its techniques aren't necessarily [...]

@wbarnha I have no experience of the former, but the latter will very likely reject the paper unless significant scientific content is added. JOM has only been around for less than a year, but I expect their standards are as high as TMTT, if not higher. Perhaps targeting such a journal is unrealistic.

I would assume MTT-S treat Microwave Magazine (MM) as a hybrid publication, meaning a fee of $2045 will have to be paid to publish an open-access article. I can't any information on this on their webpage, nor any open-access articles in MM. It would be a shame to publish an article about open-source software in a closed journal, although [the accepted version can be published elsewhere freely](https://v2.sherpa.ac.uk/id/publication/23042?template=romeo), which would allow it to be posted to the docs page.

jhillairet, 2021-07-21:
Here is the current version as of 21/07/2021 : [skrf_IEEE_Microwaves.pdf](https://github.com/scikit-rf/scikit-rf/files/6854510/skrf_IEEE_Microwaves.pdf)

Many thanks to all contributors.

jhillairet, 2021-08-29:
for future reference, github now proposes [an enhanced support for citations](https://github.blog/2021-08-19-enhanced-support-citations-github/).See also [github doc.](https://docs.github.com/en/github/creating-cloning-and-archiving-repositories/creating-a-repository-on-github/about-citation-files#about-citation-files)

dxxx9, 2021-09-23:
Well done!

Just a little annotation if not already fixed. The links on page 14 containing `scikit-rf` don't work due to the unnecessary curly brackets.

[image]

wbarnha, 2021-09-24:
Nice catch. I went ahead and fixed text formatting issue by using a combination of `\href{url}{\UrlFont{\skrf{} and LaTeX commands here}}` to fix the issue. I also went ahead and fixed the built-in links.

jhillairet, 2021-09-24:
I've asked for some news about the paper and received the following response from the IEEE Microwave Magazine Editor in Chief:

> We have tentatively scheduled your manuscript for publication in the January 2022 issue of the magazine.  Files for this issue, including your manuscript, will be forwarded to IEEE by the end of next week for production of the magazine.  Depending on how IEEE handles the production, you may receive proofs from them in 4-6 weeks, but it could be later than that date.

Some it seems it is a good news! :)

arsenovic, 2021-09-24:
great!

jjacampion, 2021-09-24:
Nice! Seems the whole thing went smoothly. Sorry I never got around to contributing anything.

jhillairet, 2021-10-31:
For reference, here is the final version sent/accepted by IEEE Microwave Magazine. To appear in January 2022.

[skrf_IEEE_Microwaves_as_submitted.pdf](https://github.com/scikit-rf/scikit-rf/files/7448821/skrf_IEEE_Microwaves_as_submitted.pdf)

jhillairet, 2021-10-31:
Thank you everyone !

jhillairet, 2021-10-31:
@arsenovic, eventually we could create a new repo to copy the source files into it, don't you think?

closed 2021-10-31 by jhillairet

jhillairet, 2021-12-02:
The scikit-rf introcing paper has been released in IEEE Microwave Magazine ( Volume: 23, Issue: 1, Jan. 2022) (DOI: [10.1109/MMM.2021.3117139](https://doi.org/10.1109/MMM.2021.3117139).

In addition, the paper sources have been stored in a dedicated repository in scikit-rf project:
https://github.com/scikit-rf/An-Open-Source-Python-Package-for-Microwave-Network-Creation-Analysis-and-Calibration

ivonnics, 2021-12-03:
Congratulations Julien!
José

Sent with [ProtonMail](https://protonmail.com/) Secure Email.

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

jhillairet, 2021-12-04:
Thank you. The paper has been written by multiple hands, so it is a community-made paper. 

By the way, speaking about community, I encourage everyone to answer the scikit-rf 2021 survey if not already:
https://forms.gle/uonurVVb4FPhTLQm9

So far, there are only 17 responses, so very little. Please spread the word!