scikit-rf/scikit-rf #492: Unhandled corner case in `Frequency.f.setter`

mbastani, 2021-07-13:
The check for sweep type in `Frequency.f.setter` [here](https://github.com/scikit-rf/scikit-rf/blob/a1c7caec4f3219ac9f5050031b40b4608adf33ef/skrf/frequency.py#L428) (introduced in one of the recent versions of skrf) is unsafe: if `new_f` is an empty array  (resulting in an empty frequency band) it raises an `IndexError`.   

In my view, `sweep_type` does not even need to be an instance variable. It is useful for building a frequency band, but once the frequency band's points are determined it doesn't matter *how* they are obtained.  Strictly speaking, `sweep_type` is not a property of a frequency band.  As far as I can say, `sweep_type` is only used in `npoints.setter`, which (again), from my point of view, does not need to exist.  (Given a frequency band we need to know how many points it has, but it does not make sense to change its number of points.  If we want finer band, we can reconstruct a new one.)

So, a clean fix would simply be to remove `sweep_type` from the instance variables of the `Frequency` class completely (and hence eliminate the code for checking the sweep type in `f.setter`.

FranzForstmayr, 2021-07-14:
I think I introduced the issue you talk about a few weeks ago, however, why does one set an empty array as the frequency vector? 
Never mind, it's an issue, but how would you handle this corner case? Throw another exception would be the only valid thing I can think of. 

Regarding the `npoints.setter`, I see your point, there's almost never a reason to increase the number of points this way.
I think this is one general issue of `skrf`, as the number of frequency points and the frequency points of the s-matrix are not synchronized. If we had a single source of truth here, nobody would need to set the number of frequency points, the only way one could change the number of points is resampling/intra-/extrapolation of the s-param matrix.

This would break backward compatibility, as the API was designed to set s-params, frequency, z0-vector directly, without any checks.

mbastani, 2021-07-14:
I think if  `new_f` is empty, `self._f` can be empty, and I don't think that would cause any issues. (This was the behavior prior to your update.)  If we insist on storing/updating `sweep_type`, it could be set to `'unknown'` in case `new_f` is empty. 

I'll have to educate myself more about what kinds of backward compatibility would break if `npoints.setter` were removed.    I imagine the corner case of empty `new_f` in `Frequency.f.setter` can be addressed independently.

FranzForstmayr, 2021-07-14:
> I'll have to educate myself more about what kinds of backward compatibility would break if `npoints.setter` were removed

I don't have any usecase either, however the method `npoints.setter` exists since many years, and I cannot guarantee that nobody is using this method :)
For me personally we could drop this setter.

FranzForstmayr, 2021-07-14:
> I imagine the corner case of empty new_f in Frequency.f.setter can be addressed independently.
@mbastani, what's about the `start`, `stop` properties and friends, we would get an IndexError too. Why do we even allow empty frequency objects?

mbastani, 2022-01-04:
@FranzForstmayr  Sorry for getting back to this so late.  I see your points.  As for the `start` and `stop` of an empty frequency band, one could throw a `ValueError`.  (That would mean `start` and `stop` should not be instance variables, but properties, where the getter checks for these cases.)   My suggestion is inspired by how Python or (NumPy's) `max` and `min` operators behave on empty lists or arrays. 

As a side note, fundamentally, I believe `Frequency` should be a totally immutable object.  (If one wants to change the number of points, sweep type, etc. they can create a new `Frequency` object.)  If we could make such an update (which will certainly require updating a lot of legacy code), most of the questions discussed in this thread would be moot.

closed 2022-06-17 by jhillairet