scikit-rf/scikit-rf #493: Uncaught ModuleNotFoundError for optional dependency networkx in rf.Circuit.plot_graph()

JAnderson419, 2021-07-15:
Calling the plot_graph method of rf.Circuit without networkx installed gives an ImportError/ModuleNotFoundError rather than failing gracefully. See code below, based on an example from the documentation.

```python
def mBVD_model(freq:rf.Frequency, cs, ls, rs, cp, rp, rc):
    tl_media = rf.DefinedGammaZ0(freq, z0=50, gamma=1j * freq.w / rf.c)
    Cs = tl_media.capacitor(cs, name='Cs')
    Ls= tl_media.inductor(ls, name='Ls')
    Rs = tl_media.resistor(rs, name='Rs')
    Cp = tl_media.capacitor(cp, name='Cp')
    Rp = tl_media.resistor(rp, name='Rp')
    Rc = tl_media.resistor(rc, name='Rc')

    gnd = rf.Circuit.Ground(freq, name='gnd')
    port1 = rf.Circuit.Port(freq, name='port1', z0=50)

    cnx = [
        [(gnd, 0), (port1, 0), (port1, 1), (Rc, 1), (Rc, 2),
         (Cp, 2), (Cp, 3), (Rp, 3), (Rp, 0),
         (Cs, 2), (Cs, 4), (Ls, 4), (Ls, 5), (Rs, 5), (Rs, 0)],
    ]
    cir = rf.Circuit(cnx)
    cir.plot_graph(network_labels=True, network_fontsize=15,
                   port_labels=True, port_fontsize=15,
                   edge_labels=True, edge_fontsize=10)
    ntw = cir.network
    return ntw
```

```python
Traceback (most recent call last):
  File "C:\Users\ander906\PycharmProjects\scratch_work\venv\lib\site-packages\skrf\circuit.py", line 359, in graph
    import networkx as nx
  File "C:\Program Files\JetBrains\PyCharm 2021.1.3\plugins\python\helpers\pydev\_pydev_bundle\pydev_import_hook.py", line 21, in do_import
    module = self._system_import(name, *args, **kwargs)
ModuleNotFoundError: No module named 'networkx'
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "C:\Users\ander906\PycharmProjects\scratch_work\venv\lib\site-packages\IPython\core\interactiveshell.py", line 3441, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-7-a01675e7785b>", line 1, in <module>
    n2 = mBVD_model(n.frequency, 1E-15, 206E-9, 12.38, 100E-15, 31, 0.1)
  File "<ipython-input-4-2c16593186ca>", line 106, in mBVD_model
    edge_labels=True, edge_fontsize=10)
  File "C:\Users\ander906\PycharmProjects\scratch_work\venv\lib\site-packages\skrf\plotting.py", line 2072, in plot_circuit_graph
    G = self.G
  File "C:\Users\ander906\PycharmProjects\scratch_work\venv\lib\site-packages\skrf\circuit.py", line 345, in G
    return self.graph()
  File "C:\Users\ander906\PycharmProjects\scratch_work\venv\lib\site-packages\skrf\circuit.py", line 361, in graph
    raise ImportError('networkx package as not been installed and is required. ')
ImportError: networkx package as not been installed and is required. 
```

jhillairet, 2021-07-15:
How did you install scikit-rf ? `networkx` is in the pip and conda requirements, so it should have been installed.

JAnderson419, 2021-07-15:
I used pip. It shows me as being on version 0.18.1 for skrf.

jhillairet, 2021-07-15:
By the way, few comments:
``` python 
    cnx = [
        [(gnd, 0), (port1, 0), (port1, 1), (Rc, 1), (Rc, 2),
         (Cp, 2), (Cp, 3), (Rp, 3), (Rp, 0),
         (Cs, 2), (Cs, 4), (Ls, 4), (Ls, 5), (Rs, 5), (Rs, 0)],
```
port1 is a 1-port Network, so `(port1, 1)` should fail. 

Same for Rc, Cp, Rp Cs Ls which are 2-ports: so `(Cs, 2)` or `(Rs,5)` refer to ports #2 and #5 of Networks Cs and Rs, which do not exist.

jhillairet, 2021-07-15:
> I used pip. It shows me as being on version 0.18.1 for skrf.

interesting... no idea why you don't have networkx installed. Installing it should solve the import problem.

If you can picture the circuit you want to build I can  provide an example of the cnx array.

JAnderson419, 2021-07-15:
Thanks for the input. I just received errors for doing the connections wrong. :/ I think it should be something like this?

```python
    cnx = [
        [(Rc, 0),  (port1, 0)],
        [(Rc, 1), (Cs, 0), (Cp, 0)],
        [(Cs, 1), (Ls, 0)],
        [(Ls, 1), (Rs, 0)],
        [(Cp, 1), (Rp,0)],
        [(gnd, 0), (Rs, 1), (Rp, 1)],
    ]
```

[image]

I'll try a fresh install of skrf in a venv and see if I can reproduce the error.

jhillairet, 2021-07-15:
from your drawing it should be:

```python
    cnx = [
        [(Rc, 0),  (port1, 0)],
        [(Rc, 1), (Rs, 0), (Cp, 0)],
        [(Rs, 1), (Ls, 0)],
        [(Ls, 1), (Cs, 0)],
        [(Cp, 1), (Rp,0)],
        [(gnd, 0), (Cs, 1), (Rp, 1)],
    ]
```

JAnderson419, 2021-07-15:
It looks like this is due to out of date requirements in setup.py:

[image]

jhillairet, 2021-07-15:
> 
> It looks like this is due to out of date requirements in setup.py:
> 

ha... so the imports should also fail for many other stuffs ...

jhillairet, 2021-07-21:
closed with #494

PR #494 (2021-07-21): Update & Streamline requirements
This is to update the setup.py (closes #493) and also makes an effort to move some of the development/build dependencies out of requirements.txt and into optional settings in setup.py. Github actions may need updating to grab the optional dependencies from setup.py for tests to run.
closed 2021-07-21 by jhillairet