scikit-rf/scikit-rf #495: Mixed-mode S-Parameter Plotting

lumylovepandas, 2021-07-16:
i read through this example from this link:
https://scikit-rf.readthedocs.io/en/latest/examples/mixedmodeanalysis/Mixed%20Mode%20S%20and%20Impedance%20Transformation.html

I don't understand how to assign the port such mm_ntwk.renumber([0,1,2],[2,1,0]) for this line. For example, I have a s4p differential pairs,  is it the mm_ntwk.renumber([1,2],[3,4]) with given that my touch file has (1,2) of the left and (3,4) on the right.

Correct if i am wrong.

Vinc0110, 2021-07-16:
Unfortunately, that example does not explain things very well. 
For a 4-port network like you have it in a `s4p` file, you should better check out the other mixed-mode example:

https://scikit-rf.readthedocs.io/en/latest/examples/mixedmodeanalysis/Mixed%20Mode%20Basics.html

The 2nd code snippet might be what you need:
`
sedata = rf.Network('my4port.s4p')

sedata.renumber([0, 1, 2, 3], [0, 2, 1, 3])  # pair ports as 1,3 and 2,4 to match experimental setup (only if you need it)

sedata.se2gmm(p=2)  # two balanced ports
`

Also see the API documentation for the `se2gmm()` and `gmm2se()` functions:
https://scikit-rf.readthedocs.io/en/latest/api/generated/skrf.network.Network.se2gmm.html
https://scikit-rf.readthedocs.io/en/latest/api/generated/skrf.network.Network.gmm2se.html

jhillairet, 2021-07-21:
@lumylovepandas, does it answer your issue?

lumylovepandas, 2021-07-21:
Hi there, thanks for answering my doubt.
The way that I did in my code is like this with s4p:

> mm_ntwk.renumber([1,2],[1,2])  <---i don't understand this code
> mm_ntwk.se2gmm(p=2)

and I manage to plot the mixed-mode S-parameter.

My doubt in my mind:

1. How does the renumber works?
2. What is the actual order for this renumber API? I always thought s4p must associated with [1,2],[1,2]

Please help me to clear my doubt.

Vinc0110, 2021-07-21:
The API documentation is published on readthedocs: https://scikit-rf.readthedocs.io/en/latest/

There, you can search for the `renumber` method:
https://scikit-rf.readthedocs.io/en/latest/api/generated/skrf.network.Network.renumber.html

Again, the examples are not always great and the explanations are limited in this case, but you should get the idea. As you can see, you always specify two lists with `N` elements each for the `renumber` function, where `N` is the number of ports of your network. The first list is for the port number *before* the renumbering (how they are), the second list is how you want them to be renumbered. This is just about names for the ports; the first port in the s4p file does not need to be port 1.

If your 4-port network is already in mixed-mode, 2 ports are for differential-mode ports, the remaining 2 ports are common-mode ports. If your 4-port is in single-ended mode, the 4 ports are all regular single-ended ports. In both cases, you obviously need to know which ones are which in your s4p file. The function `se2gmm()` requires the differential- and common-mode ports to be in the right order, see the documentation linked in my first reply.

If you want/have to renumber the ports of your 4-port network, you have to specify 2 list with length 4. 
For example:
`mm_ntwk.renumber([1, 2, 3, 4], [1, 3, 2, 4])`
Here, the ports were originally ordered 1, 2, 3, 4; after the renumbering they will be ordered 1, 3, 2, 4.

lumylovepandas, 2021-07-22:
Hi Vinc0110, thank you for your explanation. I roughly get what you mean in the sentences.
The renumber API mean change the orientation of the ports in the snp. Normally, I organize the ports according my needs in ADS and I plot the mixed mode with Scikt-RF module.

One more thing, i verify it and confirm the mm_ntwk.renumber([1, 2, 3, 4], [1, 2, 3, 4].
        ______
1 -- |  snp  | -- 3
2 -- | ______| -- 4

Am I right on this?

lumylovepandas, 2021-07-22:
I get what you mean. Sometimes the connector vendor (etc. Amphenol) gives us the order is in odd (left) and even (right) orientation. We renumber the order as we want to series order.

Connector Vendor Port Orientation:
1 -- |       | -- 2
3 -- |       | -- 4

My Order using Renumber API:

1 -- |       | -- 3
2 -- |       | -- 4

Vinc0110, 2021-07-22:
@lumylovepandas Check out the discussion in issue #499. As it turned out, my explanations above are not fully correct.

lumylovepandas, 2021-07-23:
I see, then the renumber describe need to be improved in the Scikit-RF documentation.
I  suggest you start with standard [1,2,3,4] and OEM [1,3,2,4] for explanation.

Standard:
1 -- | | -- 3
2 -- | | -- 4

OEM:
1 -- | | -- 2
3 -- | | -- 4

Vinc0110, 2021-08-05:
@lumylovepandas: Is this still an active issue or can it be closed? @jhillairet has implemented the improvements discussed in #499, so things should be much clearer now. 

See updated API doc at https://scikit-rf.readthedocs.io/en/latest/api/generated/skrf.network.Network.renumber.html

lumylovepandas, 2021-08-05:
Hi Vinc0110, please close the issue.

closed 2021-08-05 by jhillairet