scikit-rf/scikit-rf #496: Moving wiki docs into main documentation ?

jhillairet, 2021-07-21:
Currently, the [main documentation](https://scikit-rf.readthedocs.io/en/latest/index.html) does not include some of the documentation which is on the [Github wiki](https://github.com/scikit-rf/scikit-rf/wiki), such like [the development guide](https://github.com/scikit-rf/scikit-rf/wiki/Development) of the [citation/acknowledgment discussion](https://github.com/scikit-rf/scikit-rf/wiki/Citations-Acknowledgments).

Two questions:

1) should we include the wiki documentation in the main documentation?
2) should we keep the wiki documentation?

for 1) I think so, but for 2) I don't know. Google doesn't reference the documentation in the wiki, and keeping the same documentation in two places is probably a bad idea. 

What do you think?

arsenovic, 2021-07-21:
i think we should consolidate docs, and get rid of the wiki.

jhillairet, 2021-07-24:
close after #500

closed 2021-07-24 by jhillairet