scikit-rf/scikit-rf #499: Suggestion to Improve Wording on skrf.network.Network.renumber

denzchoe, 2021-07-22:
I think the Port Renumbering documentation could use some improvement.
I had to deal with 8 Ports Mixed-Mode data (with Crosstalks), and I misconstrued the definition so largely it screwed me up quite a while. lol

According to documentation
https://github.com/scikit-rf/scikit-rf/blob/7a5dcd8c017b5eb0f4a2014ad6b11580528c4aab/skrf/network.py#L2729-L2730

the wording of **becomes** is widely misleading and widely open to misinterpretation.

#### Incorrect Interpretation
Should I interpret like the below
```
Port 0 becomes Port 1
Port 1 becomes Port 2
Port 2 becomes Port 0

```
Outcome is 
```
Sequence: 	| 0 | 1 | 2 |
Old Port Info:	| A | B | C |
New Port Info:	| B | C | A |
```

#### Correct Interpretation
Took me a while to realised that the proper definition is actually like this.
```
Old Port 0 rearranged to Port 1
Old Port 1 rearranged to Port 2
Old Port 2 rearranged to Port 0

```
Outcome is
```
Sequence: 	| 0 | 1 | 2 |
Old Port Info:	| A | B | C |
New Port Info:	| C | A | B |
```

<br>
Note that for both forms, they are both rotating, just at opposite directions.

jhillairet, 2021-07-22:
Would something like that more clear ?

```
Network            Original Network Ports
Ports                               
            ┌───────────────────┐
            │                   │
0   ────────┤  0                │
            │                   │
            │                   │
1   ────────┤  1                │
            │                   │
            │                   │
2   ────────┤  2                │
            │                   │
            └───────────────────┘
```
after `bar.renumber( [0,1,2], [1,2,0] ) `
```
Network            Original Network Ports
Ports                               
            ┌───────────────────┐
            │                   │
1   ────────┤  0                │
            │                   │
            │                   │
2   ────────┤  1                │
            │                   │
            │                   │
0   ────────┤  2                │
            │                   │
            └───────────────────┘
```

jhillairet, 2021-07-22:
another example could be, using different char impedance to follow the changes :

``` python
# dummy Network with different reference impedance
f = rf.Frequency(1, 1, 1)
s = np.ones((1, 3, 3))
z0 = [10, 20, 30]
ntw = rf.Network(frequency=f, s=s, z0=z0)
print(ntw)  # before 
ntw.renumber([0,1,2], [1, 2, 0])
print(ntw)  # after
```

gives 

```
3-Port Network: '',  1.0-1.0 GHz, 1 pts, z0=[10.+0.j 20.+0.j 30.+0.j]
3-Port Network: '',  1.0-1.0 GHz, 1 pts, z0=[30.+0.j 10.+0.j 20.+0.j]
```

denzchoe, 2021-07-22:
Hmmm.. The first example. Not quite there yet. My first initial read is that, Port 1,2,0 as the new sequence instead of Port 0,1,2.
If you read the above, you can also get confused because we only have 1 form of numbering. There is no spatial location reference.

Your new example does highlight the changes in spatial location.
I thought including Alphabet or Roman number Sequencing to signify the changes in spatial location would do the trick.
Perhaps your new example is better.

denzchoe, 2021-07-22:
Or maybe a combination of both?
```
Old Network    Original Network Ports
Sequence                               
            ┌───────────────────┐
            │                   │
0   ────────┤  A  (10 Ω)        │
            │                   │
            │                   │
1   ────────┤  B  (20 Ω)        │
            │                   │
            │                   │
2   ────────┤  C  (30 Ω)        │
            │                   │
```

>>z0=[10.+0.j 20.+0.j 30.+0.j]
			
`after bar.renumber( [0,1,2], [1,2,0] )`
			
```
New Network    Original Network Ports
Sequence                               
            ┌───────────────────┐
            │                   │
1   ────────┤  A  (10 Ω)        │
            │                   │
            │                   │
2   ────────┤  B  (20 Ω)        │
            │                   │
            │                   │
0   ────────┤  C  (30 Ω)        │
            │                   │
```
>>z0=[30.+0.j 10.+0.j 20.+0.j]

Vinc0110, 2021-07-22:
For reference: this is related to the recent discussion in issue #495. I just realized that I have misinterpreted the function arguments as well, so I can only agree the documentation should be more clear. 

If the first list argument of `renumber()` is actually a list of **indices** rather than the old port locations, it should be stated much more clearly. Currently, the argument `from_ports` is pretty useless.

jhillairet, 2021-07-27:
So, how about the following docstring @lumylovepandas @Vinc0110 @denzchoe  ?
             
```python

    def renumber(self, from_ports: Sequence[int], to_ports: Sequence[int]) -> None:
        '''
        Renumbers ports of a Network (inplace).

        Parameters
        -----------
        from_ports : list-like
            list of port indices to change

        to_ports : list-like
            list of desired port indices 

        NB: from_ports and to_ports must have same size.

        Return
        ------
            None
            The reorganization of the Network's port is performed inplace.
        

        Examples
        --------
        
        In the following example, the ports of a 3-ports Network are 
        reorganized. Dummy reference impedances are set only to follow more
        easily the renumbering.
        
        >>> f = rf.Frequency(1, 1, 1)
        >>> s = np.ones((1, 3, 3))
        >>> z0 = [10, 20, 30]
        >>> ntw = rf.Network(frequency=f, s=s, z0=z0)  # our OEM Network

                    ┌───────────────────┐
                    │       OEM         │
                    │                   │
        0   ────────┤  A  (10 Ω)        │
                    │                   │
                    │                   │
        1   ────────┤  B  (20 Ω)        │
                    │                   │
                    │                   │
        2   ────────┤  C  (30 Ω)        │
                    │                   │
                    └───────────────────┘

        >>> ntw.renumber([0,1,2], [1, 2, 0])

                    ┌───────────────────┐
                    │       OEM         │
                    │                   │
        1   ────────┤  A  (10 Ω)        │
                    │                   │
                    │                   │
        2   ────────┤  B  (20 Ω)        │
                    │                   │
                    │                   │
        0   ────────┤  C  (30 Ω)        │
                    │                   │
                    └───────────────────┘

        Other examples:

        To flip the ports of a 2-port network 'foo':
        >>> foo.renumber( [0,1], [1,0] )

        To rotate the ports of a 3-port network 'bar' so that port 0 becomes port 1:
        >>> bar.renumber( [0,1,2], [1,2,0] )

        To swap the first and last ports of a N-port (N>=2) Network 'duck':
        >>> duck.renumber( [0,-1], [-1,0] )
        '''

```

Vinc0110, 2021-07-27:
With these examples in the docstring, I think it is well explained and should be clear for most users. 

The last example is very important, because it illustrates the requirement to specifiy *two* lists for the old and new port order:  when you only need to (ex)change the order of some ports in a large network, but don't want to provide the full list of all ports. Otherwise, one could question the purpose of the first argument, which would always be a sorted list of all ports in the network. To make things even more clear, we could specify the length of the lists in the parameter descriptions:
```

Parameters
        -----------
        from_ports : list-like
            List of port indices to change. Size between 1 and N_ports.

        to_ports : list-like
            List of desired port indices. Size between 1 and N_ports.

        NB: from_ports and to_ports must have same size.
```

Another minor thing to add:
@denzchoe had put labels above the old and new drawings. That is useful, so users can distinguish the original port order in the file and the old/new order in skrf. Here is my relabled version of the "header":

```
Order in              Original Order
skrf.Network      
                    ┌───────────────────┐
                    │       OEM         │
                    │                   │
        1   ────────┤  A  (10 Ω)        │
                    │                   │
                    │                   │
        2   ────────┤  B  (20 Ω)        │
                    │                   │
                    │                   │
        0   ────────┤  C  (30 Ω)        │
                    │                   │
                    └───────────────────┘                        
```

denzchoe, 2021-07-27:
> With these examples in the docstring, I think it is well explained and should be clear for most users.
>
> The last example is very important, because it illustrates the requirement to specifiy _two_ [...]

denzchoe, 2021-07-27:
Oops, accidentally pressed the wrong button there earlier (closing the issue). Tiny phone screens. :) 

I agree on the comments made by @Vinc0110. Such as this 

```
            List of port indices to change. Size between 1 and N_ports.
```
And his updated header info at Illustration.

jhillairet, 2021-07-27:
PR #504 in progress

denzchoe, 2021-07-28:
out of curiosity, what does OEM stands for here and in #495 ? 
Original Equipment Measured?

Vinc0110, 2021-07-28:
> Original Equipment Measured?

Almost right. It usually stands for [Original Equipment Manufacturer](https://en.wiktionary.org/wiki/OEM).

closed 2021-08-04 by jhillairet