scikit-rf/scikit-rf #503: Possible Future Documentation Topics

jhillairet, 2021-07-27:
I'm listing here a list of topics that could fit into the documentation. This list is a never-ending list, things I have in mind, and will be updated regularly and enriched with any proposals:

 - [ ] Measuring/Determining an impedance using a VNA. 
 Ressources : [Three Methods to Measure Impedance with the NanoVNA](https://www.youtube.com/watch?v=1UbEz73FGCU&t=333s),  [Agilent note p.4](https://studylib.net/doc/18887795/impedance-measurements)
- [ ] Introduction to Mixed-Mode S-Parameters.
Ressources: [Mixed-Mode S-Parameters](https://www.microwaves101.com/encyclopedias/mixed-mode-S-parameters), 
- [ ] Make it more clear between characteristic and reference impedances of Network and Media. 
Ressources: [Matching Circuits Reference Impedance?](https://www.researchgate.net/post/Matching-Circuits-Reference-Impedance)
- [ ] Amateur Radio & SDR oriented stuffs. 
Ressources: [W0QE Videos](https://www.youtube.com/channel/UCKSyLSu4fm_1RHoO3Jvk4YQ),  [https://www.rtl-sdr.com/](https://www.rtl-sdr.com/), [Baluns](https://www.microwaves101.com/encyclopedias/baluns)
- [ ] Various ways to create RF components
Ressources: [Top 7 Ways to Create a Quadrature (90˚) Phase Shift](https://www.markimicrowave.com/blog/top-7-ways-to-create-a-quadrature-90-phase-shift/), [6 Ways to Make an N-Way Power Splitter](https://www.markimicrowave.com/blog/how-to-make-an-n-way-power-splitter/)

jhillairet, 2021-09-06:
@vik-s it would be great if you could make a notebook example and write some tests about the de-embbeding class #480

vik-s, 2021-09-07:
> 
> 
> @vik-s it would be great if you could make a notebook example and write some tests about the de-embbeding class #480

Thanks for the followup @jhillairet! I have the notebook example written up more than half-way, I need to finish it up. It's going up on my to-do list.