scikit-rf/scikit-rf #506: Type hints in API docs

Vinc0110, 2021-08-02:
Now that scikit-rf starts to have type hints included in many classes, I'd like to propose a change in the `autodoc` configuration for the generation of the API documentation with Sphinx.

In summary, I think we should set `autodoc_typehints = "description"`.

Details:

Currently, the displayed signatures of class functions include the type hints, which provides redundant information and degrades readability when many parameters are listed. For example, here is a screenshot of the current `VectorFitting.vector_fit()` [documentation](https://scikit-rf.readthedocs.io/en/latest/api/vectorFitting.html#skrf.vectorFitting.VectorFitting.vector_fit) with a rather messy signature at the top:

[image]

This behavior can be changed in the Sphinx configuration, for example by setting `autodoc_typehints = "description"`. See www.sphinx-doc.org/en/master/usage/extensions/autodoc.html#confval-autodoc_typehints

For comparison, here is a screenshot of a [NumPy function doc](https://numpy.org/doc/stable/reference/generated/numpy.linalg.svd.html), which I think looks much nicer:

[image]

FranzForstmayr, 2021-08-02:
In addition to that, the type in the docstring is redundant, it's enough to annotate the parameter in the signature.
I already thought of that when adding the first type hints in `network.py`, but the documentation style seemed to be incomplient with the `autodoc-typehints` scheme. However it seems that this should be possible [https://github.com/agronholm/sphinx-autodoc-typehints#compatibility-with-sphinxextnapoleon](typehints with napoleon)

Vinc0110, 2021-08-05:
You are right, but I'm not sure if it's worth the extra effort of installing and configuring the extension and updating all the docstrings (do they have to be updated?). After all, the function signatures have already been updated and the rendered results on RTD would be exactly the same as they currently are.

jhillairet, 2021-08-05:
Looking to the issues in autodoc-typehints, [in particular this one](https://github.com/agronholm/sphinx-autodoc-typehints/issues/167), it seems it is redundant to install autodoc-typehints and to define:
`autodoc_typehints = "description"` 
in the sphinx config. 

I did both locally (with autodoc-typehints and without the "description" parameter and vice&versa) and I didn't find a difference in the rendering.

lenadevids, 2021-08-08:
is there any algorithm or program which detect when [Microwave Recycling](https://greencitizen.com/microwave-recycling-how-to-dispose-of-your-microwave-oven/) is process should be done.

FranzForstmayr, 2021-08-08:
@lenadevids 
I'm afraid to say this project is about radio frequency engineering and not about microwave ovens.

digitalblocs, 2021-08-24:
We are a [Houston seo service](https://digitalblocs.com/seo-for-small-businesses/) website design enhancement administration based Digital Blocs Company that works with organizations, all things considered, from one side of the country to the other and in an assortment of businesses to assess the opposition..

suorinvape, 2021-08-24:
You are right, but I'm not sure if it's worth the extra effort of installing and [configuring](https://suorinvape.com/collections/salt-e-liquid) the extension and updating all the docstrings (do they have to be updated?).

Vinc0110, 2021-08-24:
I'm closing this issue to stop the spam. We are probably done here anyways.

closed 2021-08-24 by Vinc0110

giftforhim, 2021-08-25:
[https://www.hentley.co/](url) must visit to having good and 100 percent quality handmade leather wallets .

suorinairpluspod, 2021-08-30:
is there any algorithm or program which detect when <a href="https://suorinvape.com/collections/suorin-air-plus">Suorin Air Plus Pod</a> is process should be done. if not visit our site