scikit-rf/scikit-rf #514: to get  a currents result inter the circuit

Rudy-Han, 2021-08-29:
conn = [[(port1,0), (nt1,0)], [(nt1, 1), (nt2, 0)], [(nt2, 1), (port2, 0)]]

power = [0, 0.5]
phase = np.deg2rad([0, 0])

currents = cir.currents(power, phase)

I want to get the currents on x1(  [(nt1, 1), (nt2, 0)]  )  point, but I get a error which msg is "numpy.linalg.LinAlgError: Singular matrix"

jhillairet, 2021-08-29:
I made the following example which works:

```python
import skrf as rf
print(rf.__version__)  # -> 0.19.0

ntwk1 = rf.data.ring_slot
ntwk1.name = 'ntwk1'
ntwk2 = rf.data.ring_slot
ntwk2.name = 'ntwk2'

port1 = rf.Circuit.Port(ntwk1.frequency, 'port1')
port2 = rf.Circuit.Port(ntwk1.frequency, 'port2')

cnx = [
       [(port1, 0), (ntwk1, 0)],
       [(ntwk1, 1), (ntwk2, 0)],
       [(ntwk2, 1), (port2, 0)],
       ]

cir = rf.Circuit(cnx)

powers = [0, 0.5]
phases = [0, 0]

currents = cir.currents(powers, phases)
print(currents.shape)  # ->(201, 6)
```
Are you using the latest version of scikit-rf ? If yes, can you share the two .s2p files you are using?

Rudy-Han, 2021-08-29:
I replace the ntwk1 and ntwk2 by my test.s2p. The error happens again.

this is the test.s2p :

! TOUCHSTONE file generated by CST MICROWAVE STUDIO
! Date and time: Thu Jul  8 22:28:20 2021
! Project name: test.cst
! Header version: 2018
! Port assignment regex: !\s*Touchstone port\s*([0-9]+)\s*=\s*(CST MWS port|CST DS port)\s*([0-9]+|Zmin|Zmax)\s*\(\"(.*)\"\)(\s*mode\s*([0-9]+))?.*
! Touchstone port assignment:
! Touchstone port 1 = CST MWS port 1 ("")
! Touchstone port 2 = CST MWS port 2 ("")
# GHz S MA R 50
0                              1              -0      0.000375138              -0       0.00051308             180                1              -0
0.004                   0.999992       -0.216906       0.00131214         73.2067       0.00143799         110.539         0.999987       -0.243101
0.008                   0.999968       -0.433742       0.00254271         81.1545       0.00273459         100.084         0.999948       -0.486091
0.012                   0.999928       -0.650439       0.00379131         83.7794       0.00405996         96.1676         0.999884       -0.728861
0.016                   0.999872        -0.86693       0.00504495         85.0129       0.00539134          94.008         0.999795       -0.971303
0.02                    0.999802        -1.08315       0.00630105         85.6837       0.00672339         92.5667         0.999682        -1.21331
0.024                   0.999717        -1.29904       0.00755886         86.0719       0.00805396         91.4896         0.999545        -1.45479
0.028                   0.999619        -1.51454       0.00881814         86.2982       0.00938182          90.624         0.999387        -1.69565
0.032                   0.999508         -1.7296        0.0100789         86.4233        0.0107061         89.8934         0.999208         -1.9358
0.036                   0.999386        -1.94417        0.0113412         86.4809        0.0120263         89.2555          0.99901        -2.17516
0.04                    0.999254        -2.15822        0.0126051         86.4912        0.0133418          88.685         0.998795        -2.41367
0.044                   0.999112        -2.37171        0.0138709         86.4671        0.0146523         88.1661         0.998565        -2.65127
0.048                   0.998963        -2.58461        0.0151387         86.4171        0.0159575         87.6883          0.99832         -2.8879
0.052                   0.998807        -2.79693        0.0164087         86.3471        0.0172572         87.2445         0.998065        -3.12353
0.056                   0.998647        -3.00863        0.0176811         86.2613        0.0185513         86.8298           0.9978        -3.35814
0.06                    0.998483        -3.21973        0.0189562         86.1628        0.0198399         86.4404         0.997529        -3.59171
0.064                   0.998317        -3.43024        0.0202342         86.0537        0.0211229         86.0738         0.997252        -3.82424
0.068                    0.99815        -3.64017        0.0215154         85.9356        0.0224006         85.7278         0.996974        -4.05576
0.072                   0.997985        -3.84956        0.0227999         85.8099        0.0236732         85.4009         0.996696        -4.28627
0.076                   0.997822        -4.05843        0.0240882         85.6775         0.024941         85.0917          0.99642        -4.51584
0.08                    0.997663        -4.26685        0.0253803         85.5391        0.0262046         84.7992         0.996149        -4.74451
0.084                   0.997509        -4.47485        0.0266767         85.3952        0.0274643         84.5223         0.995884        -4.97234
0.088                   0.997362        -4.68251        0.0279774         85.2461        0.0287208         84.2602         0.995629        -5.19942
0.092                   0.997223        -4.88989        0.0292828         85.0922        0.0299748         84.0121         0.995386        -5.42584
0.096                   0.997092        -5.09707        0.0305932         84.9336        0.0312269          83.777         0.995155        -5.65171
0.1                     0.996971        -5.30415        0.0319087         84.7705        0.0324779         83.5543         0.994939        -5.87714
0.104                   0.996861         -5.5112        0.0332295          84.603        0.0337287          83.343          0.99474        -6.10226
0.108                   0.996762        -5.71832        0.0345558          84.431        0.0349801         83.1424         0.994559         -6.3272
0.112                   0.996675        -5.92561        0.0358878         84.2545         0.036233         82.9516         0.994397         -6.5521
0.116                     0.9966        -6.13318        0.0372257         84.0735        0.0374883         82.7696         0.994255        -6.77713
0.12                    0.996538        -6.34113        0.0385695         83.8881         0.038747         82.5956         0.994135        -7.00244
0.124                   0.996489        -6.54957        0.0399194          83.698          0.04001         82.4286         0.994036        -7.22818
0.128                   0.996452         -6.7586        0.0412754         83.5032        0.0412782         82.2676          0.99396        -7.45454
[... cut ...]
3.988                   0.608483        -37.9296        0.0245882         126.914        0.0318321         124.398         0.362038        -58.0781
3.992                   0.608148        -38.1313         0.024023         124.961        0.0312123         121.487          0.35845        -58.0207
3.996                   0.607827        -38.3368        0.0235074         122.878        0.0306211         118.449         0.354888        -57.9504
4                       0.607518        -38.5464        0.0230424          120.66        0.0300624         115.286         0.351355        -57.8673

jhillairet, 2021-08-29:
OK, the reason is that you have a 0 frequency point 

> 0                              1              -0      0.000375138              -0       0.00051308             180                1              -0
> 0.004                   0.999992       -0.216906       0.00131214         73.2067       0.00143799         110.539         0.999987       -0.243101
> ...

if you remove this 0 Hz point, it works.

Rudy-Han, 2021-08-29:
OK ,thanks a lot.

closed 2021-08-29 by Rudy-Han