scikit-rf/scikit-rf #517: Handling of duplicate frequency values

FranzForstmayr, 2021-09-14:
Is there a recommended way how to handle S-Parameter measurements with duplicate frequency entries?
I think these measurements are taken by accident most of the time, when  specifying a segemented sweep or something in VNAs, but maybe there's another possibility.

An example for this behaviour is in the reposity:
[ntwk_noise.s2p](https://github.com/scikit-rf/scikit-rf/blob/master/skrf/tests/ntwk_noise.s2p)

With the duplicate x-value you'll get strange issues, for example scipy's interpolation functions require the x values to be unique. 

Should we create a method, to drop duplicate values? Or is there another recommended way?

jhillairet, 2021-09-14:
No, there is no recommended way to handle duplicate frequency entries. 

And indeed, a lot of Numpy functions would not like that....

A workaround could be to convert the s parameters to pandas, drop duplicates with pandas `drop_duplicates` and then recreate a new Network from this cleaned data. 

Eventually, why not a  `drop_duplicates` or, more explicit, a `drop_frequency_duplicates` method in Network?

FranzForstmayr, 2021-09-14:
> Eventually, why not a drop_duplicates or, more explicit, a drop_frequency_duplicates method in Network?

Yeah, I would prefer something like this too. Should we throw a Warning when creating a Network with duplicates to let the user decide if the duplicates should be dropped?

jhillairet, 2021-09-14:
yes, eventually, and promoting the drop_frequency_duplicates at the same time. 

I wonder if having duplicate frequency values could be a "normal" case for some users..

FranzForstmayr, 2021-09-14:
Touchstone specification states:
> All data sets must be arranged in increasing order of frequency.

So it's probably not even a legal touchstone file with two equal frequency points. 

Would you suggest to drop the duplicates silently or with Warning automatically in this case?
I would go for the warning and suggest to drop method.

jhillairet, 2021-09-14:
No, I'm not in favour to drop it silently. The warning message should just state that there is some frequency duplicate point, and if this is unexpected, one can use the xxxx method to drop them.

FranzForstmayr, 2021-09-22:
fixed with #523

closed 2021-09-22 by FranzForstmayr