scikit-rf/scikit-rf #521: Implementation of IEEE P370 De-embedding Method into De-embedding Class in Scikit-RF?

denzchoe, 2021-09-16:
As indicated at https://groups.google.com/g/scikit-rf/c/vwVKe8JaCtU

Hi all,

I have been observing this standard for quite some time from its early days. 
https://www.signalintegrityjournal.com/articles/1169-ieee-p370-a-ffixture-ddesign-and-ddata-qquality-mmetric-sstandard-for-iinterconnects-up-to-50-ghz?page=2

And only recently that it has been released as a Standard ( Published Date:2021-01-08  )
https://standards.ieee.org/standard/370-2020.html

There is an IEEE Committee of SI Engineers working in parallel like you guys over here to push their own version of the characterization & de-embedding Class for SI Community to implement using MATLAB. 

For those who doesn't own MATLAB, you can excess the .exe program over here (although it may not be the latest version):
https://www.amphenol-icc.com/software?__cf_chl_jschl_tk__=pmd_q5VhubKX2D37YIlaKn7dUKIgiBuzm0PVl3D4cWYdutA-1631516714-0-gqNtZGzNAhCjcnBszQ29

[image]

Fortunately the codes are open sourced and constantly updated.
https://opensource.ieee.org/elec-char/ieee-370/

Looking that now we have a De-Embedding Class inside scikit-rf (thanks to [Vikram](https://github.com/vik-s)); perhaps we can now consider adding the IEEE P370 de-embedding method into scikit-rf?
https://opensource.ieee.org/elec-char/ieee-370/-/tree/master/TG1
I have not tested it extensively enough yet (so far so good on just 2 samples until 70 GHz, no real world measurement yet) but looking at how some of the big Connectors, measurement & EDA companies are already member of this group, is hard to ignore its relevance.
https://ieeexplore.ieee.org/stamp/stamp.jsp?arnumber=9241568

We should also probably invite the Gitlab Code Owners over here to have a look at scikit-rf. I have had the opportunity to bump into Jason Ellison at Y2019 in a conference and spoke to him a little bit of scikit-rf, as a conference attendee after his presentation then. But.. I doubt my words carried any weight. However, some of you fellas here definitely have the Credentials to do so. 

[image]

mhuser, 2021-09-20:
Hello @denzchoe,
Looks very interesting !
I played around the .exe and the [time domain reflectometry example](https://scikit-rf.readthedocs.io/en/latest/examples/networktheory/Time%20domain%20reflectometry%2C%20measurement%20vs%20simulation.html) by simulating the stepped line alone and de-embedding the steped-line mesurement with the 100 microstripline measurement as 2x thru.
Results looks quite goods (without deep analysis).
The process also output the left and right fixtures as separated .s2p.

[image]

[image]

jhillairet, 2021-12-13:
Additional document: https://harrisburg.psu.edu/files/pdf/27581/2021/04/20/open-source-de-embedding-j-ellison.pdf

Note the "Open-Source De-embedding" and few slides later "How much does it cost?" ;)

denzchoe, 2021-12-14:
Are we allowed to post Standards Documentation here? I have the document of https://ieeexplore.ieee.org/document/9316329
but it is a paid content (searched from the web), and I am not sure whether I can post it here. 

> Additional document: https://harrisburg.psu.edu/files/pdf/27581/2021/04/20/open-source-de-embedding-j-ellison.pdf
> 
> Note the "Open-Source De-embedding" and few slides later "How much does it cost?" ;)

jhillairet, 2021-12-14:
Yes it's OK.  Please email me if you need the file.

mhuser, 2021-12-14:
Hello @jhillairet  and @denzchoe 
I just looked into the gitlab source and it seems possible with reasonnable efforts to implement the functionnality on top of python and scikit-rf fundations !
This could bring open-source de-embedding at a fraction of the cited price ;-)

denzchoe, 2021-12-14:
@jhillairet , Thanks for the feedback.
So here it is. 
</br>
[IEEE Std P370-2020.pdf](https://github.com/scikit-rf/scikit-rf/files/7710450/IEEE.Std.P370-2020.pdf)

FranzForstmayr, 2021-12-14:
@denzchoe,

I'm not sure if you are allowed to share this document online. It's tagged to (a specific/your?) university and is meant to be used only within this university.

denzchoe, 2021-12-14:
@FranzForstmayr , yeah I figured it as much as well.. To be safe, I have removed the content.

mhuser, 2021-12-14:
Now that we have dived in, I am interested to give a try converting [IEEEP3702xThru.m](https://gitlab.com/IEEE-SA/ElecChar/P370/-/blob/master/TG1/IEEEP3702xThru.m) into a monolitic scikit-rf based python script this night and "see what it happens".
In the favorable case we manage to get the same results as .exe software and s-parameters examples on the gitlab repo, we could later see how we could cleanly insert this into scikit-rf.

mhuser, 2021-12-14:
First results (mode without Z-Correction) look not bad at all but there is still some deviation:
Need to be double checked.
You can find my python file here : [IEEEP3702xThru_pof.py](https://github.com/mhuser/scikit-rf/blob/master/examples/IEEEP370_deembedding_proof_of_concept/IEEEP3702xThru_pof.py)
Anyone interested testing it with its own data for comparison with the IEEE370 .exe software ?
[image]
[image]
[image]

denzchoe, 2021-12-15:
@mhuser ,

Thanks for putting the time to experiment with it. Really appreciate it.
By the way, the official branch of the code is no longer in GitLab, I think.
It is **now located here.** [IEEE_SA](https://opensource.ieee.org/elec-char/ieee-370/-/tree/master/)
There is a recent update to Folder TG1 just 2 weeks ago.

Here are my results from your code.
Is it possible that the deviation exist because the IEEE370.exe software is not updated (January 2019) as its parent in [IEEE_SA](https://opensource.ieee.org/elec-char/ieee-370/-/tree/master/)?

[image]

[image]

mhuser, 2021-12-15:
Hello @denzchoe ,
Thank you a lot for trying the code and sharing your results and comments !
I will look at the [new repo IEEE_SA](https://opensource.ieee.org/elec-char/ieee-370/-/tree/master/) to see what is in the commits ahaead gitlab. Thanks you again for pointing it out.

It is correct that it will be difficult at some point to compare with the .exe if we cannot assess if it is the same code version. Maybe they give a file set with s-parameters input and expected output for us to check our residuals ?

I took some liberties in implementation (e.g. there is no spline kind for interp1d in scipy so I used rather cubic). Also I noticed also the IEEEP370Zc2xThru.m seems much  developped so I wonder if the .exe use this function (without the Z correction) or if it really relies on the IEEEP3702xThru.m I tried to translate into python.

Best regards,

mhuser, 2021-12-15:
Maybe it would be usefull to define a test case with a determined dataset and a residual magnitude limit so we cann assess at what point we are "reasonnably close" of IEEEP370 compliance.

denzchoe, 2021-12-16:
More examples to highlight the difference in the .exe and .py version. 
This is a recent measurement. Attaching measurement file as well. 
attaching .S2P file for reference. {convert txt to s2p} [cal.txt](https://github.com/scikit-rf/scikit-rf/files/7724453/cal.txt)

[image]

[image]

[image]

mhuser, 2021-12-16:
Looks interesting ! Thank you a lot for sharing. Again the python curve (red) is closer to the measurement (blue) than the .exe curve (green), and the difference is bigger this time.
I think we could deembedd the fixtures from the 2xthru to plot the residuals out of the box with skrf.
Then we could use the criterion you mentionned for a test case "_observe the residual return loss. If it is below -20 dB across the entire spectrum, the deembedding can be trusted_"
However, I might not be able to work on this before weekend.

mhuser, 2021-12-16:
Just noticed there is a set of s-parameters attached with the [standard](https://standards.ieee.org/standard/370-2020.html), we can probably use these for to build the test case :-)
[image]

jhillairet, 2021-12-16:
> Just noticed there is a set of s-parameters attached with the [standard](https://standards.ieee.org/standard/370-2020.html), we can probably use these for to build the test case :-) 

Yes, only if these data are compatible with the open-source licence we use. To be check.

mhuser, 2021-12-16:
Indeed. The license seems to be [BSD 3-Clause](https://opensource.org/licenses/BSD-3-Clause). But we can also use them only as a check and implement another test-case based on skrf transmission line chunks for insertion in repo.

mhuser, 2021-12-16:
I exerciced [IEEEP3702xThru_poc.py](https://github.com/mhuser/scikit-rf/blob/master/examples/IEEEP370_deembedding_proof_of_concept/IEEEP3702xThru_poc.py) on one of the dataset supplier with [the standard](https://standards.ieee.org/standard/370-2020.html). Script was not updated yet to last version IEEE repo. Dataset is in P370_TG2_generator_prj.
- 2 x thru : `case_01_2xThru.s2p`
- F-DUT-F : `case_01_F-DUT1-F.s2p`
- reference DUT : `case_01_DUT1.s2p`
- .exe was set with bandwidth 50 GHz to generate reference fixture and deembedded dut
[image]
cascaded fixtures models show excellent agreement with 2xthru s11 and s21 (kind of logical as they are built from it and we have no reference to compare to the middle point)
[image]
Agreement between dut and deembedded dut is not very good even if residual return loss is under -20 dB up to 40 GHz.

mhuser, 2021-12-19:
I have split the IEEEP3702xThru in a separated file [Ieeep370_2xthru](https://github.com/mhuser/scikit-rf/blob/master/examples/IEEEP370_deembedding_proof_of_concept/ieeep370_2xthru.py) that extend the `skrf.calibration.Deembedding` class.
It can be teste by pasting this file aside your scripts and call it like :
```python
import skrf as rf
from ieeep370_2xthru import Ieeep370_2xthru
s2xthru = rf.Network('P1-MSL_Thru_100-P2.s2p')
fdf     = rf.Network('P1-MSL_Stepped_140-P2.s2p')
dm = Ieeep370_2xthru(dummy_2xthru = s2xthru, name = '2xthru')
d_dut = dm.deembed(fdf)
d_dut.name = 'd_dut'
# optionnaly get the fixtures
s_side1 = dm.s_side1
s_side1.name = 's_side1'
s_side2 = dm.s_side2
s_side2.name = 's_side2'
```
As for now, the code is a bit different from skrf style and supporting functions are embedded inside the class, but this help to compare with the original IEEEP3702xThru.m source.
I made two example of use, [one using my stepped impedance line already on skrf repo](https://github.com/mhuser/scikit-rf/blob/master/examples/IEEEP370_deembedding_proof_of_concept/test_ieeep370.py) and the [other](https://github.com/mhuser/scikit-rf/blob/master/examples/IEEEP370_deembedding_proof_of_concept/test_ieeep370_downloaded_sparameters.py) based on the [IEEEP370 ressources](https://standards.ieee.org/standard/370-2020.html) (but without including them so you will have to download them if willing to give a try).

mhuser, 2021-12-19:
Some results for the stepped impedance line, agreement between skrf and IEEE deembedding is very good. The d_dut also have reasonnable agreement with MLine simulated dut.
[image]
For the IEEE s-parameters DUT2 (probably a Beatty standard with a portion of 3xwidth microstripline) the agreement between given dut and deembedded dut is also close. (my .exe is broken so I was not able to compare with ieee deembedding)
[image]
For the IEEE s-parameters DUT1 (probably a portion of 1xwidth microstripline) the results of 2xthru method show very poor agreement (both for skrf and ieee) with the given dut. Here the IEEE z-corrected method give a better agreement in time Z but create an oscillation in s-parameters. Maybe it is not a good example as the discontinuities at lauch seems big and the return loss of the microstripline should be small ?
[image]

denzchoe, 2021-12-23:
@mhuser , this new edition of yours follows the latest from GitLab?

> I have split the IEEEP3702xThru in a separated file [Ieeep370_2xthru](https://github.com/mhuser/scikit-rf/blob/master/examples/IEEEP370_deembedding_proof_of_concept/ieeep370_2xthru.py) that extend [...]

mhuser, 2021-12-23:
@denzchoe 
> @mhuser , this new edition of yours follows the latest from GitLab?

Yes, it is believed to by consistent with [the latest version from ieee.org](https://opensource.ieee.org/elec-char/ieee-370/-/blob/master/TG1/IEEEP3702xThru.m) which show almost no differences with the old Gitlab version.
Of course double checkings and feedbacks are most welcomes :-)
#### Minor differences of skrf against ieee are
- [ ] cubic interpolation used instead of spline because of interp1d limitations in python, but doesn't seems to change results very much
- [ ] not implemented case were DC point already in s-parameters
- [ ] not implemented case were frequency is not an harmonic sweep (need interpolate to harmonic sweep and interpolate back to original frequencies, but will obviously degrade the results)
- [ ] there is a new implementation using Z instead of ifft, but ifft is the default method

Finally, there are still [IEEEP370Zc2xThru.m](https://opensource.ieee.org/elec-char/ieee-370/-/blob/master/TG1/IEEEP370Zc2xThru.m) with impedance correction and [IEEEP370mmZc2xthru.m](https://opensource.ieee.org/elec-char/ieee-370/-/blob/master/TG1/IEEEP370mmZc2xthru.m) for mixed modes (that looks promising) that are unimplemented.

vliarop, 2022-04-05:
Hi there,

I have the 2xthru.s2p data from an evaluation board for a DUT. I also have the "embedded" s2p data for the fixture-DUT-fixture. Of course, I can use the AICC utility to extract the DUT s2p file. It's all fine up to this point.

What if I need to do the bisection of the 2xthru.s2p data and extract the individual de-embed s2p data for the left and the right side (which will be identical by definition) and use these s2p data to load in the fixture simulator of the network analyzer and perform the fixture de-embedding from the network analyzer internally?

Is there a utility to extract the bisected s2p data from the 2xthru s2p data?

I can't find it in AICC utility.

jhillairet, 2022-04-06:
Isn't chopinhalf what you are looking to? (Assuming that your fixture are
the same and reciprocal)

https://scikit-rf.readthedocs.io/en/latest/api/generated/skrf.network.chopinhalf.html

?

mhuser, 2022-04-06:
> I can't find it in AICC utility.

Hello @vliarop . If I remember properly, the AICC utility also save `fixture_model_1.s2p` and `fixture_model_2.s2p` (not 100% sur of filenames) that are the left and right from bissected 2xthru.
These can be used to de-embedd the fixtures with another process.

vliarop, 2022-04-06:
Thanks @mhuser ! Yes, I just noticed that AICC saves the fixture_model_1.s2p and fixture_model_2.s2p files.

mhuser, 2022-04-09:
[image]
A first attempt for Z-corrected IEEE de-embedding is implemented. Above is a small comparison with the AICC tool.

- DUT is a skrf simulated 3xWidth portion of microstrip fed with 1xWidth 45 ohm microstrip (black dots). The fix-dut-fix (blue line) embedd additionnal thru with same microstrip.
- De-embedded with No Z correction : skrf implementation (red diamond) show excellent agreement with AICC (red curve). The Overal Z of both results are offset from original dut.
- De-embedded with Z correction : skrf (green square) show kind a 1-sample lag with AICC (green curve). Results show very good agreement with original dut, AICC is the best.

Skrf de-embedding was used to remove Z lines while AICC relies on removing ABCD parameters, maybe I should have a try with that.

There is another question : Is it better to fit with IEEE implementation as much as possible and have class and supporting fonctions duplicated into two files of de-embedding implementation extending deembedding class or shall we merge these at one place in deembedding.py (which would probably be more clean) ?

denzchoe, 2022-04-10:
> A first attempt for Z-corrected IEEE de-embedding is implemented. Above is a small comparison with the AICC tool. 

@mhuser, I guess at some point we have to compare it with the IEEE Open Standards MATLAB code to compare the accuracy of skrf. Last I check, AICC version is still legacy although running on MATLAB runtime libraries. Too bad I don't have access to MATLAB (yet). So I can't try to compare it. Can the community with MATLAB access over here spend some time to compare it before official release on skrf?

<br>

> There is another question : Is it better to fit with IEEE implementation as much as possible and have class and supporting fonctions duplicated into two files of de-embedding implementation [...]

Just my opinion. From the looks of it everything seems to work accurately? I vote merging **but** with the possibility to maintain the code structure like in Gitlab for future ease of version upgrade / edit / comparison. (Is that even possible?)

mhuser, 2022-04-10:
Hi @denzchoe 
> Just my opinion. From the looks of it everything seems to work accurately? I vote merging but with the possibility to maintain the code structure like in Gitlab for future ease of version upgrade / edit / comparison. (Is that even possible?)

This could looks a bit ugly but it is possible and will enable to easely compare with the IEEE repo.

> I guess at some point we have to compare it with the IEEE Open Standards MATLAB code to compare the accuracy of skrf.

Unfortunately I also have no access to Matlab and rf toolbox which mean it is not possible to get intermediate results or run the very code directly (what hudge advantage whith `skrf`!).

There could be another method which is simulating a structure (e.g. with Qucs) or use a file in IEEE downloads (but not sure we can legally upload it on skrf git) and test the metrics given on the paper. In this case we would not check if we have a perfect agreement with Matlab code, but see if the deembedding quality is "within limit lines". Not sure what to do here.

I still need to implement lead-in removal and case with pullback different form left to right or only one side required. But it should cost less effort now.
The AICC seems to works also for differential .s4p lines but I did no test on that yet.

mhuser, 2022-04-11:
> > > Just my opinion. From the looks of it everything seems to work accurately? I vote merging but with the possibility to maintain the code structure like in Gitlab for future ease of version upgrade / edit / comparison. (Is that even possible?)
> > 
> > 
> > This could looks a bit ugly but it is possible and will enable to easely compare with the IEEE repo.
> 
> I'm not sure to understand how do you want to proceed: Making a file for each function like in the IEEE gitlab _and_ also adding two new methods into `deembedding.py` ?

Either keep two files **OR** add two class into `deembedding.py` extending the base class, each of one would embedd supporting functions as methods to mimic the structure of the IEEE files (which are lot of code lines). With this later case, there would be no separated files.

jhillairet, 2022-04-12:
Personally, I prefer extending the base class. @arsenovic , an advice?

mhuser, 2022-04-12:
Here is a first attempt #628. I think it should now be compared to s-parameters generated by the AICC tools, but should we embedd the files on scikit-rf or not ?

jhillairet, 2022-04-13:
I think it is possible as long as the legal notice is reproduced somewhere,
for example in the directory containing the files. Otherwise the other
option is to download them from the code itself?

mhuser, 2022-04-13:
@jhillairet thank you a lot for your feedback !
> I think it is possible as long as the legal notice is reproduced somewhere, for example in the directory containing the files. Otherwise the other option is to download them from the code itself?

This could be a solution. I also have this idea : if we save example networks generated with scikit-rf as .s2p, process them with the IEEE tool and add only the resulting fixtures .s2p, then would we need to care with legal notice at all ?

jhillairet, 2022-04-13:
Yes.

You can add the link to IEEE websites in the docstring for reference.

mhuser, 2022-04-14:
With the great help of @denzchoe I corrected an error in my NZC implementation and made some new findings:
The new AICC Deembedding tool on [Amphenol website](https://www.amphenol-cs.com/software) shows a consistent behaviour with the scikit-rf implementation.
I am building an ipython notebook with an example were the impedance of 2xthru is not the same as the line in fixture-dut-fixture (e.g. because of manufacturing variations).

- NZC that take only 2xthru as input will show a slight deviation from dut timestep
- ZC that take both 2xthru and fixture-dut-fixture as input compensate for that.

[image]
The agreement between scikit-rf and AICC tool seems now very decent.
Unfortunately the AICC tools seems to no more save the fixture model s-parameters and only the deembedded results.
So fortunately you can still get these with scikit-rf.

mhuser, 2022-04-15:
[image]
PR #628 is now ready from my point of view and the agreement of green squares is even better than before (I found a place were I had not transformed indexe that start at 1 in matlab and at 0 in python in ZC method so a chunk of TL line was left unremoved).

**What is not implemented from IEEE repo:**
- Extract ZC fixtures with asymetric pullbacks or only one side
- ZC for mixed modes (which basically call the method twice, once on differential and once on common mode and mix them again)

mhuser, 2022-04-17:
I think if we can wait some day with this PR, I will have a try to implement mixed-mode deembedding, as it is essentially a wrapper of existing ZC, NZC, `Network.se2gmm()` and `Network.gmm2se()`.
I am just building a qucs dut at the moment:
[image]

mhuser, 2022-04-19:
@jhillairet do you think we can fix this? I feel if I change the sign it would break backward compatibility for `Network.gmm2se` if anyone already use it but in the other hand it does not behave as expected...

mhuser, 2022-04-19:
I will use the solution of converting twice for this sprint, as it is supposed to be compatible with a future fix.

denzchoe, 2022-04-19:
How can I buy you a beer @mhuser? 🍺 😁
(Honestly, I haven't tested it yet but it seems promising)
https://github.com/scikit-rf/scikit-rf/pull/628

You saved so many people thousands of dollars for {insert Big Instrument Company} optional VNA software here. Now everyone has access to Open Source IEEE standard! No more, I don't trust your de-embedding methodology. 

Btw I managed to get Jason Ellison's attention on your endeavour! He certainly is impressed with scikit-rf. Hopefully he will be able to test it vs Matlab later.

jhillairet, 2022-04-19:
> @jhillairet do you think we can fix this? I feel if I change the sign it would break backward compatibility for `Network.gmm2se` if anyone already use it but in the other hand it does not behave as expected...

I still does not understand if #367 is an intrinsic numerical limitation or a real bug...

JAnderson419, 2022-04-19:
@jhillairet - #367  is a real bug. I was not sure at the original time in my understanding of what was going on and could not find a mismatch between the skrf code and the paper, but discussion since and @Ttl finding the sign error in the reference paper seems to put it to rest (hopefully - the function is still a bit experimental in that it doesn't have complete test coverage ).

mhuser, 2022-04-20:
`IEEEP370_MM_NZC_2xThru`
The impedance mismatch between 2xthru and Fixture-DUT-Fixture is not corrected, leading to a time-domain bounce. Residuals of Fixtures deembedded from 2xthru are smalls.
[image]
`IEEEP370_MM_ZC_2xThru`
The impedance mismatch is corrected while building fixture model, hence no more bounce before deembedded DUT. The residuals are more significant but within some tenth of dB and some degrees. Magnitude residuals increased by mismatch between 2xthru and fixture models and phase residuals due to a more complex way to compute the length and the line by cascading chunks of variable impedances.
[image]
Comparison with `AICC De-Embedding Utility` deembedded DUTs
This seems to be a reasonnable agreement.
[image]

jhillairet, 2022-04-21:
This new feature has been implemented in #628 by @mhuser. Closing.

closed 2022-04-21 by jhillairet

ChThorn, 2023-07-04:
> 

Hello, did you find any solution to deal with this issue? Even if the residual return loss is under -20dB, the de-embedding seems to not work at this S11 behavior-like. 

I also faced the same problem with this. If you guys have some advice, I really appreciate with that.

mhuser, 2023-07-18:
Hello @ChThorn ,

This is the results with lates version of skrf:
[image]

Last time we had the occasion to check, the agreement with the original Matlab implementation was good. This means that the original Matlab version also does not give a perfect DUT. I am afraid that like many deembedding methods, it cannot perfectly extract the DUT.