scikit-rf/scikit-rf #522: 'extrapolate_to_dc' seems to process very slowly

Bob-B-Buckley, 2021-09-16:
Hi, I am sort of new to this, so hope this is ok to post here.

It seems like the `extrapolate_to_dc` method takes a very long time (~30 s per 5,000 point dataset) on Google colab.  Is there a particular computational step in `extrapolate_to_dc` that could be done much more efficiently/quickly?

Before finding scikit-rf, I wrote amateur code that did similar extrapolations to dc which ran much faster. 

If you are reading this, thank you for your time.  Any thoughts or help here would be greatly appreciated!!
-bob

FranzForstmayr, 2021-09-16:
Hi Bob,
I think the most consuming step inside `extrapolate_to_dc` is the interpolation. While `cubic` and `linear` interpolation are pretty fast, the default method `rational` seems to be quite slow for large datasets. I triggered this behaviour too quite a while ago, I decided to use `cubic` most of the time.
I didn't dig deeper into this, I'm not sure if the performance of `rational` interpolation could be increased easily.
Can you try with `kind="cubic"`, maybe this helps already.

Bob-B-Buckley, 2021-09-16:
Hi Franz,

This indeed helped me significantly.  Switching the interpolation from 'rational' to 'cubic' reduced my calculation time from  35s down to ~5s. 

Would notifying others of this time-saving change be useful to add to the docstring of `extrapolate_to_dc`?

Thanks Franz!
Best,
bob

Bob-B-Buckley, 2021-09-16:
How much more accurate is 'rational', would it be worth changing the default fitting `kind` to `cubic`?

Vinc0110, 2021-09-17:
One could argue that extrapolation (and interpolation) by definition can never be accurate. It's an informed guess. The user needs to supply the correct dc value if it has to be accurate. 

Changing the default interpolation method to `cubic` should not be a problem. We could print a warning to inform the users about this change. For example, this has happened in [NumPy's least-squares](https://numpy.org/doc/stable/reference/generated/numpy.linalg.lstsq.html?highlight=lstsq#numpy.linalg.lstsq) function, where they change the default value for the `rcond` parameter:

`Changed in version 1.14.0: If not set, a FutureWarning is given. The previous default of -1 will use the machine precision as rcond parameter, the new default will use the machine precision times max(M, N). To silence the warning and use the new default, use rcond=None, to keep using the old behavior, use rcond=-1.`

jhillairet, 2021-10-31:
OK to change the default value from `rational` to `cubic`. 

The question is: from the next version? or just passing a warning for the moment, like: 
```python
  warnings.warn("The default `kind` parameter will change "
                "from `rational` to `cubic` in future version."
                "Use `kind='rational'` to keep the old default."
                "To silent this warning, explitly define `kind`.")
```
?

Bob-B-Buckley, 2021-11-01:
My vote would be to change to 'cubic' now with a warning that it has changed, with a warning something like:

  warnings.warn("The default `kind` parameter has been changed "
                "from `rational` to `cubic`."
                "Use `kind='rational'` to keep the old default."
                "To silent this warning, explitly define `kind`.")

I'm new enough to contributing to Git that I don't want to screw anything up.  That being said, please let me know if/how I can help with this.  

Thanks to everyone who has helped with this issue, and continues to improve this codebase!

Ttl, 2021-11-01:
There is one example comparing the interpolation methods in docs: https://scikit-rf.readthedocs.io/en/latest/examples/networktheory/Rational%20Interpolation.html

`rational` should have the best accuracy but `cubic` is close. Rational interpolation is more complicated and implemented with non-vectorized numpy functions while the other interpolation methods are from scipy library, so the difference in speed is maybe not too surprising.

Vinc0110, 2021-11-11:
@jhillairet As I'm working on the warnings for PR #559, I found this example for a `DeprecationWarning`: 
https://docs.python.org/3/library/warnings.html#warnings.warn

The type of the warning was missing from your proposed code above. That `stacklevel` parameter can probably be omitted. But apart from that, I think it would be a good solution for this issue.

jhillairet, 2021-11-12:
fix proposed in #563

PR #563 (2021-11-13): Warn users about future change of 'extrapolate_to_dc' default value from rational to cubic
Fixes #522. 

Keep the default value to `rational`, but warns users this choice is now considered as deprecated and will be changed to `cubic` by default in future versions.
closed 2021-11-13 by jhillairet