scikit-rf/scikit-rf #524: Vektor fitting passivity enforcement results in even more passivity violations for some networks

FranzForstmayr, 2021-09-21:
I tried to fit zeros and poles for the following network. (Coax cable, exported from ANSYS) [Coax_test_v6_iter-0.txt](https://github.com/scikit-rf/scikit-rf/files/7202163/Coax_test_v6_iter-0.txt) 
Rename the file to '.s2p', as Github does not allow to upload files with touchstone extension.

I started with this notebook from the examples directory. [Ex4: Passivity Evaluation and Enforcement](https://scikit-rf.readthedocs.io/en/latest/examples/vectorfitting/vectorfitting_ex4_passivity.html)

As the data is only up to 3 GHz I changed the frequency vector and the initial fit seems very promising.

```python
import skrf
import numpy as np
import matplotlib.pyplot as mplt

nw = skrf.Network('Coax_test_v6_iter-0.s2p')
vf = skrf.VectorFitting(nw)
vf.vector_fit(n_poles_real=0, n_poles_cmplx=8, fit_constant=True)

# plot fitting results
freqs = np.linspace(0, 5e9, 501)
fig, ax = mplt.subplots(2, 2)
fig.set_size_inches(12, 8)
vf.plot_s_mag(0, 0, freqs=freqs, ax=ax[0][0]) # s11
vf.plot_s_mag(0, 1, freqs=freqs, ax=ax[0][1]) # s12
vf.plot_s_mag(1, 0, freqs=freqs, ax=ax[1][0]) # s21
vf.plot_s_mag(1, 1, freqs=freqs, ax=ax[1][1]) # s22
fig.tight_layout()
mplt.show()
```

[image]

As in the example the resulting network is not passive anymore, so I tried to use `passivity_enforce`

The resulting network looks like this
[image]

The passivity test fails again, with even more non-passive frequency bands.

I attach the Notebook to reproduce this issue easily. Rename the file to ipynb, same issue as before.
[vectorfitting_ex4_passivity.txt](https://github.com/scikit-rf/scikit-rf/files/7202234/vectorfitting_ex4_passivity.txt)

FranzForstmayr, 2021-09-21:
Obviously the `passivity_enforce` did not converge in my case. 
I did not realize first, as no logging error was thrown. This is related to #526 

Increasing the iterations did not fix my issue in this case.

Vinc0110, 2021-09-21:
Thanks for the helpful report. I can reproduce the problem. By the way, `fit_constant=True` is not necessary as it is the default.

Playing around with your test file, I found out that your choice of starting poles was not ideal and caused a problem. Using `n_poles_real=1, n_poles_cmplx=5` works much better on your data and results in a successful passivation:

**Fit before passivation:**
`vf.is_passive()` returns `False`
[image]

**Fit after passivation:**
`vf.is_passive()` returns `True`
[image]

Comparing a few setups:
1. `n_poles_real=0, n_poles_cmplx=8`: fit takes 9 iterations, **passivation fails**
2. `n_poles_real=0, n_poles_cmplx=6`: fit takes 51 iterations, passivation succeeds after 16 iterations
3. `n_poles_real=0, n_poles_cmplx=5`: fit takes 6 iterations, **passivation fails/stops** after 2 iterations
4. `n_poles_real=1, n_poles_cmplx=5`: fit takes 8 iterations, passivation succeeds after 19 iterations

So, this reveals a weakness of the implemented vector fitting algorithm in that it can be quite sensitive to the choice of starting poles. It would be great if `vector_fit()` would automatically find, correct, or assist finding the optimal number of starting poles, but it currently doesn't.

In your case, there were too many poles with respect to the resonances in the fitting band. The algorithm is relocating unnecessary poles to either lower or higher frequencies outside the fitting interval. In your case, it was creating many (6 out of 11) real-valued poles, two of them with suspiciously large amplitudes (>1e10), which render them useless. 

**With `n_poles_real=0, n_poles_cmplx=8`:**
```
vf.poles = 
[-2.95343488e+11+0.00000000e+00j -4.26517045e+10+0.00000000e+00j
 -2.13358706e+09+2.53192065e+10j -1.54396072e+09+0.00000000e+00j
 -1.45410602e+09+1.50339648e+10j -1.44720801e+09+1.00240454e+10j
 -1.44032479e+09+2.00562193e+10j -1.43846469e+09+5.01407330e+09j
 -1.41615085e+09+0.00000000e+00j -2.18324951e+08+0.00000000e+00j
 -2.14396242e+08+0.00000000e+00j]
```
P-Z-diagram for S11:
[image]

**With `n_poles_real=1, n_poles_cmplx=5`:**
```
vf.poles = 
[-7.54417047e+09+3.23267462e+10j -1.45016734e+09+1.50350070e+10j
 -1.44998647e+09+1.00215550e+10j -1.43908562e+09+5.01432723e+09j
 -1.41325808e+09+0.00000000e+00j -1.37173613e+09+2.00792255e+10j]
```
P-Z-diagram for S11:
[image]

The original paper proposed the idea of detecting and removing such useless poles during the relocation process. I'll probably experiment with that and see if it relaxes the situation. Not sure how to find a universal threshold, yet...

FranzForstmayr, 2021-09-21:
Thanks for the detailed explanation.
So you expect the main issue in the `vector_fit` method itself? I assumed it's about the `passivity_enforce` method, as the samples fit quite good before, even when having too much poles.

If I understand you correct, I could implement a quick workaround by counting the poles in a bounded frequency range and run `vector_fit` again with the reduced number?

FranzForstmayr, 2021-09-21:
Thanks for the advice!

jhillairet, 2021-10-30:
@FranzForstmayr, @Vinc0110  can we close this issue ?

Vinc0110, 2021-11-01:
I think it can be closed, right @FranzForstmayr ?

FranzForstmayr, 2021-11-01:
Yes, think so too. It's more a limitation of the algorithm than an issue in
the code.

closed 2021-11-01 by jhillairet