scikit-rf/scikit-rf #525: Connecting large Network with multiple opens using circuits class

luar123, 2021-09-21:
I need to create a smaller network from a large one (~500 ports) by leaving most of the ports open. 
I have a arbitrary list of port indexes that need to be left open and the complementary list of port indexes that need to be a port in the new network.
What I have tried so far: 
- Subnetwork will not work, as the unused ports are ideally matched.
- using connect() to connect an open one-port to each unneeded port is working, but extremely slow, as I am calling connect for each port (>300 times).
- I thought the circuit class is the ideal solution, but somehow it is not working:

I tried to connect all needed ports of the network to a port network, and leave all others open:
```
f=ntwk.frequency
conn = [[(ntwk, i), (rf.Circuit.Port(f, name="port"+i),0)] for i in extract_ports_idx]
resulting_circuit = rf.Circuit(conn)
resulting_network = resulting_circuit.network
```
However, the result is the same as using subnetwork.

Is there an issue with the circuits class or am I using it wrong?

jhillairet, 2021-09-22:
A  `Port` object is similar to matched load. If you want to leave the ports open, just do nothing in the circuit description.

Example: assuming `ntwk` is a 2 port-Network, if your connection description is:

```python
conn = [
    [(port, 0), (ntwk, 0)]
]
```
it means that the port 1 of ntwk is left open.

luar123, 2021-09-22:
Yes, thanks. That is exactly what I was doing. Here is an simple example:
```
import numpy as np
import skrf as rf
tee = rf.data.tee
````
Matching port 2 using subnetwork():
```
rf.subnetwork(tee, [0,1]).s[0]
```
array([[-0.33333333+0.j,  0.66666667+0.j],
       [ 0.66666667+0.j, -0.33333333+0.j]])

And leaving port 2 open using connect():
```
f=tee.frequency
open_oneport=rf.Network(frequency=f, s=np.ones(len(f)), name='open')
tee_open=rf.connect(tee, 2, open_oneport, 0)
tee_open.s[0]
```
array([[7.50011164e-13+0.j, 1.00000000e+00+0.j],
       [1.00000000e+00+0.j, 7.50011164e-13+0.j]])

This seems correct to me. However using circuits, connecting a port to port 0 and 1 (thus leaving port 2 open):
```
conn=[[(tee, 0),(rf.Circuit.Port(f, name="port1"),0)],[(tee, 1),(rf.Circuit.Port(f, name="port2"),0)]]
resulting_circuit = rf.Circuit(conn)
resulting_network = resulting_circuit.network
resulting_network.s[0]
```
array([[-0.33333333+0.j,  0.66666667+0.j],
       [ 0.66666667+0.j, -0.33333333+0.j]])

Is the same as using subnetwork() where port 2 is matched.

jhillairet, 2021-09-22:
Hum, you're right, I need to investigate why. 

A workaround to speed up your code could be to create open networks as you did (with different name properties) and connect them using Circuit, to build the resulting Network in a single step. Like:

```python
conn=[
      [(tee, 0), (rf.Circuit.Port(f, name="port1"), 0)],
      [(tee, 1), (rf.Circuit.Port(f, name="port2"), 0)],
      [(tee, 2), (rf.Network(frequency=f, s=np.ones(len(f)), name='open1'), 0)],
      ]
resulting_circuit = rf.Circuit(conn)
resulting_network = resulting_circuit.network
print(resulting_network.s[0])
```
which gives:
```
[[7.49955653e-13+0.j 1.00000000e+00+0.j]
 [1.00000000e+00+0.j 7.49955653e-13+0.j]]
```

jhillairet, 2021-09-22:
Eventually, I can add a `rf.Circuit.Open()` to make it more clear and explicit. Do you think it is a good idea ? 
I could also add shunt/series impedances.

luar123, 2021-09-22:
The workaround works well and is fast, thank you!
(I tried something similar yesterday, but used always the same Open, which did not work)

I think additional circuit elements would be a good idea to make things more simple.
Btw, I think `rf.Circuits.Ground()` is returning a two-port network with two shorts to ground instead of a one-port as is written in the documentation.

jhillairet, 2021-09-22:
> The workaround works well and is fast, thank you!

OK, good. Just for info, how faster is it in comparison ? 

> (I tried something similar yesterday, but used always the same Open, which did not work)

Yes, in this case your N ports are all connected to the same Open via a (N+1)-junction, which is probably not what you want. 
 
> I think additional circuit elements would be a good idea to make things more simple.

OK, it is an easy addition for the next version.

> Btw, I think `rf.Circuits.Ground()` is returning a two-port network with two shorts to ground instead of a one-port as is written in the documentation.

Indeed. Thanks for the notification, I will correct the documentation as well.

luar123, 2021-09-22:
> > The workaround works well and is fast, thank you!
> 
> OK, good. Just for info, how faster is it in comparison ?
With the current dataset the circuits approach takes 29 seconds. I don't want to run the connect approach again, as it was something like 30 minutes or even more. So I thing it is 50-100 times faster.

jhillairet, 2021-10-06:
also solved with PR #529