scikit-rf/scikit-rf #553: Vector Fitting: difficult to fit data with a large negative slope

alexatadi, 2021-11-01:
Matlab has an example for this, fitting improved by introduced a "delay factor". Any way to deal with it in scirf? Thanks.
https://www.mathworks.com/help/rf/ug/using-the-delayfactor-parameter.html

jhillairet, 2021-11-01:
@Vinc0110, could you have a look?

Vinc0110, 2021-11-01:
Hi @alexatadi,
unlike the Matlab routine, the implementation in scikit-rf's `vector_fit()` does not include this kind of delay "improvement".

Out of curiosity: Why would you need such a feature? Can you maybe share some example data that you need to fit?

But regardless of its use cases, you should be able to achieve a similar result with the delay function in `Network.delay()`:
Instead of delaying the vector fitted results, you can "accelerate" the frequency responses before the fitting by applying a negative delay to them. 
If I understood the operation of the Matlab function correctly (see also https://de.mathworks.com/help/rf/ref/rationalfit.html), the effect should be the same, even though a network with negative delay is unphysical.

Example code for the delay:
```
import skrf
import matplotlib.pyplot as mplt

# using the ring slot as example 2-port
nw = skrf.data.ring_slot
nw.name = 'original'

# 300 micron delay at port 1
nw_delayed = nw.delay(300, unit='um', port=0)
nw_delayed.name = 'delayed'

# -300 micron delay at port 1
nw_accel = nw.delay(-300, unit='um', port=0)
nw_accel.name = 'acclerated'

# plot magnitude and phase
fig = mplt.figure(figsize=(6, 4))
ax = fig.subplots(2, 1)
nw.plot_s_db(1, 0, ax=ax[0])
nw.plot_s_deg_unwrap(1, 0, ax=ax[1])
nw_delayed.plot_s_db(1, 0, ax=ax[0])
nw_delayed.plot_s_deg_unwrap(1, 0, ax=ax[1])
nw_accel.plot_s_db(1, 0, ax=ax[0])
nw_accel.plot_s_deg_unwrap(1, 0, ax=ax[1])
fig.tight_layout()
mplt.show()
```
[image]

alexatadi, 2021-11-02:
Hi @Vinc0110 ,
Thanks a lot for your support! I did notice the vector_fit() significantly reduced the number of poles over matlab's function in some testing. It seems the attached network is difficult to fit by vector_fit() even with a large number of poles.
[s4p.zip](https://github.com/scikit-rf/scikit-rf/files/7456936/s4p.zip)

Vinc0110, 2021-11-06:
I finally had some time to check out your example data. It's pretty interesting, but also very strange with those huge phase shifts. May I ask what you want to do with it? Include it in a SPICE simulation?
There must be some long transmission lines inside that backplane (1-2 meters?). It does not seem very practical to me to model such an electrically big structure as a whole. It would be much easier if you had measurements of the individual parts (connectors etc.) without the lines in between, but I'm not sure if that would be good enough if you care about things like cross-talk in a back plane.

Long story short, my initial idea of removing (de-embedding) any external delay with skrf's `Network.delay()` method does not really work here. The internal delays seem to be much larger than the external ones, so this does not help much. Here is what I tried to prepare the network before vector fitting:

**Code:**
```
nw = skrf.Network('default.s4p')

deembedding_factor = 0.98
phi_remove = []
for i in range(nw.nports):
    phi_max = np.rad2deg(np.unwrap(np.angle(nw.s[:, i, i]))[-1])
    phi_remove.append(int(deembedding_factor * np.abs(phi_max)))
    print('S{}{}: Phi_max = {:.1f} degrees; Removing {} degrees ({}%) at port {}'.format(
        i+1, i+1, phi_max, phi_remove[i], deembedding_factor * 100, i+1))
    nw = nw.delay(-0.5 * phi_remove[i], unit='deg', port=i, media=None)

for i in range(nw.nports):
    print()
    for j in range(nw.nports):
        phi_max = np.rad2deg(np.unwrap(np.angle(nw.s[:, i, j]))[-1])
        l = np.abs(phi_max) / 360 * skrf.c / nw.f[-1]
        print('S{}{}: Phi_max = {:.1f} degrees; Equiv. free-space length = ={:.3f} m'.format(
            i+1, j+1, phi_max, l))
```

**Output:**
```
S11: Phi_max = -7318.5 degrees; Removing 7172 degrees (98.0%) at port 1
S22: Phi_max = -7231.2 degrees; Removing 7086 degrees (98.0%) at port 2
S33: Phi_max = -6515.2 degrees; Removing 6384 degrees (98.0%) at port 3
S44: Phi_max = -7233.4 degrees; Removing 7088 degrees (98.0%) at port 4

S11: Phi_max = -175.1 degrees; Equiv. free-space length = =0.010 m
S12: Phi_max = -28984.0 degrees; Equiv. free-space length = =1.609 m
S13: Phi_max = -5960.2 degrees; Equiv. free-space length = =0.331 m
S14: Phi_max = -29111.1 degrees; Equiv. free-space length = =1.616 m

S21: Phi_max = -28997.9 degrees; Equiv. free-space length = =1.610 m
S22: Phi_max = -173.4 degrees; Equiv. free-space length = =0.010 m
S23: Phi_max = -29604.3 degrees; Equiv. free-space length = =1.644 m
S24: Phi_max = -5626.1 degrees; Equiv. free-space length = =0.312 m

S31: Phi_max = -5957.6 degrees; Equiv. free-space length = =0.331 m
S32: Phi_max = -29604.5 degrees; Equiv. free-space length = =1.644 m
S33: Phi_max = -156.6 degrees; Equiv. free-space length = =0.009 m
S34: Phi_max = -29432.7 degrees; Equiv. free-space length = =1.634 m

S41: Phi_max = -29141.1 degrees; Equiv. free-space length = =1.618 m
S42: Phi_max = -5626.2 degrees; Equiv. free-space length = =0.312 m
S43: Phi_max = -28687.1 degrees; Equiv. free-space length = =1.593 m
S44: Phi_max = -173.6 degrees; Equiv. free-space length = =0.010 m
```
As you can see, the reflection coefficients are nearly free of delay (<360°), but the transmission coefficients still have many 1000 degrees, i.e. >1m equivalent line length. I did not notice any improvement in the fitting process with this network over the original one.

In conclusion, I see two options if you really need to fit this data:
- Brute force fitting with 100-200 poles. This worked relatively well for me with 140 real poles, but the fit took like 30 minutes on my machine. Also, the fit this was far from perfect, so this depends on your required accuracy.
- Fit only one response at a time; with or without removing the delay. Depending on what you want to do with the data, it might be enough to only analyze certain responses like S34. So can can only fit that one (and include it in your SPICE simulation?). This worked very well in my test with S21 from your example.
Procedure: Copy S21 to a new 1-port `Network`, remove the delay and fit with 40+5 poles (for example):
```
nw_orig = skrf.Network('default.s4p')
nw = skrf.Network(f=nw_orig.f, s=nw_orig.s[:, 1, 0], z0=nw_orig.z0[1, 0])
phi_max = np.rad2deg(np.unwrap(np.angle(nw.s[:, 0, 0]))[-1])
phi_remove = int(0.98 * np.abs(phi_max))
print('S21: Phi_max = {} degrees; Removing {} degrees'.format(phi_max, phi_remove))
nw_delayed = nw.delay(-0.5 * phi_remove, unit='deg', port=0, media=None)

vf = skrf.VectorFitting(nw_delayed)
vf.vector_fit(n_poles_real=40, n_poles_cmplx=5)
```

Vinc0110, 2021-11-06:
It might be obvious, but any removed length `l` needs to be re-applied after vector fitting. This can be done either by multiplying the fit with `exp(j * 2 * pi * l / lambda)` or by connecting a lossless transmission line of length `l` in SPICE.

alexatadi, 2021-11-06:
Thanks @Vinc0110 , I do appreciate your detailed guidance, really helpful 👍 The data is from Matlab's example, a long trace on backplane. In my practice, I found the vector_fit() is obviously better than Matlab's function, always perfectly fit S parameters with less poles. But met problem when fit such curve with a negative slope, so mainly consult if I missed something. As you said fit subnetwork is an option, I can also split the frequency range to several segments, which also make fitting easier.
Please close this issue, it's resolved.

Vinc0110, 2021-11-06:
Cool! I'm glad you like the vector fitting feature.
Being the creator of this issue, you can actually close it yourself (or @jhillairet does it).

closed 2021-11-06 by alexatadi

Vinc0110, 2021-11-07:
This is a good example, so I'd like to advertise the upcoming speed improvements for `vector_fit()`, which make quite a difference for this large 4-port. Stay tuned for the next release version 0.20.

Using 200 complex-conjugate starting poles, the pole relocation did not fully converge and stopped after 100 iterations, but the results are already pretty good. With the current code, it took more than 2 hours (8260 seconds) on my machine. In comparison, the new code only took 42 minutes.

```
nw = skrf.Network('/home/vinc/Desktop/default.s4p')
vf = skrf.VectorFitting(nw)
vf.vector_fit(n_poles_real=0, n_poles_cmplx=200)
```

Here's the result:

**Magnitudes**:
[image]

**Phases**:
[image]