scikit-rf/scikit-rf #554: Network.delay(): Minor inaccuracies + Better Documentation

Vinc0110, 2021-11-01:
Writing the example code for issue #553 I noticed some minor issues with the [`Network.delay()` function](https://scikit-rf.readthedocs.io/en/latest/api/generated/skrf.network.Network.delay.html). I might fix them myself, but I would first like to discuss if they are actually a problem:

- The documentation for this function should be improved: Inform the user that the delay in degrees is to be specified at the center frequency. This info is hidden deep in docstring of `Media.theta_2_d()`.
- The magnitude of a delayed network response will not remain unaffected (which it should be, correct?). According the the implementation with a T-line from `Media.line`, the magnitude of the delay line should be exactly 1, i.e. a lossless line: https://github.com/scikit-rf/scikit-rf/blob/5e52c50b5ca301b92cbaecab6d86b32e22d76b94/skrf/media/media.py#L829 However, this is not exactly the case if you look at the notch in S12 and S21 in the following example.
- In the same example, the behavior or shape of the delayed phases of S11 and S22 also "flips", which is weird and possible wrong. (I know, the phase of a signal with almost 0 magnitude is a strange thing, so maybe this is still alright...)

Please let me know what you think, so this can be fixed, improved, or at least documented.

```
import skrf
import matplotlib.pyplot as mplt

nw = skrf.data.ring_slot
nw.name = 'original'
nw_delayed = nw.delay(300, unit='um', port=0)
nw_delayed.name = 'delayed'
nw_accel = nw.delay(-300, unit='um', port=0)
nw_accel.name = 'acclerated'

fig = mplt.figure(figsize=(10, 10))
ax = fig.subplots(4, 2)
for i in range(2):
    for j in range(2):
        nw.plot_s_db(i, j, ax=ax[2 * i, j])
        nw.plot_s_deg_unwrap(i, j, ax=ax[2 * i + 1, j])
        nw_delayed.plot_s_db(i, j, ax=ax[2 * i, j])
        nw_delayed.plot_s_deg_unwrap(i, j, ax=ax[2 * i + 1, j])
        nw_accel.plot_s_db(i, j, ax=ax[2 * i, j])
        nw_accel.plot_s_deg_unwrap(i, j, ax=ax[2 * i + 1, j])
fig.tight_layout()
mplt.show()
```

[image]

jhillairet, 2021-11-02:
>     * The documentation for this function should be improved: Inform the user that the delay in degrees is to be specified at the center frequency. This info is hidden deep in docstring of `Media.theta_2_d()`.

Yes, I agree. I learned it the hard way myself also ;)
 
> * The magnitude of a delayed network response will not remain unaffected (which it should be, correct?). According the the implementation with a T-line from `Media.line`, the magnitude of the delay [...]

OK, I've seen the second message. Indeed, with the proper char impedance, this should not change the magnitude.

jhillairet, 2021-11-02:
closed with PR #556

closed 2021-11-02 by jhillairet