scikit-rf/scikit-rf #565: Coverage statistics not working

Vinc0110, 2021-11-14:
The [badge from Coveralls](https://coveralls.io/github/scikit-rf/scikit-rf) with the current coverage of the tests in scikit-rf, which is linked on the front page, is showing a surprisingly low value of 38%.
When you go to that website, all values for "skrf" are at 0%. Only when you expand a massive tree for each Python release in `/opt/hostedtoolcache/Python/...`, you will eventually find some coverage values for the modules. However, it's not showing any details about hits and misses within the code, which used to be there. Instead, this error is shown:
```
Source Not Available

The file "/opt/hostedtoolcache/Python/3.7.12/x64/lib/python3.7/site-packages/scikit_rf-0.20.0-py3.7.egg/skrf/vectorFitting.py" isn't available on github. Either it's been removed, or the repo root directory needs to be updated.
```

Is this maybe related to the changes from Nose to Pytest in PR #560 ? Or is it just about a missing update of the test reports after the last package release?

jhillairet, 2021-11-14:
Yes indeed. The coverage dropped from  after https://github.com/scikit-rf/scikit-rf/commit/624bb2f748bf17dc3538b8e706b8373753e04620 which indeed indicates it comes from the change to pytest. Looks like the directory skrf is not included.

FranzForstmayr, 2021-11-14:
I'll have a look later, sorry for the inconvenience!

jhillairet, 2021-11-14:
with the previous version of the CI (nose), the coverage file tree looked like that:
[image]

while now it looks like: 
[image]

jhillairet, 2021-11-14:
ok, just have seen your message in #566 
let's try.

commit 7217259c (2021-11-17): Merge pull request #568 from FranzForstmayr/fix/coveralls

fixes #565
closed 2021-11-17 by jhillairet