scikit-rf/scikit-rf #569: scikit-rf TypeError: must be real number, not str

cloidnerux, 2021-11-16:
Hello,

I upgraded to 0.20 and found some old code to stop working with the mentioned error message
`scikit-rf TypeError: must be real number, not str`
I found that the importer in `touchstone.py` at line 221 just writes the string literal into `self.resistance`, which of course is  a string causing the error in the function `get_gamma_z0_from_fid` line 545. As far as I see it the loaded string literal should be converted to a float value

`self.resistance = float(toks[4])`

jhillairet, 2021-11-16:
Thank you @cloidnerux for reporting.

So the error is trigged when importing a touchstone file, that right? To reproduce the error, could you give us:
- the python version
- the header and first data lines of the Touchstone file you are using 
?

jhillairet, 2021-11-16:
Seems like Numpy (1.20) knows how to  cast string into numerical values:
```python
>>> np.array('50', dtype=complex)
Out[11]: array(50.+0.j)

>>> np.__version__
Out[12]: '1.20.3'
```
what is your Numpy version?

jhillairet, 2021-11-16:
This is a consequence of the changes made in https://github.com/scikit-rf/scikit-rf/commit/6399460a3909951576f8891656a7b1a6c7cee45e# from
`z0 = npy.complex(self.resistance)`
to
`z0 = npy.array(self.resistance, dtype=complex)`

I guess that updating your Numpy to more recent version would solve the issue.

cloidnerux, 2021-11-16:
Yes, I have python 3.6.9 and numpy 1.17.4

jhillairet, 2021-11-16:
> Yes, I have python 3.6.9 and numpy 1.17.4

OK. As the test suite runs are OK with Python 3.6 (it uses numpy 1.19.5), updating your numpy should solve your issue.

jhillairet, 2021-11-16:
@cloidnerux let us know if updating Numpy solves your issue

apage224, 2021-11-16:
I'm really liking all the improvements lately. Perhaps, it might be useful to start using pyproject.toml or poetry to manage and enforce dependency versions such as numpy >=1.2.* and >1.9.5

closed 2022-01-14 by jhillairet