scikit-rf/scikit-rf #570: Vector Fitting: Error in equation?

dxxx9, 2021-11-16:
Is it possible that there is an description mistake in the [first Vector Fitting equation](https://scikit-rf.readthedocs.io/en/latest/tutorials/VectorFitting.html) of the docs? The one to calculate the transfer function of a system.

My suggestion is that instead of the zeros (z_k) in the numerator it should be the residues (c_k) because it's a sum of partial fractions and therefore has no zeros in this form. Sure you can calculate the zeros from that partial form but I'm wondering if that should be the residues instead of the zeros there.

So, instead of this:
[image]
is should be this?
[image]

Please let me know if I'm wrong here.
Thanks!

[Github Permalink to docs](https://github.com/scikit-rf/scikit-rf/blob/c50c7cc9444d03145be22d40e3c72d863b0abdf8/doc/source/tutorials/VectorFitting.ipynb?short_path=5e8554d#L19)

Vinc0110, 2021-11-16:
You are right, thanks for the hint! I was lazy with the naming/notation (also in the source code).

Of course, the tutorial should not contain errors and needs to be corrected, but I highly recommend you use the actual vector fitting paper instead if you want to study the algorithm in more detail. It confirms your suspicion:

[image]

dxxx9, 2021-11-18:
Thanks for the response! Yes, I use the the original paper. That's how I noticed the difference. Just wanted to let you guys know. 👍

Vinc0110, 2021-11-18:
That's good. Thanks again. I'll correct the tutorial and the variable names in the source code soon.

By the way: Depending on how deep you dig into vector fitting, you might find other mistakes or weird things in the examples or the code. Feel free to report any problems or make suggestions/requests for extra features. The module is still relatively new and, so far, I only implemented a limited set of features, which I find useful or interesting. Lot's of things could be extended or improved...

closed 2021-12-02 by jhillairet

dxxx9, 2021-12-16:
@Vinc0110 Thanks for correcting. If I stumble over something, I will let you know.

PR #577 (2021-12-02): Vector Fitting: Notation and other minor improvements
Fixes #570