scikit-rf/scikit-rf #572: How to use UnknownThru

YasuoKawahara, 2021-11-22:
Hello,

I compared UNknownTHRU and SOLT.
I focused on the return loss of LINE1, but the result of UNknownTHRU (result_UNknownTHRU.jpg) is about 20 dB, while the result of SOLT (result_SOLT.jpg) is about 40 dB.
I thought both would give similar results, but is this result reasonable?

I have attached the archive file (UNknownTHRUvsSOLT.zip) .
Its contents are a set of my Python script files and measurement data files.
I would be very grateful if you could check my Python script for any mistakes.

Thank you.

[image]
[image]
[UNknownTHRUvsSOLT.zip](https://github.com/scikit-rf/scikit-rf/files/7578448/UNknownTHRUvsSOLT.zip)

jhillairet, 2021-11-27:
Dear @YasuoKawahara , 

How did you generate your ideal components? And measured ones?

YasuoKawahara, 2021-12-01:
Dear @jhillairet,

Thank you for your inquiry.

The calkit I used is 85052D.
The standard of this calkit is defined as follows.
Open:
C0 = 49.433e-15
C1 = -310.13e-27
C2 = 23.168e-36
C3 = -0.15966
Delay = 2.9243E-11
Loss = 2.2
Z0 = 50
Short:
L0 = 2.0765e-12
L1 = -108.54e-24
L2 = 2.1705e-33
L3 = -0.01e-42
Delay = 3.1785E-11
Loss = 2.36
Z0 = 50
Load:
Delay = 0
Loss = 0
Z0 = 50
Thru:
Delay = 0
Loss = 0
Z0 = 50

I entered the above values into ADS to create ideal data.

The measurement data is the measurement result of PNA-X.
I measured each standard of 85052D in UnCal state and saved it in TouchStone format.

Please let us know if you have any additional information you need.
Thank you.

JAnderson419, 2021-12-16:
Looking at the calibration module, SOLT inherits the TwelveTerm error model class whereas SOLR/UnknownThru inherits the EightTerm model, so there may be some underlying difference in those two base classes (or error models) that is causing the difference in response. The following are the values I saw looking into the error matrices for the first frequency point, with  differences bolded (the defined isolation terms at 1E-5 are basically 0).

As an aside, the SOLR cal also gives me the following error when I try to access the coefs_8term parameter:
```
cal.coefs_8term
Traceback (most recent call last):
  File "...\site-packages\IPython\core\interactiveshell.py", line 3441, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-3-aba298da4f98>", line 1, in <module>
    cal.coefs_8term
  File "...\site-packages\skrf\calibration\calibration.py", line 669, in coefs_8term
    return convert_12term_2_8term(d)
  File "...\site-packages\skrf\calibration\calibration.py", line 5265, in convert_12term_2_8term
    Etf = coefs_12term['forward transmission tracking']
KeyError: 'forward transmission tracking'
```

## SOLT/TwelveTerm
'forward directivity': -0.03183546+0.05610464j
'forward reflection tracking':  1.0390609 -0.05947319j
'forward source match': 0.17416413-0.22408532j
'forward isolation': 0.+0.j
'forward transmission tracking': **0.99350004-0.30306149j**
'forward load match': 0.03555198-0.36210156j
'reverse directivity': -0.04106086+0.05751813j
'reverse reflection tracking':  0.88253415-0.49415965j
'reverse source match': **0.04062471-0.27210813j**
'reverse isolation': 0.+0.j
'reverse transmission tracking': **0.98228011-0.26166231j**
'reverse load match': **0.22259158-0.31805504j**

EightTerm terms (idk why they appear in 12Term list):
_'forward switch term': 0.03853388-0.0800554j 
'reverse switch term': 0.05095088-0.08748669j
'k': 1.0032331 +0.21260638j_

## SOLR
'forward directivity': -0.03183547+0.05610463j
'forward source match': 0.17416412-0.22408533j
'forward reflection tracking':  1.0390609 -0.05947319j
'reverse directivity': -0.04106087+0.05751814j
'reverse reflection tracking': 0.88253415-0.49415965j
'reverse source match': 0.04062471-0.27210813j
'forward isolation': -2.67440023e-06-3.32344733e-05j
'reverse isolation': -6.59150214e-05+2.81065325e-05j
'forward load match': **0.09814989-0.34094962j**
'reverse load match':  **0.14926924-0.32691822j**
'forward transmission tracking': **0.98884152-0.28189551j**
'reverse transmission tracking': **0.98797707-0.28589575j**

Ttl, 2021-12-16:
Comparing the solved error parameters the biggest difference is switch terms.

If you export the switch terms from the SOLT calibration with:

```
fsw = rf.Network(s=cal.coefs_8term['forward switch term'], f=cal.frequency.f)
fsw.write_touchstone('fsw.s1p')
rsw = rf.Network(s=cal.coefs_8term['reverse switch term'], f=cal.frequency.f)
rsw.write_touchstone('rsw.s1p')
```

and use them in the UnknownThru calibration the results are much more similar to SOLT results.

arsenovic, 2021-12-16:
there are conversions functions between 8-12 term error models, which might  be helpful, 
https://github.com/scikit-rf/scikit-rf/blob/master/skrf/calibration/calibration.py#L5235

YasuoKawahara, 2021-12-27:
Thank you for your reply.
I was convinced that my usage was correct as I could get reasonable results by replacing it with SOLT's SWterm.
Probably there is a problem with the measurement result of SWterm, so I would like to carry out remeasurement.
I will report again if good results are obtained.

closed 2022-03-22 by jhillairet

xianglong1988, 2025-12-27:
@YasuoKawahara   Is the issue solved now? I had  the same  doubts.  Any  good  comments, please kindly share with us!   thank you !

xianglong1988, 2025-12-27:
> Comparing the solved error parameters the biggest difference is switch terms.
>
> If you export the switch terms from the SOLT calibration with:
>
> ```
> fsw = rf.Network(s=cal.coefs_8term['forward switch [...]

@Ttl  I try it, but the issue still being!

xianglong1988, 2026-01-03:
[image]

The switch items errors import to the SOLR, the results is very similar with SOLT result. I summarized the result in one figure, present  to you!

the switch itens can be caculated based on the 12 items errors.
the theory you can find in the paper: Formulations of the Basic Vector Network Analyzer Error Model including Switch-Terms
equation(30) and (31)