scikit-rf/scikit-rf #573: scikit-rf 2021 survey results

jhillairet, 2021-11-22:
Dear scikit-rf users,

We have made a short survey intended for all the users of the scikit-rf, to:
- better know the users and the usages,
- determine what are the functionalities used the most and the less,
- find the ones which could be improved,
- define new directions for developments.

Could you please fill this short (5 min or less) and anonymous survey by following the link below:

https://forms.gle/uonurVVb4FPhTLQm9

Thanks in advance for your time,

The results will be posted in this issue.

jhillairet, 2021-11-27:
[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

On improvements/doc :
Could you give some details on your previous answer?
•	Ways to work with, run functions on, and extra various summary parameters from multi-dimensional collections of Networks, with varied independent variable (like temperature, bias conditions, etc). I was not aware of NetworkSet from the above question so I'll have to check that one out too. 
•	More examples and case studies to not only explain the basics of the package but also to show some neat use of it...
•	More examples; sometimes the documentation is somewhat obscure,
•	It would be nice to have filter design features such as coupling matrix synthesis, etc.
•	Perhaps integration with ADS/CST or other well known EM simulation packages? 
•	hands-on tutorial notebooks/presentation that could be given to engineering colleagues to introduce them to skrf
•	Multidimensional touchstone arrays, like sweep over temperature or something else. 
•	Line-up calculation tools.
•	More flexibility in the option of circuit analysis using S-parameter blocks and typical passive circuit elements.
•	More documentation is always good :-)
•	Multi-language support for documentation
•	More examples, tutorials of features

jhillairet, 2021-12-15:
updated answers (28 so far... quite a small number!)

closed 2022-01-08 by jhillairet

arsenovic, 2022-01-08:
julien you should have a link to your GH sponsors on scikit-rf page, since you are current maintainer.  looks like quite a few people consider sponsorship.