scikit-rf/scikit-rf #574: Is scikit-rf capable of parallel computing?

ruizhi92, 2021-11-23:
Hi,

I'm wondering if any parallel computing is enabled in scikit-rf package? It took quite some time computing for example `build_circuit()` function (normally 1~2 seconds). What I'm trying to do is to optimize the configuration of a cuicuit, which requires fast and repeated computation of build_circuit(). Is it possible to implement any method to speed it up? Thank you.

jhillairet, 2021-11-23:
Hello,

What do you mean by `build_circuit` ? Creating a circuit using `Circuit` ?

In `Circuit`, the main bottleneck is the use of the scipy `block_diag` function. This function is currently called for every frequency, which become problematic when the number of frequency points is big. 

https://github.com/scikit-rf/scikit-rf/blob/7217259c2f94abdd28f163379624cf2ed0270a8f/skrf/circuit.py#L763-L769

I didn't manage to convert this function using a faster 100% Numpy version like I did for other algebra operations here and there. 

`block_diag` creates a diagonal matrix using N matrices of (M,M) shape (so resulting in a (NxM, NxM) shape), and this is repeated to create a (f, NxM, NxM) big matrix. 

If you can find a smarter way to create the (f, NxM, NxM) matrix, you would boost the speed of creating a Circuit.

ruizhi92, 2021-11-23:
> Hello,
>
> What do you mean by `build_circuit` ? Creating a circuit using `Circuit` ?
>
> In `Circuit`, the main bottleneck is the use of the scipy `block_diag` function. This function is currently called [...]

Sorry for the confusion, I actually refer `build_circuit()` to 

    cir = rf.Circuit(cnx)
    ntwk = cir.network

I think `cir.network` takes most of the time.

jhillairet, 2021-11-26:
@ruizhi92, @FranzForstmayr has improved by a decade the performances of Circuit. Please  have a look at the latest github master version of scikit-rf.

closed 2021-11-26 by jhillairet

ruizhi92, 2021-11-29:
@jhillairet @FranzForstmayr Really appreciate that!