scikit-rf/scikit-rf #579: Some figures in the deembedding tutorial do not appear

keikawa, 2021-12-03:
Some figures added in #557 do not appear, as reported in #576.
https://scikit-rf.readthedocs.io/en/latest/tutorials/Deembedding.html

However, no problem on the GitHub repository.
https://github.com/scikit-rf/scikit-rf/blob/master/doc/source/tutorials/Deembedding.ipynb

I have no idea what caused it...

Vinc0110, 2021-12-03:
[nbsphinx](https://nbsphinx.readthedocs.io/en/0.8.7/index.html) sometimes behaves weirdly. I have had similar problems with SVG images, but your tutorial is also missing a few PNG images. 
You could try the <img> tag instead, as mentioned in [nbsphinx docs](https://nbsphinx.readthedocs.io/en/0.8.7/markdown-cells.html#Using-the-HTML-%3Cimg%3E-tag). Maybe this works better, i.e. more reliably:

`<img src="images/notebook_icon.png" alt="Jupyter notebook icon" width="300">`

keikawa, 2021-12-03:
Thank you for your advice :)

PR #580 (2021-12-03): Change syntax of the images from Markdown to HTML in the de-embedding tutorial
Fix #579
TODO: check the online documentation to make sure it works properly.
closed 2021-12-03 by jhillairet