scikit-rf/scikit-rf #584: Multi-parameters NetworkSet

jhillairet, 2021-12-04:
Currently `NetworkSet` does not explicitly support the definition of multiple parameters/dimensions (issue #535). While it is possible to create a NetworkSet and interpolate the results with respect to a parameter (from the [interpolate_from_network](https://scikit-rf.readthedocs.io/en/latest/api/generated/skrf.networkSet.NetworkSet.interpolate_from_network.html) method), it only concerns a unique parameter/dimension. 

## Purpose and objectives

A desired feature would be to allow creating NetworkSet in which the parameters for which each Network has been obtained (like temperature, voltage or whatever can be changed between the Networks of the NetworkSet) can be set, then to extract or interpolate data for specific parameters.  

prerequisites: 
 * retrocompatibility (do not create a new object NetworkSetX)

below we use the same nomenclature like in [xarray](http://xarray.pydata.org/en/stable/index.html), as xarray is now a defacto  standard for N-dimensional data. So the parameters for which a network can   be stored in a NetworkSet are _dimensions_ and the values of these parameters are the _coordinates_. 

Let be M(=5) Networks which depends on 2 dimensions `a` and `b`:
 * a = 0, b = 10 -> ntwk_0
 * a = 1, b = 10 -> ntwk_1
 * a = 2, b = 10 -> ntwk_2
 * a = 0, b = 20 -> ntwk_3
 * a = 1, b = 20 -> ntwk_4

## Initializations
multi-dimensional parameters NetworkSet _desired_ behaviour pseudo-code
```python
# the frequency band should be the same for all networks of the ns (as usual)
freq = rf.Frequency(1, 10, npoints=10)
# total number of different networks
M = 5
# create M dummy networks
ntwks = [rf.Network(frequency=freq, s=np.random.rand(10,2,2), name=f'ntwk_{m}') for m in range(M) ]
# dimensions
dims = ('a', 'b')
# coordinates
a = [0,1,2]
b = [10, 20]
```
### from a list of tuples (dims, coordinates, data)
```python
# data can be a rf.Network of s-matrix ndarray (f,n,n)
data_vars = [
    (('a', 'b'), (0, 10), ntwks[0]),
    (('a', 'b'), (1, 10), ntwks[1]),
    (('a', 'b'), (2, 10), ntwks[2]),
    (('a', 'b'), (0, 20), ntwks[3]),
    (('a', 'b'), (1, 20), ntwks[4]),
    ]
ns = rf.NetworkSet(data_vars)
```
with also:
```python
ns.dims -> dict
ns.coords -> tbd
```

### from (and to) a (G)MDIF file (ADS) and MDL format (ICCAP)
issue #359 

MDIF (Measurement Data Interchange Format)
Syntax and example file https://awrcorp.com/download/faq/english/docs/Users_Guide/data_file_formats.html#generalized_mdif

```python
rf.NetworkSet.from_mdif(filename)

rf.NetworkSet.from_mdl(filename)
```

### from a xarray Dataset ?
```python 
rf.NetworkSet.from_dataset(ds)
```
where `ds` structure is TBD.

## Possible new applications
### Extracting subset of the NetworkSet
Should be compatible with ns['3-5GHz']

```python
ns.loc[dict(a=1)] -> NetworkSet of len 2
```
or
```python
ns.loc[dict(a=slice(0,3), b=10)] -> NetworkSet of len 3
```

### N-D Interpolations?
```python
ns.interpolate_from_network(dict(a=0.5, b=10), x) -> Network
```

### plotting given quantity (mean,std,etc of s,z, etc) against specific dimensions?

jhillairet, 2021-12-04:
very open to comments and suggestions of course. I will update it according to the discussion.

arsenovic, 2021-12-05:
this is a great idea, and i think one of the biggest new features for skrf.  you have outlined it out perfectly.

jhillairet, 2021-12-10:
@FranzForstmayr, I move the discussion in #585 here:

> At the moment all the dims are stored inside the NetworkSet. Wouldn't it be better to use a 'params' dictionary property in each Network and create the dims in the NetworkSet dynamically? NetworkSet holds it's Networks in a simple list, and is able to query by parameter.

That would simplify the way the Networks are looped in a NetworkSet (in fact, the way it was), but it moves the parameter-definition to the user-side. I mean, users must define the `params` dictionary themselves before creating a NetworkSet (as usual). 

@vik-s, @jrmagers and others who would like to have multi-param NetworkSet, how and when would like to define your parameters? 

PS : Actually, it is already possible to define some parameters in a Network (like `my_ntwk.temperature`) and to filter the Network in a NetworkSet using these parameters as a key. But I'm not sure that it is widely known.

FranzForstmayr, 2021-12-10:
> PS : Actually, it is already possible to define some parameters in a Network (like `my_ntwk.temperature`) and to filter the Networks in a NetworkSet using these parameters as a key. But I'm not sure that it is widely known.

As a quite regular skrf user, I can confirm that it's not widely used :)

I just think the handling is easier, because i can add the networks to multiple sets, or move the networks to another sets, without the need of handling the params. The Network creation method could be able to parse params from the comments of the sNp file. Or when creating a NetworkSet from a directory (or zip) we could use parts of the filename. The user has to submit a valid regex for parsing them.

vik-s, 2021-12-11:
@jhillairet - thank you for laying out this so clearly; this should cover multi-dimensional s-parameter datasets without having to re-architect base classes with `xarray`. I'm especially excited for the "MDL" import feature, since that is pretty much what i work with :-)

Defining `params` dictionary property for each network would work for me, and it makes sense to parse out these independent parameters from a s-parameter filename like @FranzForstmayr suggested. Given that `network` already supports `temperature` as a parameter, associating a `params` dict to a network would be logical and consistent perhaps.

jhillairet, 2021-12-14:
OK. So to be clear, something like this?
```python
params = [
        dict(a=0, b=10, c='A'),
        dict(a=1, b=10, c='A'),
        dict(a=2, b=10, c='A'),
        dict(a=0, b=20, c='A'),
        dict(a=1, b=20, c='A')
    ]
# create M dummy networks which all contain a 'params' dictionnary
ntwks_params = [rf.Network(frequency=self.freq1, 
                                s=np.random.rand(len(self.freq1),2,2), 
                                name=f'ntwk_{m}',
                                params=params) \
                     for (m, params) in enumerate(params) ]

# Create NetworkSet from a list of Network containing a .params dict parameters
ns_params = rf.NetworkSet(ntwks_params)   
```

vik-s, 2021-12-16:
This looks good to me. It is nice to make the `params` part of the `network` object like you have described, so it can be looked up when dealing with a collection of networks. Even outside of the context of multi-parameter networks, it is helpful to record some key properties associated with the network, such as date, lot, part or product number, etc.

FranzForstmayr, 2021-12-16:
This looks great!

jhillairet, 2021-12-16:
The new `.sel(param, value)` method works for simple selection (key, value)

```python
# return None is no match found
ns.sel('XX', 0)  # -> None
ns.sel('a', 500)  # -> None

# Select all Network(s) having the parameter key 'a' equal to 1 in the NetworkSet 
ns.sel('a', 1) # -> ns of len == 2

# Selection based on multiple criteria can be made by chaining sel
ns.sel('model', 'A').sel('voltage', 0).sel('temperature', 25)
```
or for iterable values:

```python
ns.sel('a', range(0,2))  # -> ns of len == 4

ns.sel('a', [0,2])  # ns of len == 4
```

vik-s, 2021-12-17:
This is nice! Will `sel()` also work with dicts?
```
ns.sel(dict(model='A', voltage=0, temperature=25))
```
It would be nice to access elements of the multi-parameter NetworkSet using the same way it was created.

jhillairet, 2021-12-17:
> This is nice! Will `sel()` also work with dicts?

Eventually, good idea.

FranzForstmayr, 2021-12-17:
Is it possible to return an empty NetworkSet instead of None?
Type definition would be easier, and there's no type error raised, if you wan't to iterate over the networks afterward.

jhillairet, 2021-12-17:
> Is it possible to return an empty NetworkSet instead of None? Type definition would be easier, and there's no type error raised, if you wan't to iterate over the networks afterward.

OK. We need to allow empty NetworkSet then.

jhillairet, 2022-01-09:
I'm working on interpolating between multiple parameters. Using the example just above, 

* The easy case: the interpolation for set of parameters `{'a': 1.2, 'X': 10}`: interpolates the s-param between the following two Networks: `{'a':1, 'X':10, 'c':'A'}` and `{'a':2, 'X':10, 'c':'A'}`, which turns out to be similar to what is already possible to do with `interpolate_from_network()`

However, there are some tricky cases I would like to discuss:

* what about interpolating for `{'a': 1.2, 'X': 17}` ? 
At the difference of [scipy.interpolate.interpn](https://docs.scipy.org/doc/scipy/reference/generated/scipy.interpolate.interpn.html#scipy.interpolate.interpn) or [interpolation in xarrays](https://xarray.pydata.org/en/stable/user-guide/interpolation.html#multi-dimensional-interpolation), here for each coordinate (a,X) we have a s-param array of shape (F,N,N)=(101,2,2) instead of a (len(a), len(X))=(3,2) array...

* what do to in case of redundant parameter values?

* which parameter value to choose if not specified ? 
Example: interpolating for {'a': 0.5} could be made either with 'X'=10 or 20...

jhillairet, 2022-01-09:
Could anyone send me some examples of  MDIF, MDL and CITI for 1, 2 and N-ports ?

FranzForstmayr, 2022-01-10:
> However, there are some tricky cases I would like to discuss:
> 
> what about interpolating for `{'a': 1.2, 'X': 17}` ?

This is more like extrapolation, I think it's fine to raise an Exception in this case.

> what do to in case of redundant parameter values?

You mean a network is well described with two parameters A and B, and then one would create a parameter C = A. The interpolation will weight each parameter for 1/3, so in fact the parameter has a weight of 2/3.....I'm not sure if this is necessary at the first step.

> which parameter value to choose if not specified ?
We could be strict and require each parameter specified, or use the median value of the unspecified parameters. 
This would mitigate interpolation error, as the mean could be far away from a sample point.

jhillairet, 2022-01-13:
OK, I propose the following API:
```python
ns.interpolate_from_params(param, x, sub_params, interp_kind)
```
where `sub_params` is a dictionary that can be used to remove ambiguity. 

Example with previous `ns`

```python
ns.interpolate_from_params('a', 0.5)  # --> will fail (ValueError) 
ns.interpolate_from_params('a', 0.5, {'X': 10})  # --> OK
```

jhillairet, 2022-01-22:
closed with PR #585

closed 2022-01-22 by jhillairet