scikit-rf/scikit-rf #586: Touchstone parser does not accept alternative encodings

CratesDeThebes, 2021-12-14:
Any encoding that is an alternative to default UTF-8 makes touchstone parsing fail.

This is because encoding should be an argument to skrf.Network and in /skrf/io/touchstone.py the line 74 that is
`fid = get_fid(file)`
should be replaced by 
`fid = get_fid(file,encoding)`

jhillairet, 2021-12-14:
Welcome and Thanks for reporting this issue. Do you have an example of a failing Touchstone file you could attach?

CratesDeThebes, 2021-12-15:
Thank you for your quick comment. Here is a MWE, it is a s2p file including a comment with a french accent and encoded with ISO-8859-15.
[test.s2p.zip](https://github.com/scikit-rf/scikit-rf/files/7718032/test.s2p.zip)

jhillairet, 2021-12-15:
Raah, sacré français alors ;-)

I can't reproduce under Windows 10:
```python
>>> import skrf as rf
>>> rf.__version__
Out[17]: '0.20.0'

>>> ntwk = rf.Network('test.s2p')
>>> ntwk.s
Out[18]: 
array([[[ 1.-1.j,  1.-1.j],
        [-1.+1.j,  1.-1.j]]])

>>> ntwk.comments
Out[16]: ' Comment with a french accent : é\n'
```
Which version and what OS are you using?

CratesDeThebes, 2021-12-15:
Tout à fait !
Here are the versions of scikit-rf and the OS I use:

```
> pip3 show scikit-rf
Name: scikit-rf
Version: 0.19.0
```

> uname -a
Linux XXXXXX 5.10.0-0.bpo.9-amd64 #1 SMP Debian 5.10.70-1~bpo10+1 (2021-10-10) x86_64 GNU/Linux

jhillairet, 2021-12-16:
@CratesDeThebes were you aware of the particular encoding of the file or not before running into the bug? 

I mean, if the encoding is on purpose and known in advance, one could specify the encoding when creating the Network, by passing an `encoding` parameter. 

If not, we need to guess the particular encoding, which is not easy and mostly a trial-and-error thing. In #587 I ended up using latin1 as a fallback if utf8 fails, which should work in most situation, but is not perfect either.

CratesDeThebes, 2021-12-16:
Thank you for the fix.

I was not aware of the encoding, I noticed the bug when I received files from users who run a different OS (MS-Windows whereas I use a GNU-Linux distro.

In my opinion, the best way would be to enable the possibility for the user to specify an encoding kwarg, defaulting this argument to UTF-8, try to guess the encoding if reading fails, falling back to latin-1 if it fails again and finally output an error if all that fails, which should be rare.

jhillairet, 2021-12-16:
OK, I've added the possibility to define the encoding type in `Network`. If not defined, `Touchstone()` makes a trial-and-error.

PR #587 (2022-01-09): fix reading ISO-8859-1 encoded Touchstone file and add encoding param to Network() and Touchstone()
Fixes #586

Guessing the encoding used in a file seems is a complicate problem per se. 

Here this fix only fixes bug #586, but the problem may still raise for other kinds of encoding... 

Is a better way exist?
closed 2022-01-09 by jhillairet