scikit-rf/scikit-rf #590: How to perform one-path two-port calibration (SOLT)?

0xCoto, 2021-12-20:
Is there a documentation page/example script for one-path two-port calibration (SOLT)? I am _only_ interested in **S11** and **S21** measurements. I'm able to take `.s2p` measurements from my NanoVNA, so I could have 4 files in total (one for each calibration standard). Would this suffice, or are more measurements necessary? I'm having a hard time understanding the 'ideal' responses described in the docs. How would I obtain them? Should I treat them like a perfect SHORT (S11=-1, S21=0), OPEN (S11=1, S21=0), LOAD (S11=0, S21=0), THRU (S11=0, S21=1)?

I've tried generating the ideals with skrf, but I get **divide by 0** issues when trying to run the calibration. Any ideas? I also tried setting S12 and S22 as `np.nan`, but skrf doesn't seem to work well with that.

```py
load_cal = input('Would you like to import a local measurement calibration from file? [Y/n]: ')
if load_cal != 'n' and load_cal != 'N':
	calibrated = True
else:
	calibrate = input('Would you like to perform a measurement calibration? [y/N]: ')
	if calibrate == 'y' or calibrate == 'Y':
		### OPEN ###
		input('Connect OPEN to Port 1 and press Enter.')
		# Measure S11 and S21
		s11 = nv.data(0)
		s21 = nv.data(1)

		f_arr = np.linspace(f_start/1e6, f_stop/1e6, num=sweep_pts)
		f = skrf.Frequency.from_f(f_arr, unit='MHz')
		s = np.zeros((len(f), 2, 2), dtype = 'complex_')
		s[:,0,0] = s11
		s[:,0,1] = np.zeros_like(s11) #np.ones_like(s11)*np.nan
		s[:,1,0] = s21
		s[:,1,1] = np.ones_like(s11) #np.ones_like(s11)*np.nan
		ntw_init = skrf.Network(frequency=f, s=s)

		ntw_init.write_touchstone('measured/open')

		### SHORT ###
		input('Connect SHORT to Port 1 and press Enter.')
		# Measure S11 and S21
		s11 = nv.data(0)
		s21 = nv.data(1)

		f_arr = np.linspace(f_start/1e6, f_stop/1e6, num=sweep_pts)
		f = skrf.Frequency.from_f(f_arr, unit='MHz')
		s = np.zeros((len(f), 2, 2), dtype = 'complex_')
		s[:,0,0] = s11
		s[:,0,1] = np.zeros_like(s11) #np.ones_like(s11)*np.nan
		s[:,1,0] = s21
		s[:,1,1] = np.zeros_like(s11) #np.ones_like(s11)*np.nan
		ntw_init = skrf.Network(frequency=f, s=s)

		ntw_init.write_touchstone('measured/short')

		### LOAD ###
		input('Connect LOAD to Port 1 and press Enter.')
		# Measure S11 and S21
		s11 = nv.data(0)
		s21 = nv.data(1)

		f_arr = np.linspace(f_start/1e6, f_stop/1e6, num=sweep_pts)
		f = skrf.Frequency.from_f(f_arr, unit='MHz')
		s = np.zeros((len(f), 2, 2), dtype = 'complex_')
		s[:,0,0] = s11
		s[:,0,1] = np.zeros_like(s11) #np.ones_like(s11)*np.nan
		s[:,1,0] = s21
		s[:,1,1] = np.ones_like(s11) #np.ones_like(s11)*np.nan
		ntw_init = skrf.Network(frequency=f, s=s)

		ntw_init.write_touchstone('measured/load')

		### THRU ###
		input('Connect THRU between Port 1 & 2 and press Enter.')
		# Measure S11 and S21
		s11 = nv.data(0)
		s21 = nv.data(1)

		f_arr = np.linspace(f_start/1e6, f_stop/1e6, num=sweep_pts)
		f = skrf.Frequency.from_f(f_arr, unit='MHz')
		s = np.zeros((len(f), 2, 2), dtype = 'complex_')
		s[:,0,0] = s11
		s[:,0,1] = np.ones_like(s11) #np.ones_like(s11)*np.nan
		s[:,1,0] = s21
		s[:,1,1] = np.zeros_like(s11) #np.ones_like(s11)*np.nan
		ntw_init = skrf.Network(frequency=f, s=s)

		ntw_init.write_touchstone('measured/thru')
		calibrated = True
if calibrated:
	### Create ideals ###
	s = np.arange(101*2*2, dtype='complex_')
	s = s.reshape((101,2,2))
	# Open
	s_open = 1*np.ones_like(s)
	s_open[:,0,1] = 0 #np.nan+np.nan*1j
	s_open[:,1,0] = 0
	s_open[:,1,1] = 1 #np.nan+np.nan*1j
	ntw_open = skrf.Network(frequency=f, s=s_open)
	ntw_open.write_touchstone('ideal/open')
	# Short
	s_short = -1*np.ones_like(s)
	s_short[:,0,1] = 0 #np.nan+np.nan*1j
	s_short[:,1,0] = 0
	s_short[:,1,1] = 1 #np.nan+np.nan*1j
	ntw_short = skrf.Network(frequency=f, s=s_short)
	ntw_short.write_touchstone('ideal/short')
	# Load
	s_load = 0.0001*np.ones_like(s)
	s_load[:,0,1] = np.nan+np.nan*1j
	s_load[:,1,0] = 0
	s_load[:,1,1] = np.nan+np.nan*1j
	ntw_load = skrf.Network(frequency=f, s=s_load)
	ntw_load.write_touchstone('ideal/load')
	# Thru
	s_thru = 0.0001*np.ones_like(s)
	s_thru[:,0,1] = 1+0j #np.nan+np.nan*1j
	s_thru[:,1,0] = 1+0j
	s_thru[:,1,1] = 0.0001 #np.nan+np.nan*1j
	ntw_thru = skrf.Network(frequency=f, s=s_thru)
	ntw_thru.write_touchstone('ideal/thru')

	# a list of Network types, holding 'ideal' responses
	my_ideals = [
		skrf.Network('ideal/thru.s2p'),
		skrf.Network('ideal/short.s2p'),
		skrf.Network('ideal/open.s2p'),
		skrf.Network('ideal/load.s2p'),
		]
	my_measured = [
		skrf.Network('measured/thru.s2p'),
		skrf.Network('measured/short.s2p'),
		skrf.Network('measured/open.s2p'),
		skrf.Network('measured/load.s2p'),
		]

	## Create SOLT instance
	cal = skrf.calibration.SOLT(
		ideals = my_ideals,
		measured = my_measured,
		)
	# Run calibration algorithm
	cal.run()
	input('Optimization complete. Press Enter to begin DUT optimization.')
```

Ttl, 2021-12-20:
`SOLT` calibration method requires two-port measurements and can't be used in this case. There is `TwoPortOnePath` method specifically for this application. See this example in documentation: https://scikit-rf.readthedocs.io/en/latest/examples/metrology/TwoPortOnePath%2C%20EnhancedResponse%2C%20and%20FakeFlip.html .

In all calibrations `ideals` is the response that should be obtained of that standard with a perfect calibration. If for example you know that open has some amount of capacitance you would need to add it to the `ideals` definition.

0xCoto, 2021-12-20:
@Ttl Thanks for your input. Could you perhaps further explain the ideal calibration a bit? Are those instrument-specific? When I calibrate my VNA (which only supports S11 and S21 measurements) with its own interface, it asks me to connect an open, short, load (port 1) and thru (port 1 to port 2).

What kind of ideals might be considered there? It certainly doesn't ask me to define anything of that nature (it just asks me to make these 4 measurements), so does the VNA have built-in ideal measurements to carry out this calibration?

(I'm not a calibration expert, so any help is appreciated)

Ttl, 2021-12-20:
They are calibration standard specific. Usually VNAs have a menu somewhere where they can be defined and the calibration kit comes with the definitions. If you are not measuring very high frequencies and are not doing very accurate measurements you might just assume they are perfect (open and short lossless, no delay and load perfectly matched) and the error shouldn't be too high.

0xCoto, 2021-12-20:
@Ttl Alright, this makes sense. I definitely don't want super accurate measurements or anything, and I'm working around 3000 MHz. So, haven't I defined the ideals properly then?
```py
### Create ideals ###
s = np.arange(101*2*2, dtype='complex_')
s = s.reshape((101,2,2))
# Open
s_open = 1*np.ones_like(s)
s_open[:,0,1] = 0 #np.nan+np.nan*1j
s_open[:,1,0] = 0
s_open[:,1,1] = 1 #np.nan+np.nan*1j
ntw_open = skrf.Network(frequency=f, s=s_open)
ntw_open.write_touchstone('ideal/open')
# Short
s_short = -1*np.ones_like(s)
s_short[:,0,1] = 0 #np.nan+np.nan*1j
s_short[:,1,0] = 0
s_short[:,1,1] = 1 #np.nan+np.nan*1j
ntw_short = skrf.Network(frequency=f, s=s_short)
ntw_short.write_touchstone('ideal/short')
# Load
s_load = 0.0001*np.ones_like(s)
s_load[:,0,1] = np.nan+np.nan*1j
s_load[:,1,0] = 0
s_load[:,1,1] = np.nan+np.nan*1j
ntw_load = skrf.Network(frequency=f, s=s_load)
ntw_load.write_touchstone('ideal/load')
# Thru
s_thru = 0.0001*np.ones_like(s)
s_thru[:,0,1] = 1+0j #np.nan+np.nan*1j
s_thru[:,1,0] = 1+0j
s_thru[:,1,1] = 0.0001 #np.nan+np.nan*1j
ntw_thru = skrf.Network(frequency=f, s=s_thru)
ntw_thru.write_touchstone('ideal/thru')

# a list of Network types, holding 'ideal' responses
my_ideals = [
	skrf.Network('ideal/thru.s2p'),
	skrf.Network('ideal/short.s2p'),
	skrf.Network('ideal/open.s2p'),
	skrf.Network('ideal/load.s2p'),
	]
my_measured = [
	skrf.Network('measured/thru.s2p'),
	skrf.Network('measured/short.s2p'),
	skrf.Network('measured/open.s2p'),
	skrf.Network('measured/load.s2p'),
	]
```

Vinc0110, 2021-12-20:
> I'm having a hard time understanding the 'ideal' responses described in the docs. How would I obtain them?

The parameter name `ideals` seems to confuse many users (including myself). Calling it "standards", "standards_response", or something similar would make more sense to me. After all, the response of real-world impedance references (calibration standards) is far from ideal, as pointed out above. @Ttl and other devs: Please don't take this the wrong way; I'm just trying to improve the usability.

Now, changing the parameter name would probably cause a mess, but how about a few more sentences in the docstrings to clear things up? Also, the tutorial and most of the examples are not really helpful when it comes to non-ideal `ideals`.

To make things worse, the docstrings for the base class and for its children are all different and the info is hidden in the constructor docs (`__init__()`). It might be easier to include the parameter descriptions on the top level of the class. For example, see the docs of `Circuit` and `Network`.

https://scikit-rf.readthedocs.io/en/latest/api/calibration/generated/skrf.calibration.calibration.Calibration.__init__.html
https://scikit-rf.readthedocs.io/en/latest/api/calibration/generated/skrf.calibration.calibration.LRM.__init__.html

https://scikit-rf.readthedocs.io/en/latest/api/generated/skrf.circuit.Circuit.html
https://scikit-rf.readthedocs.io/en/latest/api/generated/skrf.network.Network.html

arsenovic, 2021-12-21:
@0xCoto , the example @Ttl  linked is what you want.  threre is an example script on the page, that you can use  which is the best thing to do). you cant use SOLT, because you dont have a full VNA.  the way you are defining your ideals looks fine, although its a little easier to use a media type.

( also looking at that example the corrected_full is incorrect, its the same as _fakeflip. i think it is meant to have SOLT defined also to illustrate the difference...)

i like the idea of  moving __init__ docs into class docs.

jhillairet, 2021-12-22:
@Vinc0110 do not hesitate to create an issue in which we will list all the possible/desired changes. Indeed, changing the `ideals` parameter to something else may break compatibility with the previous versions. However, improving the documentation is always good. 

In particular, is someone could make a dedicated example related to NanoVNA, that would be great.

0xCoto, 2021-12-23:
Just trying to follow the documentation example. Is this better?
```py
### Create ideals ###
coax = Coaxial(frequency=f, z0=50)
my_ideals = [
	coax.short(nports=2),
	coax.open(nports=2),
	coax.match(nports=2),
	coax.thru()
]
my_measured = [
	skrf.Network('measured/short.s2p'),
	skrf.Network('measured/open.s2p'),
	skrf.Network('measured/load.s2p'),
	skrf.Network('measured/thru.s2p'),
	]

## Create SOLT instance
cal = skrf.calibration.TwoPortOnePath(
	measured = my_measured,
	ideals = my_ideals,
	)
# Run calibration algorithm
cal.run()

# Apply Pseudo-full Correction (FakeFlip)
cal.apply_cal((ntw,ntw))
```

I get these warnings:
```
/usr/local/lib/python3.7/dist-packages/skrf/calibration/calibration.py:1454: UserWarning: n_thrus is None, guessing which stds are transmissive
  warn('n_thrus is None, guessing which stds are transmissive')

/usr/local/lib/python3.7/dist-packages/skrf/calibration/calibration.py:1934: UserWarning: only gave a single measurement orientation, error correction is partial without a tuple
  warnings.warn('only gave a single measurement orientation, error correction is partial without a tuple')
/usr/local/lib/python3.7/dist-packages/skrf/mathFunctions.py:265: RuntimeWarning: divide by zero encountered in log10
  out = 20 * npy.log10(z)
```

arsenovic, 2021-12-24:
looks right to me .

0xCoto, 2021-12-24:
@arsenovic It doesn't look like the data is calibrated at all. I assume all these
```py
zero encountered in log10
  out = 20 * npy.log10(z)
```
divide by zero warnings that I'm getting might just leave the data uncalibrated to avoid crashing, so that definitely doesn't seem right.

0xCoto, 2021-12-25:
Aha! I've finally fixed it! Looks like the division error was because I tried replacing my `nan`s with`0`s etc. Apparently I was using `cal.apply_cal((ntw,ntw))` without applying it to a variable... `	ntw = cal.apply_cal((ntw,ntw))` fixed it though, and the calibration looks perfect!

Thank you very much - closing this issue for now.

closed 2021-12-25 by 0xCoto