scikit-rf/scikit-rf #592: Improving documentation for Calibration

Vinc0110, 2021-12-22:
This is a fork of issue #590 intended to collect and discuss possible improvements of the API documentation, the tutorial, and the examples for the `Calibration` module. See https://scikit-rf.readthedocs.io/en/latest/api/calibration/index.html

Here is what I have or propose so far:
- Parameter `ideals` causes confusion and should be explained and documented in more detail in the API docs.
- Unifying the parameter description of the classes `Calibration` and its successors such that they share the same (verbose) docstring and that they are explained in the class docstring rather than in the docstring for `__init__()`. 
- Proposal: Entirely remove `__init__()` from the API docs and include parameter descriptions directly in the class docstrings **in every skrf module**, just like they do in NumPy. [The NumPy style guide](https://numpydoc.readthedocs.io/en/latest/format.html) in not very explicit here, but see [numpy.array](https://numpy.org/doc/stable/reference/generated/numpy.array.html) as an example.
- Examples and use cases of calibrations with non-ideal (real-world) standards should be given [examples/tutorials]
- Add your points below

jhillairet, 2022-01-03:
- I'd like to see some documentation specific for nanoVNA calibration

Vinc0110, 2022-04-06:
- Improve parameter description of `TwoPortOnePath.apply_cal()`, see issue #623

jhillairet, 2022-04-06:
* improve SOLT:  #624