scikit-rf/scikit-rf #593: read_all() contains argument not working if string has 2 characters

denzchoe, 2021-12-23:
Wrote this script and both Touchstone files got loaded into the NetworkSet:
```
filetxt = '90'
meas = rf.read_all_networks(file_path_model, contains = filetxt)
```

> ```
> In [22]: meas
> Out[22]: 
> {'touchstone_78': 2-Port Network: 'touchstone_78',  0.0-70.0 GHz, 1004 pts, z0=[50.+0.j 50.+0.j],
>  'touchstone_90': 2-Port Network: 'touchstone_90',  0.0-70.0 GHz, 1004 pts, z0=[50.+0.j 50.+0.j]}
> ```

Tried the alternative instead and it **performs as expected** to only load the Touchstone file with '90' string

```
filetxt = '_90'
meas = rf.read_all_networks(file_path_model, contains = filetxt)
```

> ```
> In [25]: meas
> Out[25]: {'touchstone_90': 2-Port Network: 'touchstone_90',  0.0-70.0 GHz, 1004 pts, z0=[50.+0.j 50.+0.j]}
> ```

Not sure why this is happening. 
I have enclosed 2 random Touchstone files with .txt extension. Just convert them to .s2p for your test.:

</br>

[touchstone_78.txt](https://github.com/scikit-rf/scikit-rf/files/7767630/touchstone_78.txt)

[touchstone_90.txt](https://github.com/scikit-rf/scikit-rf/files/7767629/touchstone_90.txt)

wbarnha, 2022-01-03:
That's strange. I ran it on my computer and I got:
```
In[8]: read_all_networks('.')
Out[8]: 
{'touchstone_90': 2-Port Network: 'touchstone_90',  0.0-70.0 GHz, 1004 pts, z0=[50.+0.j 50.+0.j],
 'touchstone_78': 2-Port Network: 'touchstone_78',  0.0-70.0 GHz, 1004 pts, z0=[50.+0.j 50.+0.j]}
In[9]: read_all_networks('.', contains='90')
Out[9]: {'touchstone_90': 2-Port Network: 'touchstone_90',  0.0-70.0 GHz, 1004 pts, z0=[50.+0.j 50.+0.j]}
```

What version of skrf do you have installed?

jhillairet, 2022-01-03:
I also confirm it works fine for Python 3.8, Windows.

```python
rf.read_all('.')
Out[4]: 
{'touchstone_78': 2-Port Network: 'touchstone_78',  0.0-70.0 GHz, 1004 pts, z0=[50.+0.j 50.+0.j],
 'touchstone_90': 2-Port Network: 'touchstone_90',  0.0-70.0 GHz, 1004 pts, z0=[50.+0.j 50.+0.j]}

rf.read_all('.', contains='90')
Out[5]: {'touchstone_90': 2-Port Network: 'touchstone_90',  0.0-70.0 GHz, 1004 pts, z0=[50.+0.j 50.+0.j]}

rf.read_all('.', contains='_90')
Out[6]: {'touchstone_90': 2-Port Network: 'touchstone_90',  0.0-70.0 GHz, 1004 pts, z0=[50.+0.j 50.+0.j]}

```

denzchoe, 2022-01-04:
strange.. It is working again on my end now. I am not sure why it even happened the first place. 
It was working on the previous skrf version; Updated skrf, then it didn't (I had to alter the _contain_ string for my old py files)
Now it is working again. 
Thanks for testing it out.

denzchoe, 2022-01-04:
strange.. It is working again on my end now. I am not sure why it even happened the first place.
It was working on the previous skrf version; Updated skrf, then it didn't (I had to alter the contain string for my old py files)
Now it is working again.
Thanks for testing it out.

closed 2022-01-04 by denzchoe