scikit-rf/scikit-rf #594: `Network` constructor fails if `file` is a `StringIO` stream

mbastani, 2022-01-04:
The constructor of `Network` class allows the `file` argument to be either a string (path to the touchstone) or a file-like object (as the comment [here](https://github.com/scikit-rf/scikit-rf/blob/a1c7caec4f3219ac9f5050031b40b4608adf33ef/skrf/network.py#L419) states).

However, passing a file-like stream to the constructor raises an `AttributeError`, as the following MWE shows:

    import io
    import skrf as rf

    with open('path/to/touchstone/file', 'r')  as fh:
        contents = fh.read()

    stream = io.StringIO(contents)
    
    net = rf.Network(stream)  # <--- raises AttributeError
       

The issue is that `StringIO` does not have a `name` attribute. And it seems that knowing the file name is necessary for constructing the network.  (As the number of ports need to be determined from the file extension.)

FranzForstmayr, 2022-01-04:
Assign a name property on the stream as a quick workaround

```python
stream = io.StringIO(contents)
stream.name = 'simple-network.s2p'
net = rf.Network(stream) 
```

Not sure, if this is a bug or a feature :)

jhillairet, 2022-01-04:
It is a kind of feature let's say ;)

The documentation of Touchstone reading is under update in #587 but not merged yet. It will be:

```python
        Examples
        --------
        From filename
        >>> t = rf.Touchstone('network.s2p')
        
        File encoding can be specified to help parsing the special characters:
        
        >>> t = rf.Touchstone('network.s2p', encoding='ISO-8859-1')
        From file-object
        >>> file = open('network.s2p')
        >>> t = rf.Touchstone(file)
        
        From a io.StringIO object
        
        >>> link = 'https://raw.githubusercontent.com/scikit-rf/scikit-rf/master/examples/basic_touchstone_plotting/horn antenna.s1p'
        >>> r = requests.get(link)
        >>> stringio = io.StringIO(r.text)
        >>> stringio.name = 'horn.s1p'  # must be provided for the Touchstone parser
        >>> horn = rf.Touchstone(stringio)
```

mbastani, 2022-01-04:
I'm not sure if requiring the user to add a property to the (standard) `StringIO` is a good solution. 

In fact, by definition, a [file-like object](https://docs.python.org/3/glossary.html#term-file-object) is only required to implement `read`  and `write` methods in its API.   So, either the constructor should be changed to use only the `read` method, or the argument of the constructor must be asked to implement a specific (more stringent) API that includes a `name` property. 

I understand that the complication arises because, according to the [Touchstone File Format Specification](https://ibis.org/touchstone_ver2.0/touchstone_ver2_0.pdf), in v1.0 files, the number of ports is (indirectly) embedded into the file extension rather than being recorded inside the file.

jhillairet, 2022-06-11:
I don't really like the idea of adding a mandatory name to pass. I'd rather prefer to document how creating a Network from a `StringIO` in the [Network Tutorial](https://scikit-rf.readthedocs.io/en/latest/tutorials/Networks.html#from-other-network-parameters-(Z,-Y,-ABCD,-T)). which anyway should be revised now to reflect the latest changes simplifying creating Network from other parameters than S. Any comment on this?

closed 2024-11-21 by FranzForstmayr