scikit-rf/scikit-rf #606: VF: Varying order phenomenon

dxxx9, 2022-02-04:
Hey guys!
I stumbled across something strange and hope you can help me here.

This [coax_samples.zip](https://github.com/scikit-rf/scikit-rf/files/8006347/coax_samples.zip) contains two `.s2p` files of two simple coax cables which vary slighly in their geometry (inner/ outer diameter, eps_r and length).

I'm using the VectorFitting implementation with __no constant and proportionals terms__. I also don't want any pure real poles, just complex conjugated poles.

With the `coax_sample_true.s2p` it works as expected. But I can't explain the results of `coax_sample_false.s2p`.

Not only are the two real poles included but the order is also increased by one (`n_poles_cmplx + 1`). If you increase the number of complex starting poles 

__So my questions in general are:__
- Why is the algorithm adding more poles? Is there a reason for that?
- Why are there real poles although I have excluded them in the funtion call?
- And is there a way to prevent this from happening?

You can use these lines of code to reproduce this behavior:
```python
import numpy as np
import skrf as rf

netws = ['coax_sample_true.s2p', 'coax_sample_false.s2p']
vf_list = []
for e in netws:
    netw = rf.Network(e)
    vf = rf.VectorFitting(netw)

    vf.vector_fit(
        n_poles_real=0,
        n_poles_cmplx=7, # the behavoir remains the same even when varying this number
        fit_proportional=False,
        fit_constant=False
    )
    print('Order:', vf.poles.size)
    print(vf.poles)

    vf_list.append(vf)

```

dxxx9, 2022-02-05:
@Vinc0110 I think you are the expert here.

Vinc0110, 2022-02-05:
Hi. I'll have a look at your examples. 

Regarding your general questions:
- The type of poles cannot be constrained to real-only or complex-only (at least not our implementation in scikit-rf). The only thing you can really enable or disable are the constant and proportional terms.
- During the relocation process, complex-conjugate poles can become real poles (and vice versa). Please also see [this comment](https://scikit-rf.readthedocs.io/en/latest/examples/vectorfitting/vectorfitting_ex2_190ghz_active.html#comment) in the docs, which probably explains what you get in your examples.
- I don't think you can prevent this behavior, but maybe there is some clever workaround. Maybe certain choices for the starting poles happen to end up as a bunch of complex-only poles. However, in general, if the algorithm places a real pole for your data, that probably means your data requires a real pole for a successful fit. After all, the objective for the algorithm is to minimize the deviation between the model and the data (least-squares).

Out of curiosity: what is wrong with real poles and why did you disable the constant term? For the fitting of scattering parameters, it is usually helpful to include the constant (unlike the proportional term, which is for admittance parameters and makes it worse for scattering parameters).

dxxx9, 2022-02-05:
My usecase includes neural networks which require a dataset with objects of constant length. Therefore the order must not change arbitrarily!

I refrain to use constant or proportional terms since the implementation is less complicated when you don't have to distinguish between different types of parameters (= only a sum of partial fractions). So just a practical reason.

Ok, thanks. I'll try to find a workaround.

Vinc0110, 2022-02-05:
That makes sense. But I'm afraid the current implementation in scikit-rf is not optimal for what you want to to with it. Being an open source project, you can of course have a look at our implementation and try to modify it according to your requirements. 

In PR #605 you mentioned the Matlab implementation. Does it work better for you? Do they have an option to use only complex-only poles for the model (I don't use Matlab and never tried it)? And did you try [Matlab's rationalfit routine](https://de.mathworks.com/help/rf/ref/rationalfit.html) or [vfit3 by Gustavsen] (https://www.sintef.no/projectweb/vectorfitting/downloads/vfit3/), the inventor of vector fitting?

dxxx9, 2022-02-05:
Yes, I used the original `vfit3` from Gustavsen's website. (Didn't know that matlab made their own copy)

But I didn't go much into detail because I'm no longer used to the MATLAB syntax. ^^
So I haven't checked if the behavior mentioned above is identical to your implementation. The only thing I noticed was the extremely fast processing time.

Vinc0110, 2022-02-06:
After a closer look at your example networks I can confirm my earlier speculations with 1 complex-conjugate pair splitting into 2 real poles. For your example networks, starting with 0 real poles and 6 complex-conjugate poles (0+6) results in the same complex-only configuration, but that's just a coincidence, I guess. The RMS error is still acceptable, but not great. It gets better if you include the constant term. See my results below.

I'm not sure if this is any help for you, but if you define the model order differently, it becomes completely predicable and the starting order matches the final order of the model. A complex pole in `vf.poles` is actually one part of a complex-conjugate pair, even though only one part is stored in `vf.poles` (the one with the positive imaginary part). So, the "correct" model order is actually `N = N_real + 2 * N_complex`, which does not change with the type of the poles.

Fitting of `coax_sample_true.s2p` **without** constant or proportional term:
- 0+8 starting poles (order 0+16=16) --> 0+8 final poles (order 0+16=16); RMS error = 0.00007
- 0+7 starting poles (order 0+14=14) --> 0+7 final poles (order 0+14=14); RMS error = 0.00083
- 0+6 starting poles (order 0+12=12) --> 0+6 final poles (order 0+12=12); RMS error = 0.00792

Fitting of `coax_sample_false.s2p` **without** constant or proportional term:
- 0+8 starting poles (order 0+16=16) --> 2+7 final poles (order 2+14=16); RMS error = 0.00018
- 0+7 starting poles (order 0+14=14) --> 2+6 final poles (order 2+12=14); RMS error = 0.00063
- 0+6 starting poles (order 0+12=12) --> 0+6 final poles (order 0+12=12); RMS error = 0.00444

Fitting of `coax_sample_true.s2p` **with** constant term:
- 0+8 starting poles (order 0+16=16) --> 0+8 final poles (order 0+16=16); RMS error = 0.00006
- 0+7 starting poles (order 0+14=14) --> 0+7 final poles (order 0+14=14); RMS error = 0.00050
- 0+6 starting poles (order 0+12=12) --> 0+6 final poles (order 0+12=12); RMS error = 0.00553

Fitting of `coax_sample_false.s2p` **with** constant term:
- 0+8 starting poles (order 0+16=16) --> 0+8 final poles (order 0+16=16); RMS error = 0.00007
- 0+7 starting poles (order 0+14=14) --> 2+6 final poles (order 2+12=14); RMS error = 0.00041
- 0+6 starting poles (order 0+12=12) --> 0+6 final poles (order 0+12=12); RMS error = 0.00299

Vinc0110, 2022-02-06:
Thinking about some more, if you write the individual terms of the model (ignoring *d* and *h*) in
[image]
you will always get the length (number of summands) calculated with `N = N_real + 2 * N_complex`, which matches the order of your starting poles, regardless of their types. Isn't that exactly what you want for your neural network?

dxxx9, 2022-02-07:
Thanks for the hint. That could work as well.

Found another approach in literature.
You shift the frequency vector far away from the zero frequency. That results in no more real poles without changing the shape of the sparams. Solves the issue very well. 👍

Vinc0110, 2022-02-07:
How does that work, I don't quite understand. Can you share the reference to literature (or some example code)?

Here is how I tried to do it:
```
nw = skrf.Network('/home/vinc/Downloads/coax_sample_false.s2p')
f_shift = 1e12

# frequency-shifted VF
nw.frequency.f += f_shift
vf_shifted = skrf.VectorFitting(nw)
vf_shifted.vector_fit(n_poles_real=0, n_poles_cmplx=7, fit_constant=False, fit_proportional=False)

# un-shift parameters
nw.frequency.f -= f_shift
vf = skrf.VectorFitting(nw)
vf.poles = vf_shifted.poles - 2j * np.pi * f_shift
vf.residues = vf_shifted.residues
vf.constant_coeff = vf_shifted.constant_coeff
vf.proportional_coeff = vf_shifted.proportional_coeff
```

The fit is fine, but when shifting it back, something happens. Only S11 and S22 go bad for some reason... (RMS error = 0.90791). And there is this massive peak just below the fitting interval. 

**Shifted VF**:
[image]

**Un-shifted VF**:
[image]

**Un-shifted VF extended range**:
[image]

Vinc0110, 2022-02-07:
Ah ok, thanks! So it's some kind of trick, i.e. approximation, but whatever. I'm glad it solves your problem.

closed 2022-02-07 by dxxx9