scikit-rf/scikit-rf #607: se2gmm behavior is not expected

ASLOBEAR, 2022-02-24:
Hello,
I am still having trouble using the se2gmm network method. The new documentation helps a ton, thanks for the improvements. Here is the issue I am working through.
16 port network, all single ended and need to be differential pairs. when I read in the touchstone file, the ports are ordered as:
[Port1_near_N,Port1_far_N,Port1_near_P,Port1_far_P, Port2_near_P,Port2_far_P,Port3_near_P,Port3_far_P, Port2_near_N,Port2_far_N,Port3_near_N,Port3_far_N,Port4_near_P,Port4_far_P,Port4_near_N,Port4_far_N,]

I would like these to be ordered [Port1_near_P, Port1_near_N, Port1_far_P, Port1_far_N... etc]
I have a list of from_ports and to_ports for renumbering as:
from_ports = [2, 3, 4, 5, 6, 7, 12, 13, 0, 1, 8, 9, 10, 11, 14, 15]
to_ports = [0, 2, 4, 6, 8, 10, 12, 14, 1, 3, 5, 7, 9, 11, 13, 15]

So, when the ntwk.se2gmm(p=8) method is ran, I expect the results to be:
[Port1_near_differential, Port1_near_common, Port1_far_differential, Port1_far_common... etc]

However this does not seem to be how the result is organized. I will try to recreate the results with "dummy" data that can be shared, but is there anything obvious I'm missing?

denzchoe, 2022-02-25:
If I am not wrong, this should be your renumbering 

File sequence: 
[Port1_near_N, Port1_far_N, Port1_near_P, Port1_far_P, Port2_near_P, Port2_far_P, Port3_near_P, Port3_far_P, Port2_near_N, Port2_far_N, Port3_near_N, Port3_far_N, Port4_near_P, Port4_far_P, Port4_near_N, Port4_far_N,]

Target:
[Port1_near_P, Port1_near_N, Port1_far_P, Port1_far_N... etc]

Renumbering sequence: ( I drew some lines for clarity. Remove '|' for your code)
```
your_semodeNwk.renumber([0, 1, 2, 3, |4, 5, 6, 7, |8, 9, 10, 11, |12, 13, 14, 15], \
                        [1, 3, 0, 2, |5, 7, 4, 6, |9, 11, 8, 10, |13, 15, 12, 14]) 
```

See [this discussion as well:](https://groups.google.com/g/scikit-rf/c/AoKAHpWlkcc/m/GpuTC5FbBQAJ)

denzchoe, 2022-02-25:
Somewhat related, just be aware of it after you done your renumbering. 
I am not sure whether a fix might have been included on latest version 0.21.0 but not checked

be aware of the issue reported here
https://github.com/scikit-rf/scikit-rf/issues/367#issuecomment-832012155

and here:
https://groups.google.com/g/scikit-rf/c/1tLll6hWF9w/m/QonKwpxEBwAJ

ASLOBEAR, 2022-02-25:
I suppose I don't understand the new documentation as well as I thought I did. Thank you for the help. So the from_ports should just be a linear sequence? I don't know if I've seen an example that doesn't follow this.

ASLOBEAR, 2022-02-25:
I have created some dummy data, attached [Diff_Pair_Ntwk.txt](https://github.com/scikit-rf/scikit-rf/files/8143132/Diff_Pair_Ntwk.txt). The new "to_port" order for this file should be:
`[0,2,1,3, 4,6,5,7, 8,10,9,11, 12,14,13,15]`
When I plot the near self differential terms:
```
ntwk.plot_s_db(0,0)
ntwk.plot_s_db(4,4)
ntwk.plot_s_db(8,8)
ntwk.plot_s_db(12,12)
```
I expect these all to be the same. However, I see

[image]

I think this should indicate something is still amiss

JAnderson419, 2022-02-25:
I am not sure if this is helpful or not, but the general form of the matrix from the documentation:

[image]

is the following form from the reference paper:
 
[image]

So you are plotting Sdd11 and Sdd55, as well as Scc11 and Scc55. I cannot tell from the plot, but I am assuming that the blue and orange curves are on top of each other, and similarly for the green and red?

I should probably update the tutorial with the general form to clarify it for people using higher numbers of ports. 

As far as renumbering goes, the from_ports and to_ports exists so that you could renumber only a subset of the matrix if you wanted (for example `from_ports=[2,3]` , to_ports=[3,2]` to swap ports 3 and 4). The function maps `from_ports[i]` to `to_ports[i]`. If you are renumbering the whole matrix, it makes sense to just have a linear sequence as it makes it easier for you to define `to_ports`. 

I could update the renumbering function with a default value for `from_ports` to allow it to be used with `to_ports` only. The tricky thing is that `from_ports` is first, so it wouldn't work with positional arguments. It would have to be called as `ntwk.renumber(to_ports=[0,2,1,3])` or I would have to break backward compatibility with positional arguments, since I don't think you can have multiple constructors (ex. `renumber(to_ports)` and `renumber(from_ports, to_ports)`) in python. Others with more experience feel free to chime in on this.

#367 has not been resolved yet as far as I am aware. I have tried to look through the math once or twice but have not found the place where the -1 factor is coming from. The result could possibly be multiplied by -1, although it might be more complicated for matrices with mixed mode and single ended ("e") ports present.

ASLOBEAR, 2022-02-25:
ok. With the updated guidance, I can plot Sdd11, Sdd22, Sdd33, Sdd44 with:
```
ntwk.plot_s_db(0,0)
ntwk.plot_s_db(1,1)
ntwk.plot_s_db(4,4)
ntwk.plot_s_db(5,5)
```
and the results are expected:
[image]

When I add the code I expect to plot Sdd55, Sdd66, Sdd77, and Sdd88:
```
ntwk.plot_s_db(8,8)
ntwk.plot_s_db(9,9)
ntwk.plot_s_db(12,12)
ntwk.plot_s_db(13,13)
```
It again diverges:
[image]

JAnderson419, 2022-02-25:
You can plot the Sdd parameters with the following command:
```
nPorts = 8  # number differential ports
for i in range(nPorts): 
   ntwk.plot_s_db(i,i) # Sdd
```

Since the Sdd, Sdc, Scd, Scc matrixies are PxP (where P=# differential ports), the dd terms are in s[:, **0-7**, **0-7**], the cc terms are in s[:, p+**0-7**, p+**0-7**], the dc terms are in s[:, **0-7**, p+**0-7**] and cd are in s[:, p+**0-7**, **0-7**], where **0-7** is replaced by the number of the differential port you are trying to access.

closed 2022-02-25 by ASLOBEAR

ASLOBEAR, 2022-02-26:
I did not realize the six quadrants of the matrix of Fig 37 (for my case, 4 quadrants, as all SE pairs are converted to differential) is the mapping to the new ports array. This was a tough concept to grasp from the 3 port example

ASLOBEAR, 2022-02-27:
it is not clear to me if differential to differential crosstalk can be analyzed using this method. The paper referenced seems to ignore these effects since it is not present in the example circuit. Is this true in the general case?