scikit-rf/scikit-rf #610: Offset loss Keysight Calkit model

tuxgvf, 2022-02-27:
Hi guys,
It seems to me that the example for defining a Calkit from thé polynomial model, the offset loss are missing (the famous unit in Gohm/s). 
It can be a good complement to approach metrology measurement.
Thanks and best.

jhillairet, 2022-02-27:
Hello,
Which example are you talking about?

tuxgvf, 2022-02-27:
Hello Julien,

I can't remember exaclty where I found it but it's something like this:

import sys
import skrf
import matplotlib.pyplot as plt
import numpy as np
skrf.stylely()

def tline_input(zl, z0, t, f):
    c = 299792458
    w = c/f
    b = np.pi*2/w
    l = t*c
    return z0*(zl+1j*z0*np.tan(b*l))/(z0+1j*zl*np.tan(b*l))

def gamma(zl, z0):
    return (zl-z0)/(zl+z0)

def make_open(freqs, c0, c1, c2, c3, offset_t, offset_z0):
    c_coefs = [c0*1e-15, c1*1e-27, c2*1e-36, c3*1e-45]

    reactance = []
    for f in freqs:
        c = 0
        for i in xrange(len(c_coefs)):
            c += c_coefs[i]*f**i
        xc = (-1.0j/(2*np.pi*f*c))
        reactance.append(xc)
    reactance = np.array(reactance)
    sparam = gamma(tline_input(reactance, offset_z0, offset_t, freqs), 50)

    return skrf.Network(s=sparam, f=freqs, f_unit='Hz')

def make_short(freqs, l0, l1, l2, l3, offset_t, offset_z0):
    c_coefs = [l0*1e-12, l1*1e-24, l2*1e-34, l3*1e-42]

    reactance = []
    for f in freqs:
        c = 0
        for i in xrange(len(c_coefs)):
            c += c_coefs[i]*f**i
        xc = (1.0j*2*np.pi*f*c)
        reactance.append(xc)
    reactance = np.array(reactance)
    sparam = gamma(tline_input(reactance, offset_z0, offset_t, freqs), 50)

    return skrf.Network(s=sparam, f=freqs, f_unit='Hz')

def make_load(freqs, r0, l0, l1, l2, l3, offset_t, offset_z0):
    c_coefs = [l0*1e-12, l1*1e-24, l2*1e-34, l3*1e-42]

    reactance = []
    for f in freqs:
        c = 0
        for i in xrange(len(c_coefs)):
            c += c_coefs[i]*f**i
        xc = (1.0j*2*np.pi*f*c)
        reactance.append(xc)
    reactance = np.array(reactance)
    sparam = gamma(tline_input(r0+reactance, offset_z0, offset_t, freqs), 50)

    return skrf.Network(s=sparam, f=freqs, f_unit='Hz')

o = skrf.Network('open.s1p')
s = skrf.Network('short.s1p')
l = skrf.Network('load.s1p')
dut = skrf.Network(sys.argv[1])

freqs = o.frequency

o_i = make_open(freqs, 4.46941071, -1353.09191861,  2189.61227292,  -143.3687188, 31.1e-12, 50)
s_i = make_short(freqs, 64.1413939,  -11375.46199962,  -3975.09654213,    -77.69120398, 31.1e-12 ,50)
l_i = make_load(freqs, 50.0, 95.97419879,  -61.62497249,  262.22809906, -881.59180129, 31.1e-12, 50)

cal = skrf.OnePort(\
        measured = [o, s, l],
        ideals =[o_i, s_i, l_i]
        )

coefs = cal.coefs
print coefs.keys()

for k in coefs.keys():
    plt.figure()
    plt.title(k)
    plt.plot(20*np.log10(np.abs(coefs[k])))

plt.figure()
dut_cal = cal.apply_cal(dut)
dut_cal.plot_s_db()
plt.show(block=True)

dwidmann, 2022-04-06:
I expect it would be trivial to add that in to your example. 

[Link](https://ena.support.keysight.com/e5071c/manuals/webhelp/eng/measurement/calibration/advanced_calibrations/changing_the_calibration_kit_definition.htm)

loss [Ω/s] = loss[dB] * Z_0 [Ω]  / ( 10*log10(e) [dB]  * delay [s] )

loss [dB] = loss [Ω/s] * 10*log10(e) [dB] * delay [s] / Z_0 [Ω]

# this is a one-way loss, probably need to double it, and then subtract it from the example you showed. 

How much effect will it have? 

# using the 85062D short as an example
# Z0 = 50 [Ω]
# offset loss = 2.36e9 [Ω/s]
# offset delay = 31.785e-12 [s]
loss(dB) = .006516 dB  # note: this is a one-way loss

The effect is pretty small. If it scales well with frequency, you might be able to multiply it by sqrt ( f / 1e9), else I think people just adjust the offset loss to suit the frequency they're working at, or ignore it because it's so small.

dwidmann, 2022-04-06:
I've modified that example to add in the offset loss and also suit my purposes. 
`
#!/usr/bin/env python3
import matplotlib.pyplot as plt
import numpy as np
import skrf
import os
from skrf.media import Coaxial
from skrf.media import Freespace
skrf.stylely()

np_to_db = 10*np.log10(np.exp(1))

EPSILON = 1e-10

def tline_input(zl, z0, t, f):
    c = skrf.constants.c
    w = c/f
    b = np.pi*2/w
    l = t*c
    return z0*(zl+1j*z0*np.tan(b*l))/(z0+1j*zl*np.tan(b*l))

def gamma(zl, z0):
    return (zl-z0)/(zl+z0)

def offset_loss_db(offset_l,offset_t,offset_z0):
    return offset_l * np_to_db * offset_t / offset_z0

def make_open(freqs, c0, c1, c2, c3, offset_l, offset_t, offset_z0):
    c_coefs = [c0*1e-15, c1*1e-27, c2*1e-36, c3*1e-45]

    reactance = []
    for f in freqs.f:
        c = 0
        for i in range(len(c_coefs)):
            c += c_coefs[i]*f**i
        xc = (-1.0j/(2*np.pi*f*c))
        reactance.append(xc)
    reactance = np.array(reactance)
    sparam = gamma(tline_input(reactance, offset_z0, offset_t, freqs.f), 50)
    net = skrf.Network(s=sparam, f=freqs.f, f_unit='Hz')

    media = skrf.Freespace(frequency=freqs,z0=50)
    lossvalue = -2 * offset_loss_db(offset_l=offset_l, offset_t=offset_t, offset_z0=offset_z0)
    loss = media.attenuator(d=0,s21=lossvalue,db=True)

    modnet = loss ** net
    modnet.plot_s_db()
    return modnet

def make_short(freqs, l0, l1, l2, l3, offset_l, offset_t, offset_z0):
    c_coefs = [l0*1e-12, l1*1e-24, l2*1e-34, l3*1e-42]

    reactance = []
    for f in freqs.f:
        c = 0
        for i in range(len(c_coefs)):
            c += c_coefs[i]*f**i
        xc = (1.0j*2*np.pi*f*c)
        reactance.append(xc)
    reactance = np.array(reactance)
    sparam = gamma(tline_input(reactance, offset_z0, offset_t, freqs.f), 50)
    net = skrf.Network(s=sparam, f=freqs.f, f_unit='Hz')

    media = skrf.Freespace(frequency=freqs,z0=50)
    lossvalue = -2 * offset_loss_db(offset_l=offset_l, offset_t=offset_t, offset_z0=offset_z0)
    loss = media.attenuator(d=0,s21=lossvalue,db=True)

    modnet = loss ** net
    modnet.plot_s_db()
    return modnet

#def make_load(freqs, r0, l0, l1, l2, l3, offset_t, offset_z0):
#    c_coefs = [l0*1e-12, l1*1e-24, l2*1e-34, l3*1e-42]
#
#    reactance = []
#    for f in freqs.f:
#        c = 0
#        for i in range(len(c_coefs)):
#            c += c_coefs[i]*f**i
#        xc = (1.0j*2*np.pi*f*c)
#        reactance.append(xc)
#    reactance = np.array(reactance)
#    sparam = gamma(tline_input(r0+reactance, offset_z0, offset_t, freqs.f), 50)
#
#    return skrf.Network(s=sparam, f=freqs, f_unit='Hz')

def make_thru(freqs, media, offset_l, offset_t, offset_z0):

    line = media.line(d=offset_t, unit='s', z0=offset_z0+EPSILON,embed=True)
    lossvalue = -1 * offset_loss_db(offset_l=offset_l, offset_t=offset_t, offset_z0=offset_z0)

    loss = media.attenuator(s21=lossvalue,db=True)

    modnet = loss ** line
    return modnet

if __name__=="__main__":
    # following assume the 85062D kit
    # Short characteristics
    l0 = 2.0765#e-12
    l1 = -108.54#e-24
    l2 = 2.1705#e-33
    l3 = -0.01#e-42
    offset_loss_short = 2.36e9
    offset_delay_short = 31.785e-12
    z0 = 50

    # open characteristics
    c0 = 49.433#e-15
    c1 = -310.13#e-27
    c2 = 23.168#e-36
    c3 = -0.15966#e-45
    offset_loss_open = 2.2e9
    offset_delay_open = 29.243e-12
    # z0 = 50

    # load characteristics
    # 85062D kit doesn't define load characteristics, assumes perfect match, is typically >36dB return loss

    # "flush" thru characteristics
    offset_loss_thru = 2.51e9
    offset_delay_thru = 94.75e-12
    # z0 = 50

    freqs = skrf.Frequency(start=10e6,stop=6e9,npoints=1001,unit='Hz')
    media = Coaxial(frequency=freqs,z0=50,Dint=.00127,Dout=.00292)

    o_i = make_open(freqs, c0=c0, c1=c1, c2=c2, c3=c3, offset_l=offset_loss_open, offset_t=offset_delay_open, offset_z0=z0)
    s_i = make_short(freqs, l0=l0, l1=l1, l2=l2, l3=l3, offset_l=offset_loss_short, offset_t=offset_delay_short, offset_z0=z0)
    l_i = media.load(Gamma0=EPSILON)
    t_i = make_thru(freqs, media=media,offset_l=offset_loss_open, offset_t=offset_delay_open, offset_z0=z0)

    o_i.write_touchstone(filename='85062D_open.s1p',dir=os.environ['HOME'],form='db')
    s_i.write_touchstone(filename='85062D_short.s1p',dir=os.environ['HOME'],form='db')
    l_i.write_touchstone(filename='85062D_load.s1p',dir=os.environ['HOME'],form='db')
    t_i.write_touchstone(filename='85062D_thru.s2p',dir=os.environ['HOME'],form='db')
`

biergaizi, 2022-05-13:
Since the "example defining a Calkit from the polynomial model" doesn't exist in this project, the reporter was certainly confused by an unofficial example. So let's redefine this issue as a request for two features:

1. Include an example for creating network definition from Keysight's calibration kit coefficients in the documentation.

    * I happen to be working on a guide on this today. I only found this open issue by surprise...

2. Should scikit-rf include some built-in functions to do this task automatically?

biergaizi, 2022-05-13:
scikit-rf has `media.resistor()` just like `.capacitor()` and `inductor()`, and it can be frequency-dependent. So I think usually it's adequate to model this offset loss as a resistor, without going through the extra steps of converting to a loss on the dB scale.

jhillairet, 2022-05-13:
>     2. Should scikit-rf include some built-in functions to do this task automatically?

Eventually, yes. A `CalKit` class?

biergaizi, 2022-05-14:
When writing the guide, I found there are two definition styles for calibration standards. One is the HP style, offset delay is defined as a time delay and offset loss is defined as a frequency-dependent resistor. Another one is the Anritsu & Rohde Schwarz style, offset delay is defined as a physical length and offset loss is defined as a frequency-dependent attenuation in decibel. The potential future `CalKit` should have provisions to accept both.

jhillairet, 2022-05-14:
Related issues: #256, #317, #609, #663

jhillairet, 2022-05-14:
> > ```
> > 2. Should scikit-rf include some built-in functions to do this task automatically?
> > ```
> 
> Eventually, yes. A `CalKit` class?

@biergaizi, would like to implement it? Or would you prefer someone else to do it?

biergaizi, 2022-05-14:
I'd certainly like to contribute. I think the example code given in my work-in-progress documentation already suffices as a basic implementation. But I'm not really familiar with the overall design choices in scikit-rf and I'm not sure how the API should look like.

biergaizi, 2022-05-14:
> Related issues: https://github.com/scikit-rf/scikit-rf/issues/256, https://github.com/scikit-rf/scikit-rf/issues/238, https://github.com/scikit-rf/scikit-rf/issues/317, https://github.com/scikit-rf/scikit-rf/issues/609, https://github.com/scikit-rf/scikit-rf/pull/663

Ah, I don't know this feature has such a high demand, I think I need to try finalizing the documentation ASAP before someone reports yet another duplicate request...

biergaizi, 2022-05-22:
> I'd certainly like to contribute. [...] But I'm not really familiar with the overall design choices in scikit-rf and I'm not sure how the API should look like.

Still not sure about the API. So far, I think the API should:

1. Have provisions to accept Keysight, R&S and Anritsu formats. So there should be at least two classes for Keysight and R&S. Perhaps even a separate Anritsu class, since R&S uses per GHz and Anritsu uses per Hz, a separate class for each vendor can avoid the tedious and error-prone process of typing exponents after the numbers.
2. Have provisions to create both coaxial and waveguides standards. And waveguides include both circular and rectangular ones. How can this be done in the API? Creating a separate classes will cause a combinational explosion of 3x3=9 classes. Perhaps the API should accept a physical medium object instead, so one can reuse the same initializer.

jhillairet, 2022-05-23:
To my opinion, a `CalKit` is, like in reality, <strike>a very beautiful and expensive box</strike> a collection of standard components. In the scikit-rf world, it thus would a collection of `Network`, aka a `NetworkSet`.

Personally, I'd prefer having only one `CalKit` class than multiple ones hard to remember. The choice of the calkit type and brand could be passed in the `CalKit.__init__`. This does not prevent that under the hood multiple classes to be used, inheriting from an abstract base class, where some abstract methods should be implemented depending on the type and brand.

Finally, we can imagine that the `CalKit` object to be used directly in calibration methods. (`ideals=my_cal_kit`)

jjoonathan, 2022-05-25:
How do autocals fit in? I have a bunch of code that drives Anritsu 3658[25] autocals and I'm slowly working it into shape for an eventual PR. Right now I have:

* One class that loads the .ACD file characterizing the cal kit (this contains S parameters, serial control codes, and metadata)
* One class that drives the VNA and Autocal together to obtain measurements (right now it's specialized to 8510 because switching the autocal in the inner loop of a chunked sweep is literally 30 minutes faster, but I'll make a generic version)

To my intuition the first class feels like it should be a CalKit subclass and the second class feels like a piece of functionality that might attach to the CalKit subclass or perhaps VNA classes. I suppose structuring it as an init kwarg on CalKit and a utility function would work too, though. What do you think?

biergaizi, 2022-05-30:
My original proposal was pretty simple: Just create a built-in class for generating 2-port networks (i.e. S parameters) from various static calkit parameters. How these 2-port networks are used is up to the user. Then @jhillairet proposed better integration, perhaps the calkit can be an object directly passable into the calibration methods.

If we consider electronic calkits, since their S-parameters are already known, my first thought is that it's out of scope, at least not within the scope of the initial task. But on second thought, a ".ACD file characterizing the cal kit" can also be thought as the "parameters" that defines calkits, and there needs to be a standardize interface for getting control codes and metadata. So yes, it sounds to me that there should be a (sub)class for electronic calkits as well.

jhillairet, 2022-06-11:
@tuxgvf and @dwidmann, have you seen the recent documentation made by @biergaizi on the calkit creation with scikit-rf?
https://scikit-rf.readthedocs.io/en/latest/examples/metrology/SOLT%20Calibration%20Standards%20Creation.html

I'm closing this ; we will discuss the API of a possible future `CalKit` class in a specific thread here: https://github.com/scikit-rf/scikit-rf/discussions/699

closed 2022-06-11 by jhillairet